openwrt/packages #7988: <watchcat>: Not so much an issue but a feature?

nbollman, 2019-01-19:
I love the function of watchcat, it works, however in my situation a simple restart of a network interface, or specifically disabling and re-enabling my wifi radio (not my router) fixes my internet connection. (~45 sec down time vs 10 sec) Would it be possible to add a feature to watchcat when loosing the internet connection to trigger a reset of a specific interface? If this already exists would you mind pointing me in the right direction?

Currently using a heavily modded ExpressVPN Router Firmware based on Chaos Calmer, r49389.  I connect to a public (RV Park) WiFi with the ExpressVPN tunnel for my WAN. Sometimes the connection drops, the link still established, doesn't pass data until I disable and enable the wifi connection, and if I catch it fast enough I don't have to re connect the VPN... Any help advise would be much appreciated. Thank you coders!

feckert, 2019-05-25:
You could add a call to /etc/init.d/network resart at this line https://github.com/openwrt/packages/blob/b676ca325f7a247568e5a9823a44b8ed58f0a5de/utils/watchcat/files/watchcat.sh#L20 and add also a option if we should do a restart of the system or a network restart

yuriybabiy, 2019-11-11:
Hi everyone.
I have the same problem. It is enough to restart only the "4G" interface, not the whole system.
It would be good to have the following settings in the watchcat application.