openwrt/packages #7989: pciutils: not buildable

damnms, 2019-01-19:
Maintainer: @neheb  (find it by checking history of the package Makefile)
Environment: (arm cortex a9, clearfog pro, openwrt 18.06 branch)

Description:
I just tried to build a openwrt 18.06 image for my router with lspci enabled. I activated pciutils and tried to make it. Unfortunately, i get an error because of a 404. The file is not available.

```
mkdir -p /home/oli/openwrt/dl
SHELL= flock /home/oli/openwrt/tmp/.pci.ids.91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9.flock -c '  	/home/oli/openwrt/scripts/download.pl "/home/oli/openwrt/dl" "pci.ids.91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9" "798528092d1c58eeac99c6505033ec4ce8fe3e19d7e0c41b06790d58753a89b6" "pci.ids" "@GITHUB/pciutils/pciids/91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: raw.githubusercontent.com
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: raw.githubusercontent.com
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: raw.githubusercontent.com
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: raw.githubusercontent.com
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Transient problem: timeout Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (6) Could not resolve host: raw.githubusercontent.com
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.lede-project.org/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://downloads.openwrt.org/sources/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   184  100   184    0     0   2591      0 --:--:-- --:--:-- --:--:--  2555
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:49: /home/oli/openwrt/dl/pci.ids.91cfa8a0c994634ba9a4a8639aa2ac6dff8453b9] Error 2
[... cut ...] '/home/oli/openwrt/feeds/packages/utils/pciutils'
time: package/feeds/packages/pciutils/compile#0.23#0.15#155.73
make[2]: *** [package/Makefile:109: package/feeds/packages/pciutils/compile] Error 2
make[2]: Leaving directory '/home/oli/openwrt'
make[1]: *** [package/Makefile:103: /home/oli/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/oli/openwrt'
make: *** [/home/oli/openwrt/include/toplevel.mk:218: world] Fehler 2
```

neheb, 2019-01-19:
That’s bizzare. Must be some issue with the git repository. Guess I’ll bump and switch the source.

diizzyy, 2019-01-19:
URL works just fine for me

damnms, 2019-01-19:
my fault, sorry. one of the filters in adblock (i use the openwrt version to filter for all in my network) has that raw.githubusercontent.com listed - very annoying.
But i am wondering why all the others are not working?

neheb, 2019-01-19:
Because the files are not there. Probably need to split the ids into a separate package.

diizzyy, 2019-01-27:
@hnyman @thess @damnms 
Please close this

closed 2019-01-27 by hnyman