openwrt/packages #8012: Move libtirpc to base

jow-, 2019-01-22:
@Andy2244 

I moved conntrack-tools out of base into the feed to get rid of one of the last users of librpc. 

Unfortunately there is still busybox remaining as sole (optional) user of librpc. Since base packages shall not depend on feed packages, we have basically two choices now:

Either we move libtirpc into the base, but this depends on whether you'd be fine with the move;
Or we drop the optional RPC support from busybox and mark the features requiring it broken.

It only seems to affect some rather exotic inetd features and the like. What do you think?

Andy2244, 2019-01-22:
I opted to remove the exotic options from busybox in my PR, seemed less annoying overall.

jow-, 2019-01-22:
I lean towards this as well... let me dig out the PR