openwrt/packages #8024: postfix: fatal: statfs /overlay: No such file or directory

aquaforum, 2019-01-23:
Hello !
postfix do not accept e-mails
I checked it on these devices: Paspberry Pi 3 B+ and Nexx WT3020 (8M)
Package version:  ostfix_3.3.2-3_aarch64_cortex-a53.ipk for pi and postfix_3.3.2-3_mips_24kc.ipk for WT3020
When I try to send him a letter he says in a log:
Wed Jan 23 19:19:52 2019 mail.info postfix/smtpd[3142]: connect from unknown[192.168.5.10]
Wed Jan 23 19:19:53 2019 mail.crit postfix/smtpd[3142]: fatal: statfs /overlay: No such file or directory
Wed Jan 23 19:19:54 2019 mail.warn postfix/master[3138]: warning: process /usr/libexec/postfix/smtpd pid 3142 exit status 1
Wed Jan 23 19:19:54 2019 mail.warn postfix/master[3138]: warning: /usr/libexec/postfix/smtpd: bad command startup -- throttling

I looked at the postfix source, in my opinion the error arises in the file <postfix>/src/utils/fsspace.c in the function
void fsspace (const char * path, struct fsspace * sp)
line 76 (or 83, I do not know what vatian is used ) , the call to the statfs() ends with a value of <0
maybe in openwrt broken statfs() call ?
thanks

jow-, 2019-01-24:
> maybe in openwrt broken statfs() call ?

Wouldn't it be more likely that there simply is no `/overlay` on your system? If so, a simple `mkdir /overlay` should work around the problem at least.

aquaforum, 2019-01-24:
he exists, is located on usb stick
# df -hT 
Filesystem           Type            Size      Used Available Use% Mounted on
/dev/root            squashfs        5.3M      5.3M         0 100% /rom
tmpfs                tmpfs          29.5M     96.0K     29.4M   0% /tmp
/dev/sda             f2fs           28.9G    224.8M     28.0G   1% /overlay
overlayfs:/overlay   overlay        28.9G    224.8M     28.0G   1% /
tmpfs                tmpfs         512.0K         0    512.0K   0% /dev
I tried different media, different file systems - the result did not change.

aquaforum, 2019-01-24:
I also checked different versions - 18.06.1 and   development build - the error is in both.

aquaforum, 2019-01-24:
I made him work.
a very rough patch, disables checking the free space in general, but with it, postfix works

diff -ur postfix-3.3.2/src/util/fsspace.c postfix-3.3.2_patched/src/util/fsspace.c
--- postfix-3.3.2/src/util/fsspace.c   2006-06-15 21:07:16.000000000 +0300
+++ postfix-3.3.2_patched/src/util/fsspace.c 2019-01-24 13:53:19.045425129 +0200
@@ -67,36 +67,8 @@
 
 void    fsspace(const char *path, struct fsspace * sp)
 {
-    const char *myname = "fsspace";
-
-#ifdef USE_STATFS
-#ifdef USE_STRUCT_FS_DATA       /* Ultrix */
-    struct fs_data fsbuf;
-
-    if (statfs(path, &fsbuf) < 0)
-  msg_fatal("statfs %s: %m", path);
     sp->block_size = 1024;
-    sp->block_free = fsbuf.fd_bfreen;
-#else
-    struct statfs fsbuf;
-
-    if (statfs(path, &fsbuf) < 0)
-  msg_fatal("statfs %s: %m", path);
-    sp->block_size = fsbuf.f_bsize;
-    sp->block_free = fsbuf.f_bavail;
-#endif
-#endif
-#ifdef USE_STATVFS
-    struct statvfs fsbuf;
-
-    if (statvfs(path, &fsbuf) < 0)
-  msg_fatal("statvfs %s: %m", path);
-    sp->block_size = fsbuf.f_frsize;
-    sp->block_free = fsbuf.f_bavail;
-#endif
-    if (msg_verbose)
-  msg_info("%s: %s: block size %lu, blocks free %lu",
-      myname, path, sp->block_size, sp->block_free);
+    sp->block_free = 1000000;
 }
 
 #ifdef TEST

aquaforum, 2019-01-24:
I found the cause of this error, it is contained in this patch of the kit openwrt
100-fsstat.patch
`-    fsspace(".", &fsbuf);`
`+    fsspace("/overlay", &fsbuf);`
but the postfix processes work in the chroot and do not have access to the /overlay
2 jow-
you are almost right, command
mkdir /usr/var/spool/postfix/overlay/
solved the problem

Shulyaka, 2019-04-10:
Hi @aquaforum,
Thank you for the issue and sorry for noticing it so late.
The reason behind the 100-fsstat.patch is that otherwise pastfix had exactly the same error - fsspace(".", &fsbuf) returning a value < 0. I don't know why (a kernel bug, probably?).
But I never tested it with processes in chroot.
I think I will try to remove the patch and probably find a different workaround if it is still needed.

aquaforum, 2019-04-10:
Thanks, I'm sure to check the changes