openwrt/packages #8025: aircrack-ng: buddy-ng gets compiled without stack canary.

neheb, 2019-01-24:
Maintainer: @jbenden 
Environment: mvebu

Description:
Running checksec.sh indicates that stack canary support is missing. The other binaries in the aircrack-ng suite have it enabled. A quick look through the Makefile indicates different CFLAGS get passed.

jbenden, 2019-01-24:
I have verified that this single binary is missing stack canary support, on a tiny built OpenWrt environment. The stack canary support is present when built for x86_64 with GCC8 on Ubuntu 16.04.

I verified that the Makefiles are not the cause for the missing feature: the flag `-fstack-protector-strong` is indeed passed, and `-DFORTIFY_SOURCE` is auto-detected and added if missing.

I suspect this binary is small enough that the compiler decides to not include support for it. There are a number of default rules used, for the compiler to end up enabling the feature. [ref on StackOverflow](https://stackoverflow.com/questions/24465014/gcc-generate-canary-or-not)

neheb, 2019-01-24:
Got it. Thanks for checking.

closed 2019-01-24 by neheb