openwrt/packages #8036: haproxy service don't start with SSL config, manually with -d (debug) works

Mstaaravin, 2019-01-25:
Maintainer: @gladiac1337 
Environment: (ar71xx , TP-Link 1043-NDv1, 18.06.1)

Description:
I installes haproxy with opkg, http only config works perfectly, but https config (port 443 enabled and uhttpd changed to another) doesn't work.

With -D (daemon mode) don't work
```
haproxy -q -D -f /etc/haproxy.cfg -p /var/run/haproxy.pid
```

With -db (debug mode) works perfectly
```
haproxy -q -db -f /etc/haproxy.cfg -p /var/run/haproxy.pid
```

I check config, and it's valid
```
root@tplhome:/etc# haproxy -c -f haproxy.cfg
Configuration file is valid
```

diizzyy, 2019-01-26:
Please provide logs and error messages

gladiac1337, 2019-01-26:
Yeah, logs and error message please. I'm running haproxy with ssl enabled (since the early 1.8 days) and never had similar problems.

Mstaaravin, 2019-02-12:
I changed my router from TP-Link 1043ND v1 (almost 10 years old) to WRT3200ACM and with the same config I did not have any problems.
Probably because my 1043ND v1 was too old for the job.
I close this issue

closed 2019-02-12 by Mstaaravin