openwrt/packages #8050: openldap: memory leak

MikePetullo, 2019-01-27:
Maintainer: @MikePetullo (also @val-kulkov and @neheb as patch submitters)
Environment: x86_64 master

Description:
There appears to be a memory leak in the slapd provided by the most recent OpenLDAP packages. Here is a line from top after a few hours of running:

```
PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
28958     1 root     S     506m  51%   0% /usr/sbin/slapd -h ldap://localhost/ ldaps:///
```

This amount of memory use is new. The kernel eventually kills the slapd process due to its increasingly large size. This is my configuration file:

```
include         /etc/openldap/schema/core.schema
include         /etc/openldap/schema/cosine.schema
include         /etc/openldap/schema/inetorgperson.schema
include         /etc/openldap/schema/nis.schema
include         /etc/openldap/schema/autofs.schema
include         /etc/openldap/schema/sudoers.schema

allow bind_v2

TLSCACertificateFile /etc/openldap/ca.cert
TLSCertificateFile /etc/openldap/[EXAMPLE.COM].cert
TLSCertificateKeyFile /etc/openldap/[EXAMPLE.COM].key

pidfile         /var/run/slapd.pid
argsfile        /var/run/slapd.args

database        ldif
directory       /etc/openldap/db
suffix          "dc=[EXAMPLE],dc=[COM]"
rootdn          "cn=Manager,dc=[EXAMPLE],dc=[COM]"
rootpw          [PASSWORD]
```

val-kulkov, 2019-01-28:
@MikePetullo : OpenLDAP's high virtual memory use is normal and harmless if mdb (LMDB) is your OpenLDAP backend.

Quoting from Matthew Hardin's article, "[Understanding LMDB database file sizes and memory utilization](https://symas.com/understanding-lmdb-database-file-sizes-and-memory-utilization/)":

> There’s an often-forgotten aspect of virtual memory: the issue is not how much virtual memory is allocated beyond the amount of physical RAM that affects a system’s overall performance, it’s whether or not that memory is actually being referenced. If it’s not being referenced, it simply sits out on disk and doesn’t cause anything else to happen until it’s referenced. In the case of unused LMDB space, the memory is never referenced, so it doesn’t contribute to ‘thrashing’, ‘swapping’,  or other adverse activity.

The actual physical RAM that OpenLDAP normally uses is about 5 - 6.5 MB. This is what I normally observe on my Buffalo WXR-1900DHP (512 MB RAM, 128 MB flash memory, 750 GB extroot drive, 512 MB swap partition):
```
root@openwrt:~# cat /proc/1291/status
Name:	slapd
<snip>
VmPeak:	  113652 kB
VmSize:	  113652 kB
VmLck:	       0 kB
VmPin:	       0 kB
VmHWM:	    6488 kB
VmRSS:	    5004 kB
RssAnon:	    1372 kB
RssFile:	    3632 kB
RssShmem:	       0 kB
VmData:	   98456 kB
VmStk:	     132 kB
VmExe:	    1160 kB
VmLib:	    5512 kB
VmPTE:	      68 kB
VmPMD:	       0 kB
VmSwap:	     988 kB
<snip>
```
The important figures here are **"VmHWM: 6488 kB"** and **"VmRSS: 5004 kB"**. These are the figures I am observing today on my OpenWrt box after 74 days of continuously running slapd:
```
root@openwrt:~# uptime
 23:05:37 up 74 days, 12:19,  load average: 0.40, 0.34, 0.33
```
And this is what I am observing now with "top":
```
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1291     1 root     S     110m  22%   0% /usr/sbin/slapd -h ldap:/// ldaps:/// ldapi:/// -d 0
```
In my experience, OpenLDAP normally reaches its "cruising virtual memory usage" in one or two hours from the start. I have not observed any degraded performance of my OpenWrt box as a result of OpenLDAP's high virtual memory usage. I guess it takes a little getting used to that peculiarity of LMBD.

val-kulkov, 2019-01-28:
Sorry, forgot to add this information:
```
root@openwrt:~# free
             total       used       free     shared    buffers     cached
Mem:        515104     468964      46140       2092      54260     338172
-/+ buffers/cache:      76532     438572
Swap:      4194300       6132    4188168
```

neheb, 2020-01-15:
Closing based on @val-kulkov 's comments.

If this is still an issue, please open a new thread.

closed 2020-01-15 by neheb