openwrt/packages #8056: Stunnel

micky1067, 2019-01-28:
(Stummel) 
This time I find only stunnel in version 5.44-3. Is there a Chance to become the new version 5.49 for arm7 systems. 
In the old version I can not bind localhost to ports. 

I use it for a netgear r7000. 
Thanks

diizzyy, 2019-01-28:
It's not backported.
Current version is 5.49 so you can either build your own firmware or use snapshots.

micky1067, 2019-01-28:
Thanks for your information.
Means it will never give a new ipk for entware? Or can I compile my own one?
I don't know how to do make firmware or snapshot...

diizzyy, 2019-01-28:
@micky1067 
I have no idea what entware tracks (follows) so you need to ask the maintainers for that project.

diizzyy, 2019-01-31:
@thess @hnyman 
Please close

closed 2019-01-31 by hnyman