openwrt/packages #8087: aria2: Control file gets deleted on restart

ahmar20, 2019-01-31:
Maintainer: @kaloz @kuoruan 
Environment: TpLink TD-W8980 running v18.06.1

Description:
The issue occurs with torrent downloads for now, I'm not sure if the same applies to http downloads. When a torrent download completes it starts seeding as it should. But then if I stop and start aria2 from CLI or just restart it while the torrent is seeding the aria2 control file for that download gets deleted on aria2 start and you can not seed the torrent anymore because there's no control file. But if I pause the seeding torrent before stopping aria2 then the control file doesn't get deleted and I can start the file for seeding again when I start aria2. While the torrent file that gets saved while adding metadata from magnet link is still there it doesn't get deleted. 

Is this expected behavior or is a bug?

ahmar20, 2019-02-01:
I think something very fundamental is wrong within aria2. You can have a look at the issue I opened with aria2. 
https://github.com/aria2/aria2/issues/1353

neheb, 2020-02-23:
This is the line of code responsible for removing the control file: https://github.com/aria2/aria2/blob/master/src/RequestGroup.cc#L691

It seems passing dry-run will prevent it from being removed.

ahmar20, 2020-02-23:
Alright, I'll take a look. In the meanwhile I think you can close the issue for now. Thank you.

closed 2020-02-23 by neheb