openwrt/packages #8108: nfs-kernel-server: is absent in repo

darkblaze69, 2019-02-02:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: OpenWrt 18.06.2, MediaTek MT7621, Xiaomi Mi Router 3G

Description: Cannot install nfs server. It says nfs-kernel-server:

opkg install nfs-kernel-server
Unknown package 'nfs-kernel-server'.
Collected errors:
 * opkg_install_cmd: Cannot install package nfs-kernel-server.

diizzyy, 2019-02-03:
https://downloads.openwrt.org/releases/faillogs/mipsel_24kc/packages/nfs-kernel-server/compile.txt
The package tree rarely sees backports so you need to compile your own packages/firmware or wait for the 19.X release.

neradp, 2019-02-15:
Same for me 👎 
OpenWrt 18.06.2, D-Link DIR-645

n1be, 2019-02-18:
The NFS server seemed to work fine for me in 18.06.1.  I also hit this problem after upgrade to 18.06.2 on a Linksys EA-8500.  I will downgrade back to 18.06.1 to avoid this problem.

dterrahe, 2019-02-18:
I was just going to reply that downgrading is indeed not possible anymore either; I have held off on upgrading to 18.06.2 for this reason. But I'm wondering if it is possible to manually install the nfs-kernel-server package from Master (which has version 2.3.3)? Or even if the entware package from http://bin.entware.net/mipselsf-k3.4 would work?

n1be, 2019-02-18:
I could not just download and install nfs-kernel-server_2.3.3-3_arm_cortex-a15_neon-vfpv4.ipk.  There were other additional unmet dependencies.  In addition, it looks like there are four different packages built from the same sources (nfs-kernel-server-utils, nfs-kernel-server, nfs-utils-libs, nfs-utils).  It would be much safer to have identical versions of any of those packages installed.

```
# opkg install /tmp/nfs-kernel-server_2.3.3-3_arm_cortex-a15_neon-vfpv4.ipk
Installing nfs-kernel-server (2.3.3-3) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for nfs-kernel-server:
 * 	libblkid1
 * 	libuuid1
 * 	rpcbind
 * 	nfs-utils-libs
 * 	libkeyutils
 * opkg_install_cmd: Cannot install package nfs-kernel-server.
```

diizzyy, 2019-02-18:
No, you can't mix and mash packages. Also, you can use snaphots but be aware that it's a moving target.

outlawx, 2019-02-22:
Same happened to me today on Kirkwood Zyxel NSA325. The arch ist arm_xscale and the package nfs-kernel-server is missing in 18.06.2.
I have an identical box which was installed in February, 11th and package was available.

To get my recently updated box running a nfs-server, I installed all dependencies listed for https://openwrt.org/packages/pkgdata/nfs-kernel-server (also nfs-server-kernel-utils) and copied /usr/sbin/rpc.statd, rpc.mountd, rpc.nfsd, exportfs from the working box.

I know this is not a solution, but for me (lucky to have two boxes) a quick and dirty way around

onlyjob, 2019-02-22:
On x86 I've installed `nfs-kernel-server 2.3.3-4` (as well as matching version of `nfs-kernel-server-utils` and dependencies) but `nfsd` still does not work because `/usr/sbin/rpc.mountd -p 32780 -F` fails as follows:

    Error relocating /usr/sbin/rpc.mountd: name_to_handle_at: symbol not found

Is there a reproducible way to have a working NFS server on 18.06.02?
Can somebody upload a working `nfs-kernel-server` into 18.06.02 packages repository please?

diizzyy, 2019-02-22:
You can't mix and mash branches, either you compile your own firmware or use snapshots.
Alternatively you can try to convince the core team of OpenWrt to quarterly snapshots of packages or something for releases.

kofec, 2019-02-25:
Maybe it is related (after git pull to latest version - not clean compilation):
```
cache.c: In function 'uuid_by_path':
cache.c:343:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if (*bad == st.f_type)
             ^~
cache.c: In function 'same_path':
cache.c:431:22: error: field 'fh' has incomplete type
   struct file_handle fh;
                      ^~
cache.c:438:6: warning: implicit declaration of function 'name_to_handle_at' [-Wimplicit-function-declaration]
  if (name_to_handle_at(AT_FDCWD, p, &fchild.fh, &mnt_child, 0) != 0) {
      ^~~~~~~~~~~~~~~~~
Makefile:618: recipe for target 'mountd-cache.o' failed
make[6]: *** [mountd-cache.o] Error 1
make[6]: Leaving directory '/mnt/openwrt/OpenwrtNSA310/build_dir/target-arm_xscale_musl_eabi/nfs-utils-2.3.3/utils/mountd'
Makefile:446: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/mnt/openwrt/OpenwrtNSA310/build_dir/target-arm_xscale_musl_eabi/nfs-utils-2.3.3/utils'
Makefile:470: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1

```

kofec, 2019-03-02:
When I want to use rsync I have issue:
```
root@RpiTestowe:~# perl -MFile::RsyncP -e '1'
Can't locate AutoLoader.pm in @INC (you may need to install the AutoLoader module) (@INC contains: /usr/lib/perl5/5.28) at /usr/lib/perl5/5.28/File/RsyncP/Digest.pm line 47.
Compilation failed in require at /usr/lib/perl5/5.28/File/RsyncP.pm line 45.
BEGIN failed--compilation aborted at /usr/lib/perl5/5.28/File/RsyncP.pm line 45.
Compilation failed in require.
BEGIN failed--compilation aborted
```
And I have to install:
```
perlbase-autoloader - 5.28.1-1 - AutoLoader perl module
root@RpiTestowe:~# opkg install perlbase-autoloader
Installing perlbase-autoloader (5.28.1-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/arm_arm1176jzf-s_vfp/packages/perlbase-autoloader_5.28.1-1_arm_arm1176jzf-s_vfp.ipk
Configuring perlbase-autoloader.
```

elonden, 2019-03-08:
Same issue here. Seems the package was not included in the 8.06.2 repo for some reason.

outlawx, 2019-03-09:
@elonden at some point it was included in 18.06.2, see my answer https://github.com/openwrt/packages/issues/8108#issuecomment-466388611

dterrahe, 2019-03-09:
Please be aware that there are differences between architectures. So even though for example https://downloads.openwrt.org/releases/packages-18.06/mips_24kc/packages/nfs-kernel-server_2.3.1-1_mips_24kc.ipk does exist, there is no nfs-kernel-server package in https://downloads.openwrt.org/releases/packages-18.06/mipsel_24kc/packages/

outlawx, 2019-03-09:
@dterrahe yes I'm aware of that. For my Zyxel NSA325 the arch is arm_xscale and and the package for 18.06.2 was available around February 11th and now it ain't. So there may be a common issue...

the-w1nd, 2019-03-09:
I'm in the same boat as you guys.  Just upgraded to 18.06.2 from 18.06.1 and cannot find the "nfs-kernel-server" package anywhere.  Not even 18.06.1 repo has it anymore.  It was there, just like @outlawx described, but it's gone now and I can't even fall back.  Does anybody know of mirrors with old files that can be retrieved for 18.06.2?

diizzyy, 2019-03-10:
This is due to "blindly" making a snapshot of the package repo and little to no effort backporting from master branch. The majority of developers only tracks master which makes the branches "fall behind".

darkblaze69, 2019-03-10:
why this became so low quality? are developers aware of the problem? how many packages else are missing?

outlawx, 2019-03-10:
@the-w1nd in the official download mirror packages don't seem to be specific to the minor versions - just to the major branch.
Here I found an old mirror http://mirror.math.princeton.edu/pub/openwrt/releases/packages-18.06/ be careful though...
And you can also try snapshot-packages from https://downloads.openwrt.org/snapshots/packages/

the-w1nd, 2019-03-10:
> in the official download mirror packages don't seem to be specific to the minor versions - just to the major branch.
That explains it!  Because it would be strange if 18.06.1 branch lost a package a few months after getting released.

> Here I found an old mirror http://mirror.math.princeton.edu/pub/openwrt/releases/packages-18.06/ be careful though...
Thanks a lot!  The "nfs-kernel-server_2.3.1-1" is exactly what I was looking for, since it's compatible with 18.06.1/2 branches.  Just installed it and "opkg" pulled all necessary dependencies just fine.

As a plan "B", I was already thinking about temporarily switching to a snapshot build, or some custom community build for my router.

Much appreciated!

fabrizio2210, 2019-03-16:
HI,
I have the same problem: the nfs-kernel-server package is missing from repo. Any idea when it will be included again?

the-w1nd, 2019-03-16:
> I have the same problem: the nfs-kernel-server package is missing from repo. Any idea when it will be included again?
As @diizzyy pointed out, the package fails compilation on release branches: [https://downloads.openwrt.org/releases/faillogs/mipsel_24kc/packages/nfs-kernel-server/compile.txt](https://downloads.openwrt.org/releases/faillogs/mipsel_24kc/packages/nfs-kernel-server/compile.txt)
So, until someone addresses on 18.* branches, it will stay broken.  19.* branch has it present, but not compatible with 18.* release.

fabrizio2210, 2019-03-18:
From https://downloads.openwrt.org/releases/faillogs/arm_cortex-a9_vfpv3/packages/nfs-kernel-server/compile.txt I can see the row:
```
Package nfs-kernel-server is missing dependencies for the following libraries:
libbsd.so.0
```
An old issue was opened about the missing dependency: https://github.com/openwrt/packages/issues/6890 and closed with PR https://github.com/openwrt/packages/commit/3fc7f7b8cc151ab6490189405cb526b70c9bf6e6

In the branch openwrt-18.06 I can see that the line is missing:
```
 ac_cv_lib_bsd_daemon=no \
```
So I try to insert it in the Makefile.

Someone can test my pull request https://github.com/openwrt/packages/pull/8443 ?

fabrizio2210, 2019-04-07:
HI,
I am trying to contribute to resolve this issue.
So I downloaded from git the project and tried to compile expecting that it failed.
These were my commands:
```
git clone https://git.openwrt.org/openwrt/openwrt.git/
cd openwrt
git chekout v18.06.2
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
<selected nfs-kernel-server>
<selected kmod-fs-nfs*>
make -j 5
```
It seems to work fine, in fact I can find the nfs-kernel-server:
```
fabrizio@fabrizio-Dell:~/Progetti/openwrt ((v18.06.2))$ find . -name "*nfs-kernel*"
...
./bin/packages/arm_cortex-a9_vfpv3/packages/nfs-kernel-server-utils_2.3.1-1_arm_cortex-a9_vfpv3.ipk
./bin/packages/arm_cortex-a9_vfpv3/packages/nfs-kernel-server_2.3.1-1_arm_cortex-a9_vfpv3.ipk
...
```
I expected that it failed, but it compiles. Where am I doing wrong?

stintel, 2019-04-07:
Someone please test #8612

davygrvy, 2019-04-26:
Can't someone just fix the build scripts and add them to the repo without waiting for 19.x?

`...
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/18.06.2/packages/arm_cortex-a9_vfpv3/telephony/Packages.sig
Signature check passed.
root@OpenWrt:~# opkg install nfs-kernel-server
Unknown package 'nfs-kernel-server'.
Collected errors:
 * opkg_install_cmd: Cannot install package nfs-kernel-server.`

stintel, 2019-04-27:
Fixed with c3f6660b8bbf6830d3cdb82fa4fcf362d5cfd0f7

PR #8612 (2019-04-27): [18.06] nfs-kernel-server: backport changes from master to fix build
Maintainer: @tripolar 

Attempt to fix #8108 - someone please test this. I don't have 18.06 toolchain and I'm not using NFS so did not test compile, but these are all clean backports from master.
closed 2019-04-27 by stintel