openwrt/packages #8121: samba4-libs: Missing libcap dependency

neheb, 2019-02-04:
Maintainer: @Andy2244 
Environment: master
Description:

Selecting libcap, building, removing it, and then building samba4 will cause a missing dependency as the build system finds it in staging_dir.

Andy2244, 2019-02-04:
Yeah i suspect that will be true for any optional samba4 lib, samba just pulls in whatever it finds in staging. There is no build option for the libs, so i would have to modify waf to add extra custom switches.

If you have a idea, how to improve this you ideas are welcome.

PS: Btw isn't the real problem that staging, should not have the lib anymore after you deselected/removed it from the build system? I know that's a limitation of the build system, similar to how deselecting a main packages will not deselect the auto added dependent packages.

neheb, 2019-02-04:
Yeah I guess so. No idea if make clean can be modified to also remove InstallDev entries.

closed 2019-02-14 by neheb