openwrt/packages #8128: lighttpd mod_openssl  [mips_24kc]

souravndp, 2019-02-04:
I was trying to enable TLS/SSL in Lighttpd config and got the following error:

    --Warning: please add "mod_openssl" to server.modules list in lighttpd.conf.  A future release of lighttpd 1.4.x *will not* automatically load mod_openssl and lighttpd *will not* use SSL/TLS where your lighttpd.conf contains ssl.* directives.
    --(plugin.c.229) dlopen() failed for: /usr/lib/lighttpd/mod_openssl.so Error loading shared library /usr/lib/lighttpd/mod_openssl.so: No such file or directory
    --(server.c.1141) loading plugins finally failed
There is no `mod_openssl` in the lib folder. Thus adding "mod_openssl" to `server.modules` list in lighttpd.conf also failed.

    (plugin.c.229) dlopen() failed for: /usr/lib/lighttpd/mod_openssl.so Error loading shared library /usr/lib/lighttpd/mod_openssl.so: No such file or directory

    root@GL-AR150:/~# ls /usr/lib/lighttpd/
    mod_access.so      mod_cgi.so         mod_dirlisting.so  mod_expire.so      mod_fastcgi.so     mod_indexfile.so   mod_proxy.so       mod_staticfile.so

**I then tried to install the module but couldn't find in the repository. I am using OpenWRT 18.06, Package architecture: mips_24kc and Lighttpd version: lighttpd/1.4.48 (ssl)**

I am surprised to see ssl is mentioned on the version info. Am I doing something wrong?

Here is my config file:

    $SERVER["socket"] == ":443" {
        ssl.engine                  = "enable"
        ssl.pemfile                 = "/etc/lighttpd/ssl/domain.pem"
        server.document-root = "/www/" 	
    }

neheb, 2020-01-17:
ping @flyn-org 

ping @BKPepe since you guys use lighttpd in TurrisOS

BKPepe, 2020-01-19:
Hello guys,

First, @neheb thanks for mention me! Unfortunately, this issue is almost one year old.

@souravndp Can you please tell us, if the issue still occurs for you in the OpenWrt 18.06.06 or the latest OpenWrt version 19.07 even OpenWrt master if you prefer?

I take a look at it and you shouldn't have this issue anymore. You didn't say exactly which lighttpd version you are using. There is a package version, I see 1.4.48, but there is also a release version, that's not said and it is important in this case. The output from `opkg list-installed | grep lighttpd` would be enough.

My quick investigation lead me to this issue https://github.com/openwrt/packages/issues/5343 (thus this is just a duplicate of that issue, but it just required to update your lighttpd in that time) and it was fixed by these pull requests according to their versions:

OpenWrt master including OpenWrt 18.06 - merged 06/01/2018:
https://github.com/openwrt/packages/pull/5344

OpenWrt 17.01 - merged 20/04/2018:
https://github.com/openwrt/packages/pull/6077

Anyway, if there is no response, I will close this within 7 days.

souravndp, 2020-01-19:
Thank you @BKPepe  for your detailed reply. I was aware of the issue #5343 but I am quite sure that my installation didn't have `mod_openssl` at the time when I opened this issue. Unfortunately, I didn't notice the release version and eventually started using another webserver.

However, I tried to install Lighttpd (v1.4.48-5) today and the good news is that it has `mod_openssl.so` in `/usr/lib/lighttpd/`

I am sorry that I didn't check it earlier and kept the ticket open.

closed 2020-01-19 by souravndp