openwrt/packages #8129: gnutls build failer

stintel, 2019-02-05:
Maintainer: @nmav 
Environment: OpenWrt reboot-9254-g3b40121530 @ brcm2708/bcm2708

Description:
Unable to build gnutls 3.6.5. Reverting 485db8a7707a14e897b27d0b11bcad54e1237445 fixes it.

```
make[7]: Entering directory '/home/build/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5/src'
depbase=`echo psk.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
arm-openwrt-linux-muslgnueabi-gcc -DHAVE_CONFIG_H -I. -I..  -I./gl -I./gl -I./../lib/includes -I./../lib/includes -I./../libdane/includes -I./../extra/includes -I../src/libopts -I./libopts -I../lib/minitasn1 -I/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-7.4.0_musl_eabi/usr/include -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-7.4.0_musl_eabi/include/fortify -I/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-7.4.0_musl_eabi/include  -fno-common -W -Wabi -Waddress -Waggressive-loop-optimizations -Wall -Wattributes -Wbad-function-cast -Wbool-compare -Wbool-operation -Wbuiltin-declaration-mismatch -Wbuiltin-macro-redefined -Wcast-align -Wchar-subscripts -Wchkp -Wclobbered -Wcomment -Wcomments -Wcoverage-mismatch -Wcpp -Wdangling-else -Wdate-time -Wdeprecated -Wdeprecated-declarations -Wdesignated-init -Wdisabled-optimization -Wdiscarded-array-qualifiers -Wdiscarded-qualifiers -Wdiv-by-zero -Wdouble-promotion -Wduplicated-branches -Wduplicated-cond -Wduplicate-decl-specifier -Wempty-body -Wendif-labels -Wenum-compare -Wexpansion-to-defined -Wextra -Wformat-contains-nul -Wformat-extra-args -Wformat-security -Wformat-zero-length -Wframe-address -Wfree-nonheap-object -Whsa -Wignored-attributes -Wignored-qualifiers -Wimplicit -Wimplicit-function-declaration -Wimplicit-int -Wincompatible-pointer-types -Winit-self -Wint-conversion -Wint-in-bool-context -Wint-to-pointer-cast -Winvalid-memory-model -Winvalid-pch -Wjump-misses-init -Wlogical-not-parentheses -Wlogical-op -Wmain -Wmaybe-uninitialized -Wmemset-elt-size -Wmemset-transposed-args -Wmisleading-indentation -Wmissing-braces -Wmissing-declarations -Wmissing-field-initializers -Wmissing-include-dirs -Wmissing-parameter-type -Wmissing-prototypes -Wmultichar -Wnarrowing -Wnested-externs -Wnonnull -Wnonnull-compare -Wnull-dereference -Wodr -Wold-style-declaration -Wold-style-definition -Wopenmp-simd -Woverflow -Woverride-init -Wpacked -Wpacked-bitfield-compat -Wparentheses -Wpointer-arith -Wpointer-compare -Wpointer-sign -Wpointer-to-int-cast -Wpragmas -Wpsabi -Wrestrict -Wreturn-local-addr -Wreturn-type -Wscalar-storage-order -Wsequence-point -Wshadow -Wshift-count-negative -Wshift-count-overflow -Wshift-negative-value -Wsizeof-array-argument -Wsizeof-pointer-memaccess -Wstrict-aliasing -Wstrict-prototypes -Wsuggest-attribute=format -Wsuggest-final-methods -Wsuggest-final-types -Wswitch -Wswitch-bool -Wswitch-unreachable -Wsync-nand -Wtautological-compare -Wtrampolines -Wtrigraphs -Wtype-limits -Wuninitialized -Wunknown-pragmas -Wunused -Wunused-but-set-parameter -Wunused-but-set-variable -Wunused-function -Wunused-label -Wunused-local-typedefs -Wunused-macros -Wunused-parameter -Wunused-result -Wunused-value -Wunused-variable -Wvarargs -Wvariadic-macros -Wvector-operation-performance -Wvolatile-register-var -Wwrite-strings -Walloc-size-larger-than=2147483647 -Warray-bounds=2 -Wformat-overflow=2 -Wformat-truncation=2 -Wimplicit-fallthrough=5 -Wnormalized=nfc -Wshift-overflow=2 -Wstringop-overflow=2 -Wunused-const-variable=2 -Wvla-larger-than=4031 -Wno-missing-field-initializers -Wno-unused-parameter -Wno-format-truncation -Wimplicit-fallthrough=2 -Wabi=11 -fdiagnostics-show-option  -Os -pipe -mcpu=arm1176jzf-s -mfpu=vfp -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5:gnutls-3.6.5 -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -fpic  -MT psk.o -MD -MP -MF $depbase.Tpo -c -o psk.o psk.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from psk.c:41:0:
./psktool-args.h:63:3: error: #error option template version mismatches autoopts/options.h header
 # error option template version mismatches autoopts/options.h header
   ^~~~~
./psktool-args.h:64:3: error: unknown type name 'Choke'
   Choke Me.
   ^~~~~
./psktool-args.h:64:11: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
   Choke Me.
           ^
./psktool-args.h:85:3: warning: data definition has no type or storage class
 } teOptIndex;
   ^~~~~~~~~~
./psktool-args.h:85:3: warning: type defaults to 'int' in declaration of 'teOptIndex' [-Wimplicit-int]
In file included from ./psktool-args.h:49:0,
                 from psk.c:41:
psk.c: In function 'main':
./psktool-args.h:98:50: error: 'INDEX_OPT_PSKFILE' undeclared (first use in this function); did you mean 'VALUE_OPT_PSKFILE'?
 #define         DESC(n) (psktoolOptions.pOptDesc[INDEX_OPT_## n])
                                                  ^
../src/libopts/autoopts/options.h:319:34: note: in definition of macro 'UNUSED_OPT'
 #define UNUSED_OPT(  _od)     (((_od)->fOptState & OPTST_SET_MASK) == 0)
                                  ^~~
./psktool-args.h:100:41: note: in expansion of macro 'DESC'
 #define     HAVE_OPT(n) (! UNUSED_OPT(& DESC(n)))
                                         ^~~~
psk.c:85:7: note: in expansion of macro 'HAVE_OPT'
  if (!HAVE_OPT(PSKFILE)) {
       ^~~~~~~~
./psktool-args.h:98:50: note: each undeclared identifier is reported only once for each function it appears in
 #define         DESC(n) (psktoolOptions.pOptDesc[INDEX_OPT_## n])
                                                  ^
../src/libopts/autoopts/options.h:319:34: note: in definition of macro 'UNUSED_OPT'
 #define UNUSED_OPT(  _od)     (((_od)->fOptState & OPTST_SET_MASK) == 0)
                                  ^~~
./psktool-args.h:100:41: note: in expansion of macro 'DESC'
 #define     HAVE_OPT(n) (! UNUSED_OPT(& DESC(n)))
                                         ^~~~
psk.c:85:7: note: in expansion of macro 'HAVE_OPT'
  if (!HAVE_OPT(PSKFILE)) {
       ^~~~~~~~
./psktool-args.h:98:50: error: 'INDEX_OPT_USERNAME' undeclared (first use in this function); did you mean 'VALUE_OPT_USERNAME'?
 #define         DESC(n) (psktoolOptions.pOptDesc[INDEX_OPT_## n])
                                                  ^
../src/libopts/autoopts/options.h:319:34: note: in definition of macro 'UNUSED_OPT'
 #define UNUSED_OPT(  _od)     (((_od)->fOptState & OPTST_SET_MASK) == 0)
                                  ^~~
./psktool-args.h:100:41: note: in expansion of macro 'DESC'
 #define     HAVE_OPT(n) (! UNUSED_OPT(& DESC(n)))
                                         ^~~~
psk.c:91:7: note: in expansion of macro 'HAVE_OPT'
  if (!HAVE_OPT(USERNAME)) {
       ^~~~~~~~
./psktool-args.h:98:50: error: 'INDEX_OPT_KEYSIZE' undeclared (first use in this function); did you mean 'INDEX_OPT_PSKFILE'?
 #define         DESC(n) (psktoolOptions.pOptDesc[INDEX_OPT_## n])
                                                  ^
../src/libopts/autoopts/options.h:319:34: note: in definition of macro 'UNUSED_OPT'
 #define UNUSED_OPT(  _od)     (((_od)->fOptState & OPTST_SET_MASK) == 0)
                                  ^~~
./psktool-args.h:100:41: note: in expansion of macro 'DESC'
 #define     HAVE_OPT(n) (! UNUSED_OPT(& DESC(n)))
                                         ^~~~
psk.c:108:6: note: in expansion of macro 'HAVE_OPT'
  if (HAVE_OPT(KEYSIZE) && OPT_VALUE_KEYSIZE > MAX_KEY_SIZE) {
      ^~~~~~~~
make[7]: *** [Makefile:2114: psk.o] Error 1
make[7]: Leaving directory '/home/build/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5/src'
make[6]: *** [Makefile:2150: all-recursive] Error 1
make[6]: Leaving directory '/home/build/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5/src'
make[5]: *** [Makefile:1884: all] Error 2
make[5]: Leaving directory '/home/build/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5/src'
make[4]: *** [Makefile:1539: all-recursive] Error 1
make[4]: Leaving directory '/home/build/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5'
make[3]: *** [Makefile:1466: all] Error 2
make[3]: Leaving directory '/home/build/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5'
make[2]: *** [Makefile:250: /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.5/.built] Error 2
make[2]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/libs/gnutls'
```

nmav, 2019-02-13:
Something seems wrong here. Have you customized the configure options?

stintel, 2019-02-13:
```stijn@taz ~/Development/LEDE/source $ grep -i gnutls .config
CONFIG_PACKAGE_libgnutls=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_CRYPTODEV=y
CONFIG_GNUTLS_HEARTBEAT=y
# CONFIG_GNUTLS_OPENPGP is not set
# CONFIG_GNUTLS_SRP is not set
CONFIG_GNUTLS_PSK=y
CONFIG_GNUTLS_ANON=y
# CONFIG_GNUTLS_TPM is not set
# CONFIG_GNUTLS_PKCS11 is not set
# CONFIG_GNUTLS_EXT_LIBTASN1 is not set
# CONFIG_LIBCURL_GNUTLS is not set
# CONFIG_PACKAGE_gnutls-utils is not set```

stintel, 2019-04-07:
Same problem with gnutls 3.6.7.

nmav, 2019-04-07:
I do not know how I can help here. I cannot reproduce it.

stintel, 2019-04-07:
Yes, I migrated my /home drive recently and dumped build_dir, staging_dir, tmp. This is the output of scripts/diffconfig.sh, maybe you can reproduce it with that.
```
CONFIG_TARGET_brcm2708=y
CONFIG_TARGET_brcm2708_bcm2708=y
CONFIG_TARGET_brcm2708_bcm2708_DEVICE_rpi=y
CONFIG_DEVEL=y
CONFIG_BUILD_LOG=y
CONFIG_BUILD_PATENTED=y
CONFIG_DEBUG=y
CONFIG_DROPBEAR_ECC=y
CONFIG_DROPBEAR_ECC_FULL=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_CRYPTODEV=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_IPK_FILES_CHECKSUMS=y
CONFIG_KERNEL_CC_STACKPROTECTOR_NONE=y
# CONFIG_KERNEL_CC_STACKPROTECTOR_REGULAR is not set
# CONFIG_KERNEL_STACKPROTECTOR is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBIIO_LOCAL_BACKEND=y
CONFIG_LIBIIO_NETWORK_BACKEND=y
CONFIG_LIBIIO_XML_BACKEND=y
CONFIG_LLDPD_WITH_CDP=y
CONFIG_LLDPD_WITH_CUSTOM=y
CONFIG_LLDPD_WITH_DOT1=y
CONFIG_LLDPD_WITH_DOT3=y
CONFIG_LLDPD_WITH_EDP=y
CONFIG_LLDPD_WITH_FDP=y
CONFIG_LLDPD_WITH_LLDPMED=y
CONFIG_LLDPD_WITH_PRIVSEP=y
CONFIG_LLDPD_WITH_SONMP=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_ENGINE_BUILTIN=y
CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO=y
CONFIG_OPENSSL_ENGINE_CRYPTO=y
CONFIG_OPENSSL_OPTIMIZE_SPEED=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_EC=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_NPN=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_BRCM80211_DEBUG=y
CONFIG_PACKAGE_avahi-dbus-daemon=y
CONFIG_PACKAGE_bfgminer=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bluez-libs=y
CONFIG_PACKAGE_bluez-utils=y
CONFIG_PACKAGE_ca-bundle=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_dbus=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_domotiqz=y
[... 212 lines cut ...]
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_snmpd=y
CONFIG_PACKAGE_ss=y
CONFIG_PACKAGE_strace=y
CONFIG_PACKAGE_sysfsutils=y
CONFIG_PACKAGE_tc=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_uclibcxx=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_usbutils=y
# CONFIG_PACKAGE_wpad-basic is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_zlib=y
CONFIG_PKG_CC_STACKPROTECTOR_NONE=y
# CONFIG_PKG_FORTIFY_SOURCE_1 is not set
CONFIG_PKG_FORTIFY_SOURCE_2=y
CONFIG_STRACE_LIBDW=y
# CONFIG_STRACE_NONE is not set
CONFIG_TARGET_KERNEL_PARTSIZE=256
CONFIG_TARGET_ROOTFS_PARTSIZE=512
CONFIG_ZLIB_OPTIMIZE_SPEED=y
```

jow-, 2019-04-07:
Did you try reactivating `PKG_FIXUP:=autoreconf gettext-version` ?

stintel, 2019-04-07:
With `PKG_FIXUP:=autoreconf gettext-version` reactivated I get this:
```
Making all in gl
make[5]: Entering directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7/gl'
  GEN      alloca.h
  GEN      arpa/inet.h
  GEN      limits.h
  GEN      netdb.h
  GEN      stdint.h
  GEN      stdio.h
  GEN      stdlib.h
  GEN      string.h
  GEN      strings.h
  GEN      sys/socket.h
  GEN      sys/stat.h
  GEN      sys/time.h
  GEN      sys/types.h
  GEN      sys/uio.h
  GEN      time.h
  GEN      unistd.h
  GEN      wchar.h
make  all-recursive
make[6]: Entering directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7/gl'
Making all in .
make[7]: Entering directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7/gl'
  CC       c-ctype.lo
  CC       c-strcasecmp.lo
  CC       c-strncasecmp.lo
  CC       hash-pjw-bare.lo
  CC       read-file.lo
In file included from /home/build/openwrt/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-7.4.0_musl_eabi/include/endian.h:23:0,
                 from /home/build/openwrt/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-7.4.0_musl_eabi/include/sys/types.h:70,
                 from ./sys/types.h:39,
                 from ./stdio.h:58,
                 from read-file.h:25,
                 from read-file.c:20:
./stdint.h:89:5: error: #if with no expression
 #if
     ^
make[7]: *** [Makefile:1715: read-file.lo] Error 1
make[7]: Leaving directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7/gl'
make[6]: *** [Makefile:1735: all-recursive] Error 1
make[6]: Leaving directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7/gl'
make[5]: *** [Makefile:1595: all] Error 2
make[5]: Leaving directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7/gl'
make[4]: *** [Makefile:1555: all-recursive] Error 1
make[4]: Leaving directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7'
make[3]: *** [Makefile:1481: all] Error 2
make[3]: Leaving directory '/home/build/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7'
make[2]: *** [Makefile:247: /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/gnutls-3.6.7/.built] Error 2
make[2]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#74.05#20.36#91.28
make[1]: *** [package/Makefile:109: package/feeds/packages/gnutls/compile] Error 2
make[1]: Leaving directory '/home/stijn/Development/LEDE/source'
make: *** [/home/stijn/Development/LEDE/source/include/toplevel.mk:218: package/gnutls/compile] Error 2
```

CGretski, 2019-05-21:
This error appears to be caused by an incompatible version of autogen installed on the host.
When make calls "build-aux/missing autogen _<*.def>_" to create the .c and .h files it actually executes the host's /usr/bin/autogen ; resulting in creation of .c/.h files requiring a newer version of libopts than is bundled with gnutls (5.18.6).

Downgrading my host (gentoo) to autogen 5.18.4 allowed gnutls to compile successfully.

nmav, 2019-05-22:
Is it about the host autogen? What if you remove autogen completely from the host, does it help?

CGretski, 2019-05-24:
Removing the host's autogen worked, though isn't an ideal solution.

nmav, 2019-05-25:
Autogen should be triggered only if some modification dates in the system are not right, e.g., you should see in src/ of the unpacked source something like:
```
-r--r--r-- 1 user group 144089 Apr  2 20:02 certtool-args.c.bak
-rw-r--r-- 1 user group  36369 Mar 15 10:10 certtool-args.def
-r--r--r-- 1 user group  14896 Apr  2 20:02 certtool-args.h.bak
```
Could it be something locally which mangles your modification time?

nmav, 2019-05-25:
Hmm, it may be an upstream issue. I'll create a bug report to check.

neheb, 2019-05-29:
It's also failing on ARC:

CGretski, 2019-05-30:
> It's also failing on ARC:
> http://downloads.openwrt.org/snapshots/faillogs/arc_arc700/packages/gnutls/compile.txt
> 
> Different issue looks like.

I get this same issue building 3.6.8 for mips_24kc_gcc-8.3.0_musl
commenting out PKG_FIXUP line allows compilation to succeed ( even with autogen present on host )

nmav, 2019-05-30:
That's a different issue unrelated to autogen. The failure to compile on that is [on gnulib](https://www.gnu.org/software/gnulib/) which seems to have regressed. Let's close this to separate the issues. @CGretski which version of autogen did you have on the host?

nmav, 2019-05-30:
I've removed the fixup nevertheless.

closed 2019-05-30 by nmav

CGretski, 2019-05-30:
@nmav 
host autogen (GNU AutoGen) 5.18.16

neheb, 2019-05-30:
I've also seen this error in the wget package. Some autotools error I guess.

neheb, 2019-06-01:
https://downloads.openwrt.org/snapshots/faillogs/arm_mpcore/packages/gnutls/compile.txt

still failing. for a different reason.

nmav, 2019-06-01:
Ok, it seems to be different failures depending on what devel tools are available in the system. I'll revert the fix completely and allow failures when autogen is present as this is quite a rare tool.

CGretski, 2019-06-02:
How about including autogen in the toolchain and making it a dependency?

nmav, 2019-06-02:
I do not think that this will be a good idea. This will introduce an additional dependency on libopts, and autogen is not widely available. Overall we are trying to get rid of it on upstream:

https://gitlab.com/gnutls/gnutls/milestones/23