openwrt/packages #8134: [unbound] version in current branch vs master

n8v8R, 2019-02-05:
Maintainer: @EricLuehrsen 

Whilst `unbound` 1.8.3 been merged into master there is only 1.8.1 available in the current branch. 

1.8.3 going to be merged into the current branch as well, respectively the [recent 1.9.0 release](https://www.nlnetlabs.nl/downloads/unbound/unbound-1.9.0.tar.gz) from upstream?

EricLuehrsen, 2019-02-05:
Unless there is a hard fault, CVE, or other critical issue, I will not generally polute point releases with upstream or my own packaging changes. Upstream to master timing is a function of engineering judgement. 1.9.0 changed bits of TLS and I havent had time to really review and double check, including consequences of package patches.

n8v8R, 2019-02-05:
Thank you for the instant feedback! 

To my understanding you are saying that changes from 1.8.1 to 1.8.3 do not warrant an update in the current branch?

Since openssl 1.1.x not being available in the current branch there is now a fix in unbound 1.9.0

>  Fix #4206: support openssl 1.0.2 for TLS hostname verification

EricLuehrsen, 2019-02-05:
There is no policy, just qualitative judgement Risk-for-Reward. #8135  will put U1.9.0 in master. Its TLS improvement has the best value for a point release in O18.6. The TLS change carries risk though (another openSSL odd use mem leak?). Its best to wait time for exposure in master with bleeding edge users before retrofitting.

closed 2019-02-05 by n8v8R