openwrt/packages #8137: wireguard: adds endpoint host to routing list. ( without need )

EUA, 2019-02-06:
Maintainer: ???
Environment: ramips, OpenWrt 18.06.1, MediaTek MT7621, Xiaomi Mi Router 3G

Description:
I tried to setup router as WG client for a while. But it doesn't work on my setup.
After research, I realized that licu-wireguard application adds a static route somehow.
**After removal** of non needed record of endpoint from routing table (51.x.x.x in my setup), **Wireguard starts working proper.**

```
root@MainRouter:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.2.1     0.0.0.0         UG    0      0        0 eth0.2
10.0.2.0        *               255.255.255.0   U     0      0        0 wg0
51.x.x.x        MainRouter.lan  255.255.255.255 UGH   0      0        0 br-lan
192.168.1.0     *               255.255.255.0   U     0      0        0 br-lan
192.168.2.0     *               255.255.255.0   U     0      0        0 eth0.2

```
PS: MainRouter.lan is name of the router. (192.168.1.1)

diizzyy, 2019-02-06:
Wrong repo, either https://github.com/openwrt/luci (luci apps) or https://bugs.openwrt.org/ (main wireguard package)

EUA, 2019-02-06:
luci apps said same.  https://github.com/openwrt/luci/issues/2510
Well, I don't know if there are another package tracker available.
Thanks.

closed 2019-02-06 by EUA