openwrt/packages #8139: mwan3 - interface never recovers when using nping

jsalatiel, 2019-02-06:
OpenWrt 18.06.2
mwan3 - 2.7.10-1
luci-app-mwan3 - git-19.020.41695-6f6641d-1

Maintainer: @feckert 
Environment: (x86, model, OpenWrt version)

Description:
If track method is set to nping-tcp, the interface never recovers.

This is my config:
```

config globals 'globals'
	option mmx_mask '0x3F00'
	option rtmon_interval '5'

config interface 'wan'
	option family 'ipv4'
	option timeout '2'
	option interval '10'
	option down '3'
	option up '6'
	option initial_state 'online'
	option size '56'
	option failure_interval '10'
	option recovery_interval '10'
	option flush_conntrack 'never'
	option enabled '1'
	option keep_failure_interval '5'
	option count '3'
	option track_method 'nping-tcp'
	list track_ip 'www.google.com'
	list track_ip 'www.netflix.com'
	option reliability '1'

config interface 'wan2'
	list track_ip '8.8.8.8'
	list track_ip '208.67.222.222'
	list track_ip '8.8.4.4'
	list track_ip '208.67.220.220'
	option family 'ipv4'
	option reliability '4'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	option initial_state 'online'
	option track_method 'ping'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option failure_interval '5'
	option recovery_interval '5'
	option flush_conntrack 'never'
	option enabled '1'

config member 'wan_m1_w1'
	option interface 'wan'
	option metric '1'
	option weight '1'

config member 'wan_m2_w1'
	option interface 'wan'
	option metric '2'
	option weight '1'

config member 'wan2_m1_w1'
	option interface 'wan2'
	option metric '1'
	option weight '1'

config member 'wan2_m2_w1'
	option interface 'wan2'
	option metric '2'
	option weight '1'

config policy 'wan_only'
	list use_member 'wan_m1_w1'

config policy 'wan2_only'
	list use_member 'wan2_m1_w1'

config policy 'balanced'
	list use_member 'wan_m1_w1'
	list use_member 'wan2_m1_w1'

config policy 'wan_wan2'
	list use_member 'wan_m1_w1'
	list use_member 'wan2_m2_w1'

config policy 'wan2_wan'
	list use_member 'wan2_m1_w1'
	list use_member 'wan_m2_w1'

config rule 'https'
	option sticky '1'
	option dest_port '443'
	option proto 'tcp'
	option use_policy 'balanced'

config rule 'default_rule'
	option dest_ip '0.0.0.0/0'
	option use_policy 'balanced'
```

To test, i simply do:
```
# iptables -I OUTPUT -p tcp --dport 80 -j DROP
```
and after 3 attempts i get:

```
Wed Feb  6 10:28:08 2019 user.info mwan3track[15511]: Check (nping-tcp) failed for target "www.google.com" on interface wan (eth2)
Wed Feb  6 10:28:11 2019 user.info mwan3track[15511]: Check (nping-tcp) failed for target "www.netflix.com" on interface wan (eth2)
Wed Feb  6 10:28:24 2019 user.info mwan3track[15511]: Check (nping-tcp) failed for target "www.google.com" on interface wan (eth2)
Wed Feb  6 10:28:27 2019 user.info mwan3track[15511]: Check (nping-tcp) failed for target "www.netflix.com" on interface wan (eth2)
Wed Feb  6 10:28:40 2019 user.info mwan3track[15511]: Check (nping-tcp) failed for target "www.google.com" on interface wan (eth2)
Wed Feb  6 10:28:43 2019 user.info mwan3track[15511]: Check (nping-tcp) failed for target "www.netflix.com" on interface wan (eth2)
Wed Feb  6 10:28:43 2019 user.notice mwan3track[15511]: Interface wan (eth2) is offline
Wed Feb  6 10:28:43 2019 user.notice mwan3[16465]: Execute ifdown event on interface wan (eth2)
Wed Feb  6 10:28:44 2019 user.info mwan3[16465]: connection tracking not flushed on interface wan (eth2) ifdown
```

After that, i remove the iptables rule and the interface never recovers.
If i try to manually use netping , i get the following:

```
#  nping -e eth2 -c 1 www.netflix.com --tcp
Starting Nping 0.7.70 ( https://nmap.org/nping ) at 2019-02-06 10:59 -03
SENT (0.1102s) TCP 189.11.12.13:26422 > 52.35.229.235:80 S ttl=64 id=24928 iplen=40  seq=2170821390 win=1480 
nping_event_handler(): READ-PCAP killed: Resource temporarily unavailable
 
Max rtt: N/A | Min rtt: N/A | Avg rtt: N/A
Raw packets sent: 1 (40B) | Rcvd: 0 (0B) | Lost: 1 (100.00%)
Nping done: 1 IP address pinged in 1.12 seconds

```

If i restart mwan3, the interface is up again and working.

feckert, 2019-02-06:
That's right, I was just able to reconstruct it!
I'll see what the problem is.

jsalatiel, 2019-02-13:
Did you find anything ?

feckert, 2019-02-15:
Sorry this is strange. I do not understand this failure. 
The package leave the system on the right interface.
I could see that the firewall counter are increasing.

I will have to look deeper into this failure. This will take time sorry

jsalatiel, 2019-02-20:
> Sorry this is strange. I do not understand this failure.
> The package leave the system on the right interface.
> I could see that the firewall counter are increasing.
> 
> I will have to look deeper into this failure. This will take time sorry

No problem, thanks for this awesome tool!

haohaolee, 2019-04-09:
I got similar issues. I have a VPN connection which is monitored by mwan3, but I cannot use nping tcp to recover the connection.
It seems `nping -e some_if` ignores the interface and follows the mwan3 rules.(hence will not try VPN route even it is recovered)
However, `curl --interface some_if` works as expected.

feckert, 2019-05-25:
That is really wired!! I have no solution for this issue. If I can not fix it I will remove nping from the supported tracking methods.

haohaolee, 2020-05-10:
@feckert Hi, I am still thinking about this issue.
After some investigation, I now know why curl can send out request by ignoring the rules and routes. The key is `SO_BINDTODEVICE ` [https://linux.die.net/man/7/socket](https://linux.die.net/man/7/socket)

https://github.com/curl/curl/blob/7a86a25f5b73b86455cd977572250de0a6124524/lib/connect.c#L309

This option is only supported by linux, so tools like nping httping don't support this in general.

I am wondering if we can wrap `curl` as an alternative track method to `httping`:
I have tried this:
`curl -I -k --silent --connect-timeout 2 --output /dev/null --interface eth0 https://1.1.1.1`
seems OK for my device

feckert, 2020-05-11:
@haohaolee thanks for digging the code.
I am fine if we could replace the httping with curl.

feckert, 2020-05-29:
@haohaolee do you plan to replace httping with your suggestion in an upcoming pullrequest?

haohaolee, 2020-06-01:
@feckert Sorry, I have ignored some notifications for this issue. Allow me give it a try!

PR #13169 (2020-10-21): mwan3: mwan3track via default routing table and use procd from mwan3track & mwan3rtmon
Maintainer:  @feckert 
Compile tested: 19.07.3-ipq806x and snapshot-7673df79 MT76x8 
Run tested: 19.07.3 and snapshot - 7673df79

Description:
This P/R has a few fixes for the 19.07.3

1) Use the default routing tables for all mwan3track outgoing requests. This fixes some issues on the 5.4.x kernel where packets are not routed properly unless they go through a routing table that has the gateway. This also allows us to remove some hacks that were in place for ipv6 mwan3track
2) Comply with source based routin [... cut]