openwrt/packages #8140: L2TP doesn't reconnect

Jeroenleender, 2019-02-06:
Maintainer: @yousong 
Environment: (RAMIPS, HLKRM04, OpenWrt 18.06.2)

Description:

L2TP VPN won't autoreconnect, if WAN cable gets unplugged and gets plugged back in after you receive the message: 
"daemon.debug xl2tpd[1062]: Unable to deliver closing message for tunnel 20224. Destroying anyway. in system log"
in System Log

Everything is build from the default feeds and updated yesterday.
Earlier 18.06.1 had the same issue.

Extra question:
Would it be possible to have a default l2tp VPN connection added on firstboot?
Would allow for easy configuration when replacing devices is needed by non technician.

yousong, 2019-02-07:
hi, @Jeroenleender , I currently do not have the necessary environment to reproduce the issue.  Please share with us more content from system log to make more clear the event sequence and context.  Thank you

Jeroenleender, 2019-02-07:
Thu Feb  7 11:30:52 2019 kern.notice kernel: [    0.000000] Linux version 4.14.96 (jeroen@rm04) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r9239-f665fb058f)) #0 Wed Jan 30 14:17:23 2019
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] SoC Type: Ralink RT5350 id:1 rev:3
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] CPU0 revision is: 0001964c (MIPS 24KEc)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] MIPS: machine is HILINK HLK-RM04
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Determined physical RAM map:
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000]  memory: 02000000 @ 00000000 (usable)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Thu Feb  7 11:30:52 2019 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Thu Feb  7 11:30:52 2019 kern.warn kernel: [    0.000000] Primary data cache 16kB, 4-way, VIPT, no aliases, linesize 32 bytes
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Zone ranges:
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000001ffffff]
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Movable zone start for each node
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Early memory node ranges
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000001ffffff]
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000001ffffff]
Thu Feb  7 11:30:52 2019 kern.debug kernel: [    0.000000] On node 0 totalpages: 8192
Thu Feb  7 11:30:52 2019 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80394b70, node_mem_map 81000040
Thu Feb  7 11:30:52 2019 kern.debug kernel: [    0.000000]   Normal zone: 64 pages used for memmap
Thu Feb  7 11:30:52 2019 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Feb  7 11:30:52 2019 kern.debug kernel: [    0.000000]   Normal zone: 8192 pages, LIFO batch:0
Thu Feb  7 11:30:52 2019 kern.notice kernel: [    0.000000] random: get_random_bytes called from 0x8039872c with crng_init=0
Thu Feb  7 11:30:52 2019 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Thu Feb  7 11:30:52 2019 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8128
Thu Feb  7 11:30:52 2019 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS1,57600 rootfstype=squashfs,jffs2
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Writing ErrCtl register=0006bbf0
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Readback ErrCtl register=0006bbf0
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] Memory: 28356K/32768K available (3151K kernel code, 160K rwdata, 360K rodata, 160K init, 195K bss, 4412K reserved, 0K cma-reserved)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] NR_IRQS: 256
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] CPU Clock: 360MHz
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] systick: running - mult: 214748, shift: 32
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 10618113593 ns
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.000016] sched_clock: 32 bits at 180MHz, resolution 5ns, wraps every 11930464253ns
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.015711] Calibrating delay loop... 239.61 BogoMIPS (lpj=1198080)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.098024] pid_max: default: 32768 minimum: 301
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.107622] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.120736] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.143675] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.163234] futex hash table entries: 256 (order: -1, 3072 bytes)
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.175495] pinctrl core: initialized pinctrl subsystem
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.188676] NET: Registered protocol family 16
Thu Feb  7 11:30:52 2019 kern.warn kernel: [    0.206935] Can't analyze schedule() prologue at 8030fbe4
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.260203] rt2880_gpio 10000600.gpio: registering 22 gpios
Thu Feb  7 11:30:52 2019 kern.info kernel: [    0.271349] rt2880_gpio 10000600.gpio: registering 22 irq handlers
[... cut ...]
Thu Feb  7 11:36:39 2019 daemon.notice xl2tpd[902]: Maximum retries exceeded for tunnel 63928.  Closing.
Thu Feb  7 11:36:39 2019 daemon.info xl2tpd[902]: Connection 0 closed to xxx.xxx.xxx.xxx, port 1701 (Timeout)
Thu Feb  7 11:37:10 2019 daemon.debug xl2tpd[902]: Unable to deliver closing message for tunnel 63928. Destroying anyway.
Thu Feb  7 11:39:07 2019 kern.info kernel: [  457.250679] rt3050-esw 10110000.esw: link changed 0x18
Thu Feb  7 11:43:36 2019 daemon.err uhttpd[875]: luci: accepted login on / for root from 192.168.1.100

Jeroenleender, 2019-02-07:
above is the complete "system log" including unplugging the WAN cable and putting it back in a couple of minutes later.
After leaving it plugged in for some minutes i relogged back in using Luci so it would get added to the system log so you see nothing else is being done.

yousong, 2019-02-12:
@Jeroenleender 

Can you please try adding `option checkup_interval 10` to that VPN interface `/etc/config/network`.  It tells netifd to wait that many seconds to check if the interface is up since last setup attempt and retry if it's not the case.  The option is documented at https://github.com/openwrt/packages/tree/master/net/xl2tpd#uci-options .  It should fix the issue.

The other thing worth noting is that it seems wan down and up event did not occur along with the unplug/re-plug event (link status changed from 0x18 to 0x8, then again to 0x18).  It seems that link status of switch port do not current translate to interface up/down status of switch cpu port.

Jeroenleender, 2019-02-15:
@yousong 
Sorry for my late reply, i will check this option this weekend to see if it fixes the issue.

Is there anything we can do (workaround) for the up/down status of switch cpu port?

yousong, 2019-02-15:
> Is there anything we can do (workaround) for the up/down status of switch cpu port?

You can write script with swconfig command to query link status of the wan port and bring up/down eth0.2 interface.  But it's just too hacky to worth going further.

I wonder maybe switchdev drivers can help in this situation, but that requires driver and userspace rewrite.

yousong, 2019-02-23:
@Jeroenleender friendly remainder

yousong, 2019-03-04:
Close for lack of activity.  Feel free to report back if you have updates.  Thank you.

closed 2019-03-04 by yousong

iDarkEmpire, 2019-10-01:
The problem, checked already on all openwrt that on stable that on snapshots:
I pull out a router from a network, I connect, there is no network on L2TP
I do restart through the interface in a router, a network on L2TP is
I reboot L2TP connection in the router interface, the network also is

In what there can be a problem that the router does not connect L2TP after disconnection from a network?

yousong, 2019-10-01:
> @Jeroenleender
> 
> Can you please try adding `option checkup_interval 10` to that VPN interface in `/etc/config/network`. It tells netifd to wait that many seconds to check if the interface is up since last setup attempt and retry if it's not the case. The option is documented at https://github.com/openwrt/packages/tree/master/net/xl2tpd#uci-options . It should fix the issue.
> 
> The other thing worth noting is that it seems wan down and up event did not occur along with the unplug/re-plug action (link status changed from 0x18 to 0x8, then again to 0x18). It seems that link status of switch port do not current translate to interface up/down status of switch cpu port.

One thing worth adding is that MAYBE after unplugging the l2tp tunnel session will detect timeout and trigger teardown and reconnect.