openwrt/packages #8147: [meta] Mailing list or forum for project / meta discussions?

jefferyto, 2019-02-07:
Hi, I'm Jeff To. You may remember me from such hits as *lurking in the background*, *occasionally updating a package*, and *being pedantic about tabs and curly brackets*.

I've noticed that sometimes there are project or meta discussions happening as issues:

* #6584 Growing lack of package maintainers
* #7712 [RFC] Compile but not run testing 'compilation fixes'/updates/uclibc additions etc.

I have no problem with these discussions being filed as issues, but it's difficult to pick them out from all the other package-related issues. I don't regularly browse the entire issues list (I have subscribed to all conversations but I filter for keywords for packages I maintain). I'm only a part-time contributor but I'd like to keep up with what's happening and maybe chime in now and then.

I wanted to open this discussion (as an issue, of course) to see if others have any thoughts on this.

* Would a separate mailing list or forum be a better place to have these discussions?
* If folks like having these as GitHub issues, can we agree on a label or subject line tag (that I can use to surface these issues in my inbox)?

Thanks! :pray::wink:

cshoredaniel, 2019-02-10:
I think the -devel mailing list is a good place, as long as everyone knows to go there for such discussions.  Either that use one of the GitHub-by tools like a Wiki 'Talk' page or maybe a 'Project' board, for meta discussions (since packages is on GitHub anyway it would make sense).

feckert, 2019-03-28:
I think such discussions should be made in OpenWrt Forum
https://forum.openwrt.org/

This is a issue tracker for packages in this repository.

neheb, 2020-01-12:
There's a discussion section in the package-maitainers team (which should probably be cleaned up at some point. Too many inactive people). It's currently empty.

Anyway, I've no problem with meta Issues. The main problem is that most Issues here are unlabeled.

jefferyto, 2020-01-27:
It's been almost a year and this issue has attracted 3 comments, so I think we can say most package maintainers are not browsing the issues list for meta discussions :joy:

Some thoughts on the suggestions so far:

* openwrt-devel mailing list - Most (all?) of the threads there are patches and discussion for the main repo. (I wasn't subscribed for a long time, until I decided I really needed to stay informed and configured my Gmail filters to auto-archive all patches.) To me it doesn't feel like a place to discuss (non-core) packages.

* OpenWrt forum - To me this feels like a place where the wider OpenWrt community meets to discuss and help each other. Any packages-specific discussions may pick up comments from random end users. (It's not that I don't want their input, but end user suggestions are not always... helpful.) Maybe a separate section of the forum just for packages discussion could work.

* Package-maintainers team - I'm not on it :laughing: I think most casual maintainers won't be a member of the team. Unless the discussion section can be opened to non-members, many casual maintainers (who may be still be interested enough to participate) would be left out.

* Other GitHub tools - Maybe... I'm not familiar enough with GitHub's other collaboration tools to say one way or the other.

At this point I don't have any other useful suggestions... There are still meta-/project-related matters I would like to ask ([freezing the openwrt-19.07 branch](https://github.com/openwrt/packages/issues/10510) was one; more recently, the support status of the openwrt-18.06 branch), but it doesn't appear that GitHub issues are a good fit for these.

Probably best I close this issue as well.

closed 2020-01-27 by jefferyto