openwrt/packages #8151: python3-cryptographic: Compile OpenWRT Image with pip3 module installed

petrukngantuk, 2019-02-08:
Maintainer: @commodo 
Environment: Netgear R7800, OpenWRT 18.06, arm_cortex-a15_neon-vfpv4 

Description:
Hello Everyone,
I have Netgear R7800, I install it with openwrt with some customization,

now I want to install 'pip3 install mitmproxy' it's failed with error : 

```
Collecting mitmproxy
  Downloading https://files.pythonhosted.org/packages/e5/6c/fe3f4b4cbef7f703b1199d5443963c4c56c2182223375bdaf0930e1ad052/mitmproxy-4.0.4-py3-none-any.whl (1.3MB)
    100% |████████████████████████████████| 1.3MB 931kB/s
Collecting urwid<2.1,>=2.0.1 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/c7/90/415728875c230fafd13d118512bde3184d810d7bf798a631abc05fac09d0/urwid-2.0.1.tar.gz (604kB)
    100% |████████████████████████████████| 614kB 539kB/s
Collecting blinker<1.5,>=1.4 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/1b/51/e2a9f3b757eb802f61dc1f2b09c8c99f6eb01cf06416c0671253536517b6/blinker-1.4.tar.gz (111kB)
    100% |████████████████████████████████| 112kB 2.3MB/s
Collecting wsproto<0.12.0,>=0.11.0 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/1f/bf/a4136bd7950c49f18a7d643f962e307cbed93ca115e0634656d88e02c333/wsproto-0.11.0-py2.py3-none-any.whl
Collecting click<7,>=6.2 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/34/c1/8806f99713ddb993c5366c362b2f908f18269f8d792aff1abfd700775a77/click-6.7-py2.py3-none-any.whl (71kB)
    100% |████████████████████████████████| 71kB 1.3MB/s
Collecting pyOpenSSL<18.1,>=17.5 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/96/af/9d29e6bd40823061aea2e0574ccb2fcf72bfd6130ce53d32773ec375458c/pyOpenSSL-18.0.0-py2.py3-none-any.whl (53kB)
    100% |████████████████████████████████| 61kB 1.2MB/s
Collecting passlib<1.8,>=1.6.5 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/ee/a7/d6d238d927df355d4e4e000670342ca4705a72f0bf694027cf67d9bcf5af/passlib-1.7.1-py2.py3-none-any.whl (498kB)
    100% |████████████████████████████████| 501kB 2.9MB/s
Collecting cryptography<2.4,>=2.1.4 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/22/21/233e38f74188db94e8451ef6385754a98f3cad9b59bedf3a8e8b14988be4/cryptography-2.3.1.tar.gz (449kB)
    100% |████████████████████████████████| 450kB 408kB/s
Collecting ldap3<2.6,>=2.5 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/54/93/55d4de43abf3af1b60befa648811538b49701237e5cbda456608384b25ac/ldap3-2.5.2-py2.py3-none-any.whl (393kB)
    100% |████████████████████████████████| 399kB 409kB/s
Collecting ruamel.yaml<0.16,>=0.13.2 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/95/be/1a63cdb6053aeed8a6de1bcbca02c54e7746383fbe0facfb798e69a66d67/ruamel.yaml-0.15.87.tar.gz (316kB)
    100% |████████████████████████████████| 317kB 1.8MB/s
Collecting brotlipy<0.8,>=0.7.0 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/d9/91/bc79b88590e4f662bd40a55a2b6beb0f15da4726732efec5aa5a3763d856/brotlipy-0.7.0.tar.gz (413kB)
    100% |████████████████████████████████| 419kB 463kB/s
Collecting kaitaistruct<0.9,>=0.7 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/23/71/8577ca06e81c1dc0ba03a39ae32e315175ba2d9df51befa3a45f47950056/kaitaistruct-0.8.tar.gz
Requirement already satisfied: certifi>=2015.11.20.1 in /usr/lib/python3.7/site-packages (from mitmproxy) (2018.11.29)
Requirement already satisfied: hyperframe<6,>=5.1.0 in /usr/lib/python3.7/site-packages (from mitmproxy) (5.2.0)
Requirement already satisfied: h2<4,>=3.0.1 in /usr/lib/python3.7/site-packages (from mitmproxy) (3.1.0)
Requirement already satisfied: pyperclip<1.7,>=1.6.0 in /usr/lib/python3.7/site-packages (from mitmproxy) (1.6.5)
Collecting sortedcontainers<2.1,>=1.5.4 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/be/e3/a065de5fdd5849450a8a16a52a96c8db5f498f245e7eda06cc6725d04b80/sortedcontainers-2.0.5-py2.py3-none-any.whl
Requirement already satisfied: pyparsing<2.3,>=2.1.3 in /usr/lib/python3.7/site-packages (from mitmproxy) (2.2.2)
Requirement already satisfied: pyasn1<0.5,>=0.3.1 in /usr/lib/python3.7/site-packages (from mitmproxy) (0.4.4)
Collecting tornado<5.2,>=4.3 (from mitmproxy)
  Downloading https://files.pythonhosted.org/packages/e6/78/6e7b5af12c12bdf38ca9bfe863fcaf53dc10430a312d0324e76c1e5ca426/tornado-5.1.1.tar.gz (516kB)
    100% |████████████████████████████████| 522kB 1.7MB/s
Collecting h11~=0.7.0 (from wsproto<0.12.0,>=0.11.0->mitmproxy)
  Downloading https://files.pythonhosted.org/packages/97/57/c0734cec5a9344d1feab295046011edbc6ba8e877ce1b822373257acdb61/h11-0.7.0-py2.py3-none-any.whl (53kB)
    100% |████████████████████████████████| 61kB 3.3MB/s
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3.7/site-packages (from pyOpenSSL<18.1,>=17.5->mitmproxy) (1.12.0)
Requirement already satisfied: idna>=2.1 in /usr/lib/python3.7/site-packages (from cryptography<2.4,>=2.1.4->mitmproxy) (2.8)
Requirement already satisfied: asn1crypto>=0.21.0 in /usr/lib/python3.7/site-packages (from cryptography<2.4,>=2.1.4->mitmproxy) (0.24.0)
Requirement already satisfied: cffi!=1.11.3,>=1.7 in /usr/lib/python3.7/site-packages (from cryptography<2.4,>=2.1.4->mitmproxy) (1.11.5)
Requirement already satisfied: hpack<4,>=2.3 in /usr/lib/python3.7/site-packages (from h2<4,>=3.0.1->mitmproxy) (3.0.0)
Requirement already satisfied: pycparser in /usr/lib/python3.7/site-packages (from cffi!=1.11.3,>=1.7->cryptography<2.4,>=2.1.4->mitmproxy) (2.19)
Installing collected packages: urwid, blinker, h11, wsproto, click, cryptography, pyOpenSSL, passlib, ldap3, ruamel.yaml, brotlipy, kaitaistruct, sortedcontainers, tornado, mitmproxy
  Running setup.py install for urwid ... done
  Running setup.py install for blinker ... done
  Running setup.py install for cryptography ... error
Command "/usr/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-4xfgubc5/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-tbbrllmh/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-4xfgubc5/cryptography/
```

before, I had installed 'python3-pip python3-setuptools python3-cryptography', and I think the error is in python3-cryptographic, and after some consideration,

I want to compile OpenWRT Image with pip3 module mitmproxy installed.
I read this stackoverflow [https://stackoverflow.com/questions/50218544/how-to-include-python-pip3-modules-in-openwrt-image](https://stackoverflow.com/questions/50218544/how-to-include-python-pip3-modules-in-openwrt-image)

but I confuse about 'files/lib/python3.7/site-packages' because mitmproxy dependencies is not just one,

Anyone can help me with it? Compile OpenWRT Image with Pip3 Module Installed,

commodo, 2019-02-08:
are you trying the `-U` (i.e. upgrade option) when installing via pip3 ?
that will typically try to install `cryptography` from pypi, which may not work

the version of `python3-cryptography` in 18.06 is a bit old, but they won't be updated anytime soon;

petrukngantuk, 2019-02-08:
> are you trying the `-U` (i.e. upgrade option) when installing via pip3 ?
> that will typically try to install `cryptography` from pypi, which may not work
> 
> the version of `python3-cryptography` in 18.06 is a bit old, but they won't be updated anytime soon;

how sad...
so no otherway?,

commodo, 2019-02-08:
well, you could update packages [python3-cryptography] from master
that may work;

other than that, there are several options:
* use the latest OpenWrt from master ;that is typically up-to-date
* build locally a newer `python3-cryptography` and install it on your router

i admit the situations are a bit limiting, but OpenWrt is a system that typically runs on slow HW or HW with limited resources; so the situation is generally like this

a new release should come along of OpenWrt in some time [if it hasn't already]

jefferyto, 2019-02-11:
From the install log, it looks like mitmproxy wants to install `cryptography<2.4,>=2.1.4` (which is what [setup.py](https://github.com/mitmproxy/mitmproxy/blob/v4.0.4/setup.py#L67) is looking for). The current version of `python3-cryptography` in 18.06 is [2.1.4](https://github.com/openwrt/packages/blob/openwrt-18.06/lang/python/python-cryptography/Makefile#L11), so _technically_ it should satisfy the requirement.

My guess is, our `python3-cryptography` package is missing files that makes pip3 think it is not installed or doesn't satisfy the requirement, but I haven't debugged this yet to say for certain.

jefferyto, 2019-02-11:
Hmm... so the current version of python3 in 18.06 is [3.6.5](https://github.com/openwrt/packages/blob/openwrt-18.06/lang/python/python3-version.mk), and so `python3-cryptography` installs into `/usr/lib/python3.6`.

But according to the install log above, pip3 is checking `/usr/lib/python3.7`. In fact, the last line of the install log (the command that failed), starts with `/usr/bin/python3.7`.

@petrukngantuk Did you manually compile and install python 3.7?

petrukngantuk, 2019-02-11:
> Hmm... so the current version of python3 in 18.06 is [3.6.5](https://github.com/openwrt/packages/blob/openwrt-18.06/lang/python/python3-version.mk), and so `python3-cryptography` installs into [...]

nope, I use openwrt master branch,

jefferyto, 2019-02-11:
Environment: Netgear R7800, OpenWRT 18.06, arm_cortex-a15_neon-vfpv4

Master branch is not 18.06.

Did you compile your own image or did you use a snapshot?

jefferyto, 2019-02-11:
Wait, the version of `python3-cryptography` in master is 2.4.2, which is **too new** for mitmproxy (v4.0.4). You can see from the install log that pip3 downloaded 2.3.1, then tried to install it.

mitmproxy [bumped up its dependencies](https://github.com/mitmproxy/mitmproxy/commit/2fb2b48a060f53c10958d493415a802c5daf5830#diff-2eeaed663bd0d25b7e608891384b7298) to `cryptography>=2.1.4,<2.5` about 2 months ago. I suggest waiting for them to release a new version and trying that, using their latest version from git, or downloading the [OpenWrt SDK](https://openwrt.org/docs/guide-developer/using_the_sdk) and compiling your own `python3-cryptography` for 2.3.1.

BTW cryptography 2.5 was released a few weeks ago, and `python3-cryptography` will probably be updated to it soon. I would expect mitmproxy to update their dependencies as well, but you may want to keep an eye on the whole situation.

petrukngantuk, 2019-02-11:
> ```
> Environment: Netgear R7800, OpenWRT 18.06, arm_cortex-a15_neon-vfpv4
> ```
> Master branch is not 18.06.
> 
> Did you compile your own image or did you use a snapshot?

sorry, for wrong info, I fixed it,

closed 2020-02-22 by BKPepe