openwrt/packages #8153: shadowsocks-libev not running openwrt18.06.2 on Linksys 1900ACS

ebenfarnworth, 2019-02-08:
Thanks, I see much work has been done on ss-libev since I last tried to get it running.

I have installed:
shadowsocks-libev-config
shadowsocks-libev-ss-local
shadowsocks-libev-ss-redir
shadowsocks-libev-ss-rules
shadowsocks-libev-ss-tunnel
luci-app-shadowsocks-libev

I can't get the service to start:
[image]

I followed the instructions [here](https://medium.com/@cnnbysy/openwrt-18-06-1-配置科学上网-30e231958c38).

Here is a copy of the config file (where "my server.." = the settings I entered to link to my ss server):
```
config ss_local                                                                                          
        option server 'sss0'                                                                             
        option local_port '1080'                                                                         
        option timeout '30'                                                                              
        option mode 'tcp_and_udp'                                                                        
        option local_address '0.0.0.0'                                                                   
                                                                                                         
config ss_tunnel                                                                                         
        option disabled '1'                                                                              
        option server 'sss0'                                                                             
        option local_address '0.0.0.0'                                                                   
        option local_port '1090'                                                                         
        option tunnel_address 'example.com:80'                                                           
        option mode 'tcp_and_udp'                                                                        
        option timeout '60'                                                                              
                                                                                                         
config ss_redir 'hi'                                                                                     
        option disabled '1'                                                                              
        option server 'sss0'                                                                             
        option local_address '0.0.0.0'                                                                   
        option local_port '1100'                                                                         
        option mode 'tcp_and_udp'                                                                        
        option timeout '60'                                                                              
        option fast_open '1'                                                                             
        option verbose '1'                                                                               
        option reuse_port '1'                                                                            
                                                                                                         
config ss_redir 'hj'                                                                                     
        option disabled '1'                                                                              
        option server 'sss0'                                                                             
        option local_address '0.0.0.0'                                                                   
        option local_port '1100'                                                                         
        option mode 'tcp_and_udp'                                                                        
        option timeout '60'                                                                              
        option fast_open '1'                                                                             
        option verbose '1'                                                                               
        option reuse_port '1'        

config ss_rules 'ss_rules'                                                                               
        option disabled '1'                                                                              
        option redir_tcp 'hi'                                                                            
        option redir_udp 'hi'                                                                            
        option src_default 'checkdst'                                                                    
        option dst_default 'bypass'                                                                      
        option local_default 'checkdst'                                                                  
        list src_ips_forward '192.168.1.4'                                                               
        list dst_ips_forward '8.8.8.8'                                                                   
                                                                                                         
config server 'sss0'                                                                                     
        option server 'my server address'                                                                
        option server_port 'my server pot'                                                                        
        option method 'chacha20-ietf-poly1305'                                                           
        option password 'My server password'                                                                   
                                                                                                         
config ss_server                                                                                         
        option disabled '1'                                                                              
        option server_port '9001'                                                                        
        option password '********'                                                                       
        option method 'aes-256-cfb'                                                                      
        option bind_address '192.168.7.72'    
```

yousong, 2019-02-16:
> `root@OpenWrt:~# ss-local` = Bus error

Very likely this was caused by alignment issues in the shadowsocks-libev code, or bad binaries caused by wrong compiler options.

I will see if I can reproduce the issue on an armvirt virtual machine.  Before that you can confirm that procd was really already told to run the ss-local instance with the following command.  Other debug methods can be found at https://github.com/openwrt/packages/tree/master/net/shadowsocks-libev#notes-and-faq

```
ubus call service list '{"name": "shadowsocks-libev", "verbose": true}'
```

ebenfarnworth, 2019-02-17:
Thanks, attached is the output after running the above command:

[ss-test-02-17.txt](https://github.com/openwrt/packages/files/2872652/ss-test-02-17.txt)

I have also tried another older build of shadowsocks for Open-WRT. I could not get that to start either.
The hardware I'm using is an Linksys WRT 1900 ACS, using an: arm_cortex-a9_vfpv3

I had not found your FAQ before, I'm going to read through it now and will edit this post if I have any more information to add. Thanks again.

yousong, 2019-02-17:
The attached file seems to be snippet from apache config.

Anyway, I cannot reproduce the issue with QEMU armvirt machine.  You may have to try gdb on ss-local to find out which part fails (dependent libraries can also cause this bus error)

ebenfarnworth, 2019-02-17:
Thanks for your help.
Sorry about that! here is the correct version of the file:
[ss-test-02-17.txt](https://github.com/openwrt/packages/files/2872972/ss-test-02-17.txt)

yousong, 2019-02-17:
> `You may have to try gdb on ss-loca o find out which part fails`
> How do I do that? I have not spent much time with Open-WRT...

Maybe install gdb first with `opkg install gdb` then try something like the following to dump info on the failure site.  To find the cause is the first step, I reckon to really fix it requires some trial and error efforts on real hardware.

```
➜  ~ gdb a
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-114.el7
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/yunion/a...(no debugging symbols found)...done.
(gdb) r
Starting program: /home/yunion/a

Program received signal SIGBUS, Bus error.
0x00007ffff7a444d7 in kill () at ../sysdeps/unix/syscall-template.S:81
81	T_PSEUDO (SYSCALL_SYMBOL, SYSCALL_NAME, SYSCALL_NARGS)
(gdb) bt
#0  0x00007ffff7a444d7 in kill () at ../sysdeps/unix/syscall-template.S:81
#1  0x0000000000400530 in main ()
(gdb) q
A debugging session is active.

	Inferior 1 [process 15611] will be killed.

Quit anyway? (y or n) y

```

> What hardware would you recommend for a cheep setup to run without experimenting debugging with gdb? I will give it a go but I'm not confident I can make it work!...

Sorry, I do not know much about popular hardware for OpenWrt these days.  I am currently using HiWiFi HC6361 that has been with me for about 6 years...

ebenfarnworth, 2019-02-19:
I have been trying to install gdb but the install keeps failing. I'm thinking there might not be enough space on my device (25mb free) or that some of the download servers are blocked at my location.
I have to put this on hold for now as work is getting busy. Anyway I will have ago again next week. Maybe I will download it to my computer first... Thanks for the info above.

yousong, 2019-03-17:
@ebenfarnworth  can you maybe give the latest binaries from master branch a try?  Just extract the binary to /tmp dir and see if it runs without the old "bus error"

http://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/

ebenfarnworth, 2019-03-17:
@yousong I can try, this is going to sound like a stupid question! But which binary's should I extract to the /tmp folder? In the package list I can see many ss related pages. Thanks

yousong, 2019-03-17:
> `root@OpenWrt:~# ss-local` = Bus error

it's shadowsocks-libev-ss-local

ebenfarnworth, 2019-03-17:
I tried to install the package after downloading it to the /tmp folder but I can't get it to install:

```
root@OpenWrt:/tmp# ls
TZ                                                          overlay
board.json                                                  resolv.conf
dhcp.leases                                                 resolv.conf.auto
dnsmasq.d                                                   resolv.conf.ppp
download?path=%2F                                           run
etc                                                         shadowsocks-libev-ss-local_3.2.5-1_arm_cortex-a9_vfpv3.ipk
hosts                                                       shm
lib                                                         state
lock                                                        syscfg
log                                                         sysinfo
luci-indexcache                                             tmp
luci-modulecache                                            usr
opkg-lists

root@OpenWrt:/tmp# opkg install shadowsocks-libev-ss-local_3.2.5-1_arm_cortex-a9_vfpv3.ipk
Installing shadowsocks-libev-ss-local (3.2.5-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev-ss-local:
 * 	libmbedtls12
 * opkg_install_cmd: Cannot install package shadowsocks-libev-ss-local.

root@OpenWrt:/tmp# opkg install libmbedtls12
Unknown package 'libmbedtls12'.
Collected errors:
 * opkg_install_cmd: Cannot install package libmbedtls12.
```

yousong, 2019-03-18:
try something like the following to unpack the .ipk file directly and run the binary ss-local there

```
mkdir -p /tmp/t
cd /tmp/t
tar xzf ../shadowsocks-libev-ss-local_3.2.5-1_arm_cortex-a9_vfpv3.ipk
tar xzf data.tar.gz
ls -al usr/bin/ss-local
```

In the end we will need to turn back to the issue at 18.06.2.

ebenfarnworth, 2019-03-18:
Thanks, that all worked:
```
root@OpenWrt:/tmp/t# tar xzf ../shadowsocks-libev-ss-local_3.2.5-1_arm_cortex-a9_vfpv3.ipk
root@OpenWrt:/tmp/t# ls
control.tar.gz  data.tar.gz     debian-binary
root@OpenWrt:/tmp/t# tar xzf data.tar.gz
root@OpenWrt:/tmp/t# ls
control.tar.gz  data.tar.gz     debian-binary   usr
root@OpenWrt:/tmp/t# ls -al usr/bin/ss-local
-rwxr-xr-x    1 root     root        127407 Mar 14 21:55 usr/bin/ss-local
```
I have not worked out how to start it though. I tried running:
`usr/bin/ss-local`
and cd to: `/usr/bin/` then run `./ss-local`
I uninstalled all the other ss packages before unpacking the new file.
Sorry I really have no idea how things work inside Open-WRT!

yousong, 2019-03-19:
So it did't report "Bus error" as in the original report?

ebenfarnworth, 2019-03-19:
```
root@OpenWrt:/tmp/t/usr/bin# ls
ss-local
root@OpenWrt:/tmp/t/usr/bin# ss-local
-ash: ss-local: not found
root@OpenWrt:/tmp/t/usr/bin# 
```

yousong, 2019-03-19:
Sorry, the conversation is not very productive here.  I am afraid we will need either a live session like teamviewer or remote ssh access to the machine for me to check it in real time.  Please drop me a mail if you are ready.

ebenfarnworth, 2019-04-01:
Thanks @yousong for fixing my setup. It was user error, I had not installed the software correctly; and thanks for maintaining packaging and supporting ss on Open-WRT.

closed 2019-04-01 by ebenfarnworth