openwrt/packages #8155: bind-9.12.3-P1:bind-9.12.3-P1 - openssleddsa_link.c Ed25519 group is not known

wulfy23, 2019-02-08:
Maintainer: @Noah Meyerhans <frodo@morgul.net>
Environment: master ip806x
Description: Trying this reporting thing out.... bind  ( low prio )

```
OpenWrt-libtool: compile:  arm-openwrt-linux-muslgnueabi-gcc -I/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1 -I../.. -I. -I../../lib/dns -Iinclude -I/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1/lib/dns/include -I../../lib/dns/include -I/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1/lib/isc/include -I../../lib/isc -I../../lib/isc/include -I../../lib/isc/unix/include -I../../lib/isc/nothreads/include -I../../lib/isc/noatomic/include -I/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -DUSE_MD5 -DOPENSSL -D_GNU_SOURCE -Os -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1:bind-9.12.3-P1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -fPIC -W -Wall -Wmissing-prototypes -Wcast-qual -Wwrite-strings -Wformat -Wpointer-arith -fno-strict-aliasing -fno-delete-null-pointer-checks -c openssleddsa_link.c  -fPIC -DPIC -o .libs/openssleddsa_link.o
openssleddsa_link.c:39:2: error: #error "Ed25519 group is not known (NID_ED25519)"
 #error "Ed25519 group is not known (NID_ED25519)"
  ^~~~~
openssleddsa_link.c:42:2: error: #error "Ed448 group is not known (NID_ED448)"
 #error "Ed448 group is not known (NID_ED448)"
  ^~~~~
openssleddsa_link.c: In function 'priv_ed25519_to_ossl':
openssleddsa_link.c:138:25: error: 'NID_ED25519' undeclared (first use in this function); did you mean 'NID_sha512'?
  return (d2i_PrivateKey(NID_ED25519, NULL, &p,
                         ^~~~~~~~~~~
                         NID_sha512
openssleddsa_link.c:138:25: note: each undeclared identifier is reported only once for each function it appears in
openssleddsa_link.c: In function 'priv_ed448_to_ossl':
openssleddsa_link.c:175:25: error: 'NID_ED448' undeclared (first use in this function); did you mean 'NID_rc4_40'?
  return (d2i_PrivateKey(NID_ED448, NULL, &p,
                         ^~~~~~~~~
                         NID_rc4_40
openssleddsa_link.c: In function 'openssleddsa_sign':
openssleddsa_link.c:263:20: warning: implicit declaration of function 'EVP_MD_CTX_new'; did you mean 'EVP_MD_CTX_init'? [-Wimplicit-function-declaration]
  EVP_MD_CTX* ctx = EVP_MD_CTX_new();
                    ^~~~~~~~~~~~~~
                    EVP_MD_CTX_init
openssleddsa_link.c:263:20: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
openssleddsa_link.c:288:7: warning: implicit declaration of function 'EVP_DigestSign'; did you mean 'EVP_DigestSignInit'? [-Wimplicit-function-declaration]
  if (!EVP_DigestSign(ctx, sigreg.base, &siglen,
       ^~~~~~~~~~~~~~
       EVP_DigestSignInit
openssleddsa_link.c:298:3: warning: implicit declaration of function 'EVP_MD_CTX_free'; did you mean 'EVP_MD_CTX_create'? [-Wimplicit-function-declaration]
   EVP_MD_CTX_free(ctx);
   ^~~~~~~~~~~~~~~
   EVP_MD_CTX_create
openssleddsa_link.c: In function 'openssleddsa_verify':
openssleddsa_link.c:312:20: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
  EVP_MD_CTX* ctx = EVP_MD_CTX_new();
                    ^~~~~~~~~~~~~~
openssleddsa_link.c:337:11: warning: implicit declaration of function 'EVP_DigestVerify'; did you mean 'EVP_DigestVerifyInit'? [-Wimplicit-function-declaration]
  status = EVP_DigestVerify(ctx, sig->base, siglen,
           ^~~~~~~~~~~~~~~~
           EVP_DigestVerifyInit
openssleddsa_link.c: In function 'openssleddsa_generate':
openssleddsa_link.c:393:9: error: 'NID_ED25519' undeclared (first use in this function); did you mean 'NID_sha512'?
   nid = NID_ED25519;
         ^~~~~~~~~~~
         NID_sha512
openssleddsa_link.c:396:9: error: 'NID_ED448' undeclared (first use in this function); did you mean 'NID_rc4_40'?
   nid = NID_ED448;
         ^~~~~~~~~
         NID_rc4_40
openssleddsa_link.c: In function 'priv_ed25519_to_ossl':
openssleddsa_link.c:140:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
openssleddsa_link.c: In function 'priv_ed448_to_ossl':
openssleddsa_link.c:177:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:335: recipe for target 'openssleddsa_link.lo' failed
make[6]: *** [openssleddsa_link.lo] Error 1
make[6]: Leaving directory '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1/lib/dns'
Makefile:82: recipe for target 'subdirs' failed
make[5]: *** [subdirs] Error 1
make[5]: Leaving directory '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1/lib'
Makefile:88: recipe for target 'subdirs' failed
make[4]: *** [subdirs] Error 1
make[4]: Leaving directory '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1'
Makefile:243: recipe for target '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1/.built' failed
make[3]: *** [/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/bind-9.12.3-P1/.built] Error 2
make[3]: Leaving directory '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/feeds/packages/net/bind'
time: package/feeds/packages/bind/compile#0.13#0.01#0.56
package/Makefile:107: recipe for target 'package/feeds/packages/bind/compile' failed
make[2]: *** [package/feeds/packages/bind/compile] Error 2
make[2]: Leaving directory '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3'
package/Makefile:103: recipe for target '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3'
/xsrc/wrt/_x-ipq/ipq23-18.06.2/v2.3/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

```

dengqf6, 2019-02-08:
Does this error happen only with OpenSSL 1.1.1 and without deprecated API?

wulfy23, 2019-02-08:
Yeah, this is looking like it's something on my end. Thanks for the input.

closed 2019-02-08 by wulfy23

dengqf6, 2019-02-09:
@neheb

neheb, 2019-02-09:
Sounds like bind requires OPENSSL_WITH_EC. And maybe missing headers.