openwrt/packages #8173: Nanopi-neo-plus2 OTG port support

liftoff-sr, 2019-02-10:
linux kernel  for nanopi-neo-plus2   aka sunxi

The OTG port will not recognize a USB 2.0 device connected to it.  I checked the kernel config file and it looks like the device tree and the config file are of sync with respect to the MUSB drivers.  The devicetree compatibility string points to driver
enabled by this line in Makefile:
obj-$(CONFIG_USB_MUSB_SUNXI)			+= sunxi.o

but the corresponding configuration variable seems not to be enabled.

diizzyy, 2019-02-10:
@liftoff-sr 
Wrong repo

Report it here:
https://bugs.openwrt.org/

karlp, 2019-02-11:
Wrong place to report as @diizzyy  says, but you're going to need more than that.  You probably need more like all of this https://github.com/karlp/openwrt/commit/62642bb7a44a9a087ae9195f6ff0807b69a1b802  taht I just need to finish re-testing before I submit.

liftoff-sr, 2019-02-11:
karlp, wrong place for that:
I didn't ask for usbgadget, which I think is USB "device" oriented.  The NEO PLUS2 has a single purpose OTG port, namely it MUST be used as HOST.

antonyantony, 2019-03-01:
Which USB connector are you trying to use as OTG?  The Micro USB connector? I think it is power only no data lines. The other two are in  Host mode. That is what the device tree is set to.

liftoff-sr, 2019-03-01:
@antonyantony :  NEO plus 2 has no microusb port.  Look at the schematic.  This is a kernel configuration bug.  The MUSB drivers are deselected and need to be included, built-in or module.

liftoff-sr, 2019-03-01:
if we clear up the kernel config bug, we can then look at the devicetree again.  I think I remember it being ok, I compared it to mainline but that was weeks ago.  Nearest hurdle is kernel config, lets clear that first.

liftoff-sr, 2019-03-01:
The host port furthest away from the ethernet connector is not working.

liftoff-sr, 2019-03-01:
And that is the OTG port to be dedicated to host mode.

antonyantony, 2019-03-01:
I think we are agreeing there is no  MicroUSB port for OTG.  However, there is a Micro USB connector and it is only for power.
I do not USB on OpeWRT. It are sitting far away. So I can't test now.
However, on Armbian both USB (type A connectors) work for me.
sd 0:0:0:0: [sda] 7821312 512-byte logical blocks: (4.00 GB/3.73 GiB)
sd 1:0:0:0: [sdb] 15482880 512-byte logical blocks: (7.93 GB/7.38 GiB)

I wonder what would be bug in device tree file.

liftoff-sr, 2019-03-01:
My kernel:
root@OpenWrt:~# uname -a
Linux OpenWrt 4.14.63 #0 SMP PREEMPT Wed Aug 15 20:42:39 2018 aarch64 GNU/Linux

Would be nice to have /proc/config.gz

Can you verify that: 

a) OTG will work without the MUSB drivers, which are referenced in the device tree for the OTG port.  Not likely.

b) That the OTG IP block is indeed tied to the outside most USB port?  USB0 and USB3 are referenced in the schematic, and USB0 is the OTG port.

I did not see a bug in the device tree, I see the otg node being hardwired to "host" mode, and this is not to be a variable with this particular schematic.

liftoff-sr, 2019-03-01:
Is the MUSB driver perhaps a separate package that you installed?

antonyantony, 2019-03-01:
> My kernel:
> root@OpenWrt:~# uname -a
> Linux OpenWrt 4.14.63 #0 SMP PREEMPT Wed Aug 15 20:42:39 2018 aarch64 GNU/Linux
> 
> Would be nice to have /proc/config.gz
> 
> Can you verify that:
> 
> a) OTG will work without the MUSB drivers, which are referenced in the device tree for the OTG port. Not likely.

no, rather I don't know.  On Armbian CONFIG_USB_MUSB_SUNXI is enabled also other setting karlp  recommend.

> b) That the OTG IP block is indeed tied to the  outside most USB port? USB0 and USB3 are referenced in the schematic, and USB0 is the OTG port.

I am curious to see if OTG works. I use both ports as host port on Armbian.

> I did not see a bug in the device tree, I see the otg node being hardwired to "host" mode, and this is not to be a variable with this particular schematic.

I can test OpenWRT next week. However I feel [karlp](/karlp) is right you may need more config changes.

liftoff-sr, 2019-03-01:
OK, I see a newer release than what I am using.  If MUSB has been enabled that explains yours working and mine not.  But typically extra GADGET options you reference are for a USB port that is in "device" mode.  I don't have experience putting a USB A port into device mode.  Maybe its possible, and if so gadget features could be helpful to somebody.  Otherwise I might have left them out, unless the kernel wouldn't build.

I suspect this may be a closed issue now, I will do the upgrade posthaste.
Thanks @antonyantony !
I appreciate the help, I was not anxious to fire up the build system myself, got too many others already stretching me in 6 different directions :)

liftoff-sr, 2019-03-01:
New theory:  the upgrade will not fix the issue, as the kernel seems unchanged.   I hope whoever is responsible for the release can also include /proc/config.gz support next time.  This machine is not memory challenged, and it can reduce frustration with uncertainty of build config.

Please consider me waiting patiently for a fix.  Bye.

diizzyy, 2019-03-28:
Closing, wrong repo

closed 2019-03-28 by diizzyy