openwrt/packages #8175: [libpam] feature suggestion - add PAM limit module

n8v8R, 2019-02-10:
@nmav 

if feasible I would appreciate the `pam_limits.so` to be added in order to be able to fine grain/tune ulimit limits for user-land sessions, which imho would add to security and resource management.

nmav, 2020-01-13:
@n8v8R would you like to send an MR?

neheb, 2020-01-24:
This is already the case:

```
mangix@mangix-trapnet ~/d/openwrt> find build_dir/target-mips_74kc+dsp2_musl/Linux-PAM-1.3.1/ipkg-mips_74kc_dsp2/ | grep limits
build_dir/target-mips_74kc+dsp2_musl/Linux-PAM-1.3.1/ipkg-mips_74kc_dsp2/libpam/etc/security/limits.d
build_dir/target-mips_74kc+dsp2_musl/Linux-PAM-1.3.1/ipkg-mips_74kc_dsp2/libpam/etc/security/limits.conf
build_dir/target-mips_74kc+dsp2_musl/Linux-PAM-1.3.1/ipkg-mips_74kc_dsp2/libpam/lib/security/pam_limits.so
```

Closing.

closed 2020-01-24 by neheb