openwrt/packages #8179: softethervpn5: running OK with snapshots, missing it in releases

lars18th, 2019-02-11:
Hi,

The package `softethervpn5` has now official support: https://github.com/openwrt/packages/pull/7936 https://github.com/openwrt/packages/pull/6577

I test it (as client) in x86_64 at it works like a charm for a week.
- However, it only appears in the snapshot repository:
https://downloads.openwrt.org/snapshots/packages/x86_64/packages/
- And it's missing from the release ropositories:
http://downloads.openwrt.org/releases/18.06.2/packages/x86_64/packages/
http://downloads.openwrt.org/releases/18.06.1/packages/x86_64/packages/

So, to use/install it's required to install some snapshot libraries.

I suggest to move it to the release branch.

lars18th, 2019-02-12:
Hi,

Regarding this package `softethervpn5`, please take note that it supersedes the package `softethervpn` that is legacy:
- Package `softethervpn` is included in the release branch of packages for 18.06:
https://github.com/openwrt/packages/tree/openwrt-18.06/net/softethervpn
- Package `softethervpn` doesn't compile at all:
https://downloads.openwrt.org/releases/faillogs/x86_64/packages/softethervpn/compile.txt
- Package `softethervpn5` is merged to the master: (https://github.com/openwrt/packages/pull/6577)
https://github.com/openwrt/packages/tree/2636fb9a169613eaee46e124da07a2dc06afb953/net/softethervpn5
- However, `softethervpn5` isn't already in the release branch:
https://github.com/openwrt/packages/tree/openwrt-18.06/net/softethervpn5

So, as it is tested for x86_64, arm/mvebu and mips/qemu; then I suggest to merge it to the 18.06 release branch, and remove the ancient `softethervpn` from it (as the developer of the package doesn't maintain it).

Regards.

Andy2244, 2019-02-12:
Hi, i guess i can setup a PR for 18.06, need to retest against 18.x again.

poranje, 2019-02-23:
While you're at it: also adding samba4 to 18.06 would be much appreciated.
Paul

Andy2244, 2019-02-23:
ok if i have some more time next week, i will try

diizzyy, 2019-03-28:
@Andy2244 
Seeing that 19.XX is to be released "soon-ish" is this something you still intend to backport?

Andy2244, 2019-03-28:
Not sure, i have little time atm and my to-do list is already filled. 
The actual pull requests for 18.x are trivial to setup, but i'm not sure i want to maintain updates for my extra feed + snapshots + 19.x and than 18.x for every change i do. On top i also have to maintain separate luci changes for snapshots, 19.x and 18.x .

I wonder how other maintainers handle this, it seems a simple softether/samba/cifsd version update will result in 3 separate pull requests for me than.

I wonder if i should instead just focus on stable 19.x support + snapshots.

diizzyy, 2019-03-28:
I'd suggest that you drop 18.X support but that's just me...

Andy2244, 2019-04-10:
I will not merge it for 18.x and maintain it for 19.x and snapshots. I don't have the time/setup to verify, test possible 18.x backports. It should still be possible to build a compatible package via my package-builder for 18.x.

poranje, 2019-04-13:
> 
> I will not merge it for 18.x and maintain it for 19.x and snapshots. I don't have the time/setup to verify, test possible 18.x backports. It should still be possible to build a compatible package via my package-builder for 18.x.

No problem, already using the package-buider, but would have been nice.
Bye, Paul

lars18th, 2019-06-07:
> I will not merge it for 18.x, but maintain it for 19.x and snapshots. I don't have the time/setup to verify, test possible 18.x backport changes. It should still be possible to build a compatible package via my package-builder for 18.x.

Hi @Andy2244 ,

I think that until the end of 2019 or the beginning of 2020 no new release of OpenWRT will be made. So, 19.x will be out of scope _for months_.

Please, can you reconsider a backport to 18.x?
Thank you!

diizzyy, 2019-06-09:
@lars18th 
The release schedule/roadmap/* is determined by the OpenWrt core team, if you have issues with this feel free to voice your opinion on the mailinglists or use master branch (ie compile your own firmware).  Please respect other people's time and effort and keep in mind that this is more or less a hobby project people spend their free/spare time on.

closed 2019-06-09 by diizzyy

lars18th, 2019-06-10:
Hi @diizzyy ,

I feel like you misunderstood my message. I never wanted to criticize the development roadmap of the OpenWRT project. I have simply shown that right now the date of the next release has been postponed.

And so I encouraged @Andy2244 to decide if it would be good to reconsider supporting this package for the 18.x branch, as it is going to be active for a long time.

If you understood otherwise, I apologize for my mistake.
Please reopen this issue.
Thank you!

lars18th, 2019-06-11:
Hi @diizzyy ,

I see today the creation of the new "branch 19.07" in the repository: https://github.com/openwrt/openwrt/commit/b344ca9ca68322a023414e7e38ddd4d81aa8d378

So, please don't care about my previous message. I hope @Andy2244 will update his package for the next version "19.07" when the beta is available.

Thank you all for your good work!