openwrt/packages #8180: nodogsplash: libmicrohttpd dependency selects libmicrohttpd-ssl instead

mwarning, 2019-02-11:
On a current snapshot:

```
opkg update
opkg install nodogsplash
```

.. installs  libmicrohttpd-ssl (+ libgcrypt, libgnutls, ...).
But in the nodogsplash package, the package control file states: `Depends: libc, libpthread, iptables-mod-ipopt, libmicrohttpd`

From `opkg install --verbosity=4 nodogsplash`:
```
pkg_hash_fetch_unsatisfied_dependencies: satisfying_pkg=0x9a0410
pkg_hash_fetch_best_installation_candidate: Best installation candidate for kmod-ipt-core:
pkg_hash_fetch_best_installation_candidate: Adding kmod-ipt-core to providers.
pkg_hash_fetch_best_installation_candidate: kmod-ipt-core arch=mipsel_24kc arch_priority=10 version=4.14.98.
pkg_hash_fetch_best_installation_candidate: Candidate: kmod-ipt-core 4.14.98.
pkg_hash_fetch_unsatisfied_dependencies: satisfying_pkg=0x99ebe0
pkg_hash_fetch_best_installation_candidate: Best installation candidate for libmicrohttpd:
pkg_hash_fetch_best_installation_candidate: apkg=libmicrohttpd nprovides=2.
pkg_hash_fetch_best_installation_candidate: Adding libmicrohttpd-ssl to providers.
pkg_hash_fetch_best_installation_candidate: Adding libmicrohttpd to providers.
pkg_hash_fetch_best_installation_candidate: libmicrohttpd-ssl arch=mipsel_24kc arch_priority=10 version=0.9.62.
pkg_hash_fetch_best_installation_candidate: libmicrohttpd arch=mipsel_24kc arch_priority=10 version=0.9.62.
pkg_hash_fetch_best_installation_candidate: 2 matching pkgs for apkg=libmicrohttpd:
pkg_hash_fetch_best_installation_candidate: libmicrohttpd 0.9.62 mipsel_24kc
pkg_hash_fetch_best_installation_candidate: libmicrohttpd-ssl 0.9.62 mipsel_24kc
pkg_hash_fetch_best_installation_candidate: No matching pkg out of 2 matching_apkgs.
pkg_hash_fetch_unsatisfied_dependencies: satisfying_pkg=0
pkg_hash_fetch_best_installation_candidate: Best installation candidate for libmicrohttpd:
pkg_hash_fetch_best_installation_candidate: apkg=libmicrohttpd nprovides=2.
pkg_hash_fetch_best_installation_candidate: Adding libmicrohttpd-ssl to providers.
pkg_hash_fetch_best_installation_candidate: Adding libmicrohttpd to providers.
pkg_hash_fetch_best_installation_candidate: libmicrohttpd-ssl arch=mipsel_24kc arch_priority=10 version=0.9.62.
pkg_hash_fetch_best_installation_candidate: libmicrohttpd arch=mipsel_24kc arch_priority=10 version=0.9.62.
pkg_hash_fetch_best_installation_candidate: Candidate: libmicrohttpd 0.9.62.
pkg_hash_fetch_best_installation_candidate: Candidate: libmicrohttpd-ssl 0.9.62.
pkg_hash_fetch_best_installation_candidate: 2 matching pkgs for apkg=libmicrohttpd:
pkg_hash_fetch_best_installation_candidate: libmicrohttpd 0.9.62 mipsel_24kc
pkg_hash_fetch_best_installation_candidate: libmicrohttpd-ssl 0.9.62 mipsel_24kc
pkg_hash_fetch_unsatisfied_dependencies: satisfying_pkg=0x990780
pkg_hash_fetch_best_installation_candidate: Best installation candidate for libc:
pkg_hash_fetch_best_installation_candidate: Adding libc to providers.
pkg_hash_fetch_best_installation_candidate: libc arch=mipsel_24kc arch_priority=10 version=1.1.21.
pkg_hash_fetch_best_installation_candidate: Candidate: libc 1.1.21.
```

mwarning, 2019-02-11:
Discussion started here: https://github.com/openwrt/packages/pull/8075

bluewavenet, 2019-02-11:
@mwarning 
Would this be better entitled:
"libmicrohttpd: nodogsplash dependency selects libmicrohttpd-ssl instead" ?

mwarning, 2019-02-11:
Related: https://github.com/nodogsplash/nodogsplash/issues/339

bluewavenet, 2019-03-13:
No progress on this - still an issue a month later.
As far as I can see it is an issue for ANY package that wants only the non-ssl version.
It makes a likely uncomfortable fit in a 64MB device, let alone a 32MB unfortunately.
@mwarning @lynxis ?

mwarning, 2019-03-14:
I have no time right now to address the issue.

bluewavenet, 2019-03-26:
@diizzyy @lynxis @mwarning 
The Nodogsplash new release(3.3.0-1) is now ONLY compatible with Snapshot because of this issue. For 18.06, preinstalling the old version of libmicrohttpd-nossl is necessary.
Do we have to make a special backport version of Nodogsplash for 18.06 to work around this issue or can it be fixed here?
In addition, for Snapshot/19.xx onwards, any packages with libmicrohttpd as a dependency will end up with the ssl version regardless.
Having named -nossl versions seems to be the norm elsewhere, could we not revert to that way of doing things?

mwarning, 2019-04-01:
@jow- do you happen to know if this is an opkg bug? Thanks!

bluewavenet, 2019-04-02:
This issue is maybe showing up a bug in opkg but that is not really the long term issue here. The problem really is the change from previous naming that breaks things and makes supporting packages across versions of OpenWrt more complicated. There does not seem to be any real technical reason for the change that caused this issue as far as I can see.

closed 2020-01-12 by neheb