openwrt/packages #8184: banIP: whitelist ignores IPv6 addresses

R-Adrian, 2019-02-11:
Maintainer: @dibdot  

Environment: 
```
system type             : Qualcomm Atheros QCA9558 ver 1 rev 0
machine                 : TP-LINK TL-WR1043ND v2
cpu model               : MIPS 74Kc V5.0

root@MyRouter-v2:~# opkg list banip
banip - 0.1.0-1 - Powerful banIP script to block ip addresses via ipsets.

Firmware Version	SWRT SNAPSHOT r8392+901-3dba852547 / LuCI Master (git-19.038.66435-4e5111e)
Kernel Version	4.14.98
```

Issue description: banIP ignores IPv6 addresses when added to the whitelist.

I tried adding the addresses from CloudFlare to the whitelists ( https://www.cloudflare.com/ips/ ) , and banIP created a whitelist ipset only for the IPv4 addresses, it completely ignored the IPv6 ones.

dibdot, 2019-02-11:
Please provide an example whitelist, to reproduce your problem  - thanks.

R-Adrian, 2019-02-11:
this is my whitelist:
(basically, it's https://www.cloudflare.com/ips-v4 and https://www.cloudflare.com/ips-v6 copied and pasted in the same list since the whitelist doesn't have a place where to define download URLs)

```
103.21.244.0/22
103.22.200.0/22
103.31.4.0/22
104.16.0.0/12
108.162.192.0/18
131.0.72.0/22
141.101.64.0/18
162.158.0.0/15
172.64.0.0/13
173.245.48.0/20
188.114.96.0/20
190.93.240.0/20
197.234.240.0/22
198.41.128.0/17
2400:cb00::/32
2405:b500::/32
2606:4700::/32
2803:f800::/32
2c0f:f248::/32
2a06:98c0::/29

```

dibdot, 2019-02-11:
Hmm, works for me:
[image]

Please provide/check your whitelist config section.

R-Adrian, 2019-02-11:
hmm... apparently _6 is turned off by default and there's no option in the Luci UI to turn it on.. i'll try turning it on via config.

```
config source 'whitelist'
        option ban_src '/etc/banip/banip.whitelist'
        option ban_src_6 '/etc/banip/banip.whitelist'
        option ban_src_desc 'Always allow these IPs (IPv4/IPv6)'
        option ban_src_rset '/^(([0-9]{1,3}\.){3}[0-9]{1,3}(\/[0-9]{1,2})?)([[:space:]]|$)/{print \"add whitelist \"\$1}'
        option ban_src_rset_6 '/^([0-9a-fA-F]{0,4}:){1,7}[0-9a-fA-F]{0,4}(:\/[0-9]{1,2})?([[:space:]]|$)/{print \"add whitelist_6 \"\$1}'
        option ban_src_settype 'net'
        option ban_src_ruletype 'src+dst'
        option ban_src_on '1'
        option ban_src_on_6 '0'

```

R-Adrian, 2019-02-11:
i see now... the default option ban_src_on_6 '0'  should be turned on... because even if it says "ban_src" it's actually "allow src" in this context.

closed 2019-02-11 by R-Adrian

dibdot, 2019-02-11:
Reset your LuCI cache if you don't see the 'enable IPv6' column (rm -rf /tmp/luci-*) ...
[image]

R-Adrian, 2019-02-11:
that was it... i had a stale cache.