openwrt/packages #8194: unbound: no forward is used

mistepien, 2019-02-12:
openwrt: trunk
unbound - 1.9.0-1

this is /etc/config/unbound generated by luci-app-unbound

> config unbound
	option dns64 '0'
	option edns_size '1280'
	option hide_binddata '1'
	option localservice '1'
	option manual_conf '0'
	option recursion 'default'
	option root_age '9'
	option ttl_min '120'
	option verbosity '1'
	option enabled '1'
	option trigger_interface 'wan'
	option unbound_control '1'
	option listen_port '5353'
	option dhcp_link 'none'
	option add_wan_fqdn '0'
	option add_extra_dns '0'
	option protocol 'ip4_only'
	option add_local_fqdn '0'
	option validator '0'
	option rebind_protection '0'
	option resource 'medium'
	option extended_stats '1'
	option rebind_localhost '0'
	option domain_type 'transparent'

> config  zone
	option zone_type 'forward_zone'
	option tls_upstream '1'
	option enabled '1'
	list server '176.103.130.132'
	list server '176.103.130.134'
	option tls_index 'dns-family.adguard.com'
	option tls_port '853'
	option fallback '0'

unbound does not use given above formard

mistepien, 2019-02-12:
the was mistake, no "list zone_name '.'" in config zone section

EricLuehrsen, 2019-03-13:
if resolved, can this be closed?

EricLuehrsen, 2019-03-24:
@hnyman  administrative closure then?

closed 2019-03-24 by hnyman