openwrt/packages #8200: mariadb: wrong dependencies relation between packages

p4u, 2019-02-12:
Maintainer: @micmac1 
Environment: master

Description:
There is a very new strange relation between mariadb subpackages. So, for instance

`mariadb-client` is strict dependent on `mariadb-client-base` which is also strict dependent on `mariadb-base`

https://github.com/openwrt/packages/blob/master/utils/mariadb/Makefile#L243
https://github.com/openwrt/packages/blob/master/utils/mariadb/Makefile#L258

That means that unless `mariadb-base` and `maria-client-base` are explicitly selected `mariadb-client` cannot be installed (or included as dependency for a third package).

This behavior breaks the standard operation which is the opposite: if `mariadb-client` is selected, its dependences are automatically selected too.

Is it possible that a '+' character has been forgotten?

micmac1, 2019-02-12:
No, I kind of like it this way. It keeps the view in utils/databases clean.

p4u, 2019-02-12:
So, I'm using mariadb-server as dependency for several packages and it's very hard to understand how this dependences work. I believe this behavior will mess up many users and developers. Usually when "mariadb-server" is selected the common files package ("mariadb-base") is also selected, that's the standard way to do it.

I'd like to know more opinions regarding this.

p4u, 2019-02-12:
If I want to use mariadb-client as dependency for another package, then I must include as dependency `mariadb-base`,  `maria-client-base` and  `mariadb-client`. Sorry, but I don't see the point, it does not makes sense to me.

micmac1, 2019-02-12:
That's not correct, you just need mariadb-client as a dep. Your package will work as expected.

Building is a different subject. You need to make the proper selections. But in that regard there is no difference to other packages that work the same way.

p4u, 2019-02-12:
So, I have the following package:

```
include $(TOPDIR)/rules.mk

PKG_NAME:=motionalarm
PKG_VERSION=0.1
include $(INCLUDE_DIR)/package.mk

define Package/$(PKG_NAME)
  SECTION:=utils
  CATEGORY:=Utilities
  TITLE:=alarm monitor
  MAINTAINER:=Pau
  DEPENDS:= +i2c-tools +kmod-i2c-gpio +kmod-i2c-core +mariadb-server
endef

define Package/$(PKG_NAME)/description
  Check the i2c accelerometer and perform actions
endef

define Build/Compile
endef

define Package/$(PKG_NAME)/install
	$(INSTALL_DIR) $(1)/
	$(CP) ./files/* $(1)/
endef

$(eval $(call BuildPackage,$(PKG_NAME)))
```

Now I want to select the package on "menuconfig", but it is not listed because

```
Depends on: GPIO_SUPPORT [=y] && PACKAGE_mariadb-server-base [=n]                                                                                                                                            
```

So until I explicitly select `mariadb-server-base` and `mariadb-base` my package won't be available in menuconfig. That's very confusing.

In addition, now I made another package which depends on the first one (motionalarm). This second package will have a broken dependency because motionalarm cannot be selected (unless explicitly someone selects `mariadb-server-base` and `mariadb-base`).

p4u, 2019-02-12:
> That's not correct, you just need mariadb-client as a dep. Your package will work as expected. Building is a different subject. You need to make the proper selections. But in that regard there is no difference to other packages that work the same way.

So which part is not correct?

neheb, 2020-01-12:
What's the progress here?

micmac1, 2020-01-12:
There is no wrong dependency here. This issue can be closed from my point of view.

closed 2020-01-12 by neheb