openwrt/packages #8209: python-cryptography: Builds have been failing

jefferyto, 2019-02-13:
Maintainer: me / @commodo 
Environment: too many to list here

Description:
Faillogs:

* [aarch64_generic](https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/python-cryptography/)
* [arm_cortex-a15_neon-vfpv4](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/packages/python-cryptography/)
* [i386_pentium](https://downloads.openwrt.org/snapshots/faillogs/i386_pentium/packages/python-cryptography/)
* [mips_24kc](https://downloads.openwrt.org/snapshots/faillogs/mips_24kc/packages/python-cryptography/)
* and many more

The earliest faillog I found was for [arm_mpcore_vfp](https://downloads.openwrt.org/snapshots/faillogs/arm_mpcore_vfp/packages/python-cryptography/), from Jan 18.

I believe the platforms where failures have been happening are where we are compiling OpenSSL 1.1 and/or OpenSSL is compiled with `no-engine`. I found a few related issues (termux/termux-packages#2847, pyca/cryptography#4412), but it doesn't look like there will be a fix in cryptography (pyca/cryptography/#3867).

I'm a little out of my depth on this one, would there be folks more familiar with OpenSSL be able to help? (@neheb ?)

commodo, 2019-02-13:
I'll take a look these days as well.

neheb, 2019-02-13:
https://github.com/openwrt/packages/pull/7268

Seems I was too early

jefferyto, 2019-02-16:
I've tried #7268 locally and it appears to compile fine.

jefferyto, 2019-02-18:
@commodo @neheb Would reopening / merging #7268 be the right solution?

commodo, 2019-02-18:
yep; let's re-do #7268  
https://github.com/openwrt/openwrt/pull/965#issuecomment-462948386  has been merged 5 days ago
weird/funny timing

commodo, 2019-02-18:
@neheb    do you want to re-open #7268   or shall i ?

neheb, 2019-02-18:
Did so.

jefferyto, 2019-02-19:
For reference, #8245 is the new PR.

neheb, 2019-02-22:
Dirty patch to fix both dependencies: https://gist.github.com/neheb/4e8266ab730a91ea303c3467b8bbf02a

commodo, 2019-02-22:
that patch doesn't seem too dirty, and we could probably make it cleaner
we could also propose this upstream
let me try to re-iterate on it

neheb, 2019-02-22:
Already did:

https://github.com/pyca/cryptography/pull/4760
https://github.com/pyca/cryptography/pull/4761
https://github.com/pyca/cryptography/pull/4762

commodo, 2019-02-22:
Oh.
Let me try those patches.
They look very nice

neheb, 2019-02-24:
I looked into the engine stuff again. Turns out it was enabled when I was testing.

After looking at it again, this is going to be more problematic. CMAC_Init and HMAC_Init are functions that take ENGINE as a parameter. I don't see an easy way of disabling those without creating stub functions.

Deprecated APIs should be doable though.

jefferyto, 2019-02-25:
From pyca/cryptography#3867 and [this comment](https://github.com/pyca/cryptography/issues/4412#issuecomment-415589704) it seems they have a custom engine that implements a fork-safe RNG.

There is a patch in pyca/cryptography#4714 to use the internal RNG in OpenSSL 1.1.1 but it wasn't merged. Perhaps getting a version of this merged may help?

neheb, 2019-02-25:
NO ENGINE support was merged: https://github.com/pyca/cryptography/commit/76c784340c3851f402abc38dff8fa5f008cdc4d4

jefferyto, 2019-02-25:
Oh great :tada::tada::joy:

commodo, 2019-02-25:
Even with that patch, it doesn't compile for me.
This is getting a bit frustrating, but I'll keep looking.

I tried to backport that patch to 2.5, and I get:
```
build/temp.linux2-2.7/_openssl.c: In function 'Cryptography_pem_password_cb':
build/temp.linux2-2.7/_openssl.c:3014:35: warning: conversion to 'size_t {aka unsigned int}' from 'int' may change the sign of the result [-Wsign-conversion]
         memcpy(buf, st->password, st->length);
                                   ^~
build/temp.linux2-2.7/_openssl.c: In function '_cffi_const_ENGINE_METHOD_ALL':
build/temp.linux2-2.7/_openssl.c:7765:12: error: 'ENGINE_METHOD_ALL' undeclared (first use in this function); did you mean 'ENGINE_METHOD_RAND'?
   int n = (ENGINE_METHOD_ALL) <= 0;
            ^~~~~~~~~~~~~~~~~
            ENGINE_METHOD_RAND
build/temp.linux2-2.7/_openssl.c:7765:12: note: each undeclared identifier is reported only once for each function it appears in
build/temp.linux2-2.7/_openssl.c: In function '_cffi_const_ENGINE_METHOD_CIPHERS':
build/temp.linux2-2.7/_openssl.c:7772:12: error: 'ENGINE_METHOD_CIPHERS' undeclared (first use in this function); did you mean 'ENGINE_METHOD_RAND'?
   int n = (ENGINE_METHOD_CIPHERS) <= 0;
            ^~~~~~~~~~~~~~~~~~~~~
            ENGINE_METHOD_RAND
build/temp.linux2-2.7/_openssl.c: In function '_cffi_const_ENGINE_METHOD_DIGESTS':
build/temp.linux2-2.7/_openssl.c:7779:12: error: 'ENGINE_METHOD_DIGESTS' undeclared (first use in this function); did you mean 'ENGINE_METHOD_RAND'?
   int n = (ENGINE_METHOD_DIGESTS) <= 0;
            ^~~~~~~~~~~~~~~~~~~~~
            ENGINE_METHOD_RAND
build/temp.linux2-2.7/_openssl.c: In function '_cffi_const_ENGINE_METHOD_DSA':
build/temp.linux2-2.7/_openssl.c:7786:12: error: 'ENGINE_METHOD_DSA' undeclared (first use in this function); did you mean 'ENGINE_METHOD_RAND'?
   int n = (ENGINE_METHOD_DSA) <= 0;
            ^~~~~~~~~~~~~~~~~
            ENGINE_METHOD_RAND
build/temp.linux2-2.7/_openssl.c: In function '_cffi_const_ENGINE_METHOD_NONE':
build/temp.linux2-2.7/_openssl.c:7793:12: error: 'ENGINE_METHOD_NONE' undeclared (first use in this function); did you mean 'ENGINE_METHOD_RAND'?
   int n = (ENGINE_METHOD_NONE) <= 0;

```

neheb, 2019-02-25:
The patch before it removes those functions I believe. I will take a crack at this.

commodo, 2019-02-25:
that fixed it :)
thanks

do you want to do a PR, or shall i ?

neheb, 2019-02-25:
Go ahead.

commodo, 2019-02-25:
created PR #8293

jefferyto, 2019-03-11:
Fixed. @neheb @commodo thanks for all your help!

closed 2019-03-11 by jefferyto