openwrt/packages #8210: prometheus: Builds have been failing

jefferyto, 2019-02-13:
Maintainer: @aparcar 
Environment: too many to list here

Description:
Faillogs:

* [aarch64_generic](https://downloads.openwrt.org/snapshots/faillogs/aarch64_generic/packages/prometheus/)
* [arm_cortex-a15_neon-vfpv4](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/packages/prometheus/)
* [i386_pentium](https://downloads.openwrt.org/snapshots/faillogs/i386_pentium/packages/prometheus/)
* [mips_24kc](https://downloads.openwrt.org/snapshots/faillogs/mips_24kc/packages/prometheus/)
* and many more

Looks like it is failing because `bzr` isn't installed on buildbots.

aparcar, 2019-03-20:
Can't we install `bzr` on the buildbots?

aparcar, 2019-03-20:
If I remember correctly we did the same for the CI build containers...

diizzyy, 2019-03-20:
The general rule of thumb is to not require additional external deps except the ones specified/required by OpenWrt. I personally think that requiring bzr on host should be avoided if possible.

jefferyto, 2019-03-21:
As @neheb [noted](https://github.com/openwrt/packages/pull/8451#issuecomment-475034811), upstream has removed the bzr dependency (prometheus/prometheus#5378). I assume the change will be reflected in the next upstream release.

aparcar, 2019-03-22:
@jefferyto could you helpme backport that patch to 2.8.0? Or would that be a waste of time?

jefferyto, 2019-03-26:
Fixed by #8468.

closed 2019-03-26 by jefferyto