openwrt/packages #8213: Several packages which use OpenSSL have failed to build since the update to 1.1.1a

dengqf6, 2019-02-14:
Such as softethervpn5, freeradius3. For details see faillogs
@neheb Could you please take a look?

neheb, 2019-02-14:
Yeah. some of them I have open pull requests for. Not those.

lucize, 2019-02-15:
is libinput related ? it worked some days ago 
https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/libinput/compile.txt

neheb, 2019-02-15:
https://github.com/openwrt/packages/pull/8203

lucize, 2019-02-17:
@neheb what do you think about this one ? (without deprecated api)

```
ssh.c: In function 'main':
ssh.c:819:8: warning: implicit declaration of function 'SSLeay_version'; did you mean 'RSA_get_version'? [-Wimplicit-function-declaration]
        SSLeay_version(SSLEAY_VERSION)
        ^~~~~~~~~~~~~~
        RSA_get_version
ssh.c:819:23: error: 'SSLEAY_VERSION' undeclared (first use in this function); did you mean 'SSH_VERSION'?
        SSLeay_version(SSLEAY_VERSION)
                       ^~~~~~~~~~~~~~
                       SSH_VERSION
ssh.c:819:23: note: each undeclared identifier is reported only once for each function it appears in
ssh.c:1041:2: warning: implicit declaration of function 'ERR_load_crypto_strings'; did you mean 'ERR_load_DH_strings'? [-Wimplicit-function-declaration]
  ERR_load_crypto_strings();
  ^~~~~~~~~~~~~~~~~~~~~~~
  ERR_load_DH_strings
Makefile:160: recipe for target 'ssh.o' failed

```

neheb, 2019-02-17:
Fixed upstream. Unfortunately, backporting all the patches would be a huge pain. It's probably better to wait for the next stable release.

I have a patch that fixes it in the meantime but it's probably not worth posting.

lucize, 2019-02-17:
I'm trying to debug this one ssh_dispatch_run_fatal: Connection from user xxxx zzzz port yyyy: Connection corrupted on rsync after the update

neheb, 2019-02-17:
No idea. I've been using rsync after the update with no issues.

lucize, 2019-02-17:
sorry, rsync ubuntu client and openwrt openssh-server, not dropbear

neheb, 2019-02-17:
right. I use the same.

lucize, 2019-02-17:
maybe aes-ni cpu problem ?

neheb, 2019-02-17:
could be. I force ChaCha20-Poly1305 as it's faster.

lucize, 2019-02-18:
so without engine support and force poly over gcm aes seems to work, I'll try without poly and without engine

neheb, 2019-02-18:
@lucize there's a patch on the mailing list called "openssl: backport devcrypto changes from master". This will probably fix your issues.

lucize, 2019-02-18:
with af_alg seems to work, didn't enable devcrypto so maybe that's the problem, I'll try when is merged for now I'm good

closed 2019-04-03 by dengqf6