openwrt/packages #8218: openconnect can not connect with DTLS while choose openssl

xseven007, 2019-02-14:
openconnect can not connect with DTLS while build with openssl. When the openwrt update the libopenssl to 1.1.1a, then the openconnect cannot connect with dtls. But It can be connected with DTLS while build with gnutls.

nmav, 2019-02-15:
May I suggest to bring that upstream on https://gitlab.com/openconnect/openconnect ? I do not believe we have any patches that change the behavior.

xseven007, 2019-02-15:
The latest release is OpenConnect v8.02 (PGP signature), released on 2019-01-16 with the following changelog:

    Fix GNU/Hurd build.
    Discover vpnc-script in default packaged location on FreeBSD/OpenBSD.
    Support split-exclude routes for GlobalProtect.
    Fix GnuTLS builds without libtasn1.
    Fix DTLS support with OpenSSL 1.1.1+.
    Add Cisco-compatible DTLSv1.2 support.
    Invoke script with reason=attempt-reconnect before doing so.

Maybe upgrade to 8.02 would be fix this issue

nmav, 2019-05-30:
I've updated master to 8.03. Please re-open if it doesn't address it.

commit ebb45bb5 (2019-05-30): openconnect: updated to 8.03

Resolves: #8218
closed 2019-05-30 by nmav