openwrt/packages #8223: ddns-scripts: failures are logged as successes

tmo1, 2019-02-15:
Maintainer: @chris5560
Environment: TP-Link C2600, 18.06.2

For (at least some) failures, the logs still indicate success. E.g.:

112844       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/afraid.dat -o /var/run/ddns/afraid.err --ca-directory=/etc/ssl/certs/ --no-proxy 'https
://freedns.afraid.org/dynamic/update.php?***PW***&address=xxx.xxx.xxx.xxx
 112846       : DDNS Provider answered:
ERROR: Invalid update URL (2)
 112846  info : Update successful - IP 'xxx.xxx.xxx.xxx' send
 112847  info : Forced update successful - IP: 'xxx.xxx.xxx.xxx' send

 133925       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/afraid.dat -o /var/run/ddns/afraid.err --ca-directory=/etc/ssl/certs/ --no-proxy 'https
://sync.afraid.org/u/***PW***/?address=xxx.xxx.xxx.xxx'
 133927       : DDNS Provider answered:
Couldn't find any matching DNS records via randomkey method!
 133928  info : Update successful - IP 'xxx.xxx.xxx.xxx' send
 133928  info : Forced update successful - IP: 'xxx.xxx.xxx.xxx' send

chris5560, 2019-02-15:
Looks like afraid.org has informed it's users a (long) time ago, that their update url will change.
Somebody familiar with afraid.org needs to change services-files and the used config-file fields.

For the moment You could try the following config entry's.
I have no afraid accout to check.

username:	guest
password:	guest
domain:		dns_name,update_key	(Example: zeus.afraid.org,AFFEEFFFCCDDEEEFFFCC==)
lookup_host:	dns_name		(Example: zeus.afraid.org)

tmo1, 2019-02-17:
On Fri, 15 Feb 2019 07:52:37 +0000 (UTC)
Christian Schoenebeck <notifications@github.com> wrote:

> Looks like afraid.org has informed it's users a (long) time ago, that their update url will change.
> Somebody familiar with afraid.org needs to change services-files and the used config-file fields. [...]

Thanks. I actually did get it to work with the following configuration:

	option enabled '1'
	option lookup_host 'nnnnnnnn'
	option use_https '1'
	option cacert '/etc/ssl/certs/'
	option service_name 'afraid.org-v2-token'
	option password 'xxxxxxxx'

I just think that the script should do a better job of reporting
failure as failure rather than success. As per my original report, the
problem is not limited to the update url error - the same thing occurs
with other sorts of errors. If afraid is reporting any sort of failure,
the script should understand and report that, rather than reporting
success.

Tmo <tmore1@gmx.com>