openwrt/packages #8227: rp-pppoe: pppoe-server fails with BRIDGE Interfaces

lars18th, 2019-02-15:
Hi,

Using the package "rp-pppoe" the "pppoe-server" daemon fails when selecting as the listening interface a Bridge.

The problem is because the init script is searching for the physical device, but it doesn't test if it's a bridge.
Here the code:
- https://github.com/openwrt/packages/blob/openwrt-18.06/net/rp-pppoe/files/pppoe-server.init#L42
Repacing this line with: `device=$interface` it works. However, that's a dirty fix.

I suggest the original developer (@alinnastac) to fix this.

Regards.

alinnastac, 2019-02-15:
Hi,

Bridge interfaces have device attributes:
`root@mymodem:~# ifstatus lan | head -n 10
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 19268,
        "l3_device": "br-lan",
        "proto": "static",
        "device": "br-lan",
`
 What kind of interface are you using? 

BR,
Alin

lars18th, 2019-02-15:
Hi @alinnastac ,

Thank you! Now I see the problem: I'm using the "br-NAME" as the listening interface, instead of the "NAME".

So, I close this issue! :wink:

closed 2019-02-15 by lars18th