openwrt/packages #8232: ss - Cannot open netlink socket: Protocol not supported

Strykar, 2019-02-16:
@russelsenior
@jow- unsure if this should be opened on bugs.openwrt.org since it's part of iproute2?
```
Linux 4.14.95 #0 SMP Jan 2019 x86_64 GNU/Linux OpenWrt 18.06.2 r7676-cddd7b4c77
```
The ss package reports `Cannot open netlink socket: Protocol not supported` errors when run.

The last reference to this issue I found was from two years ago, marked Fixed - https://dev.archive.openwrt.org/ticket/20544.html
`[strykar@i-5 ~/github/openwrt]$ grep CONFIG_NETLINK_DIAG .config` in my buildroot does not show it and the strace log looks like it's indeed the same issue.
```
root@apu:~# ss
Cannot open netlink socket: Protocol not supported
Cannot open netlink socket: Protocol not supported
Cannot open netlink socket: Protocol not supported
Cannot open netlink socket: Protocol not supported
Cannot open netlink socket: Protocol not supported
Cannot open netlink socket: Protocol not supported
Cannot open netlink socket: Protocol not supported
Netid                    State                      Recv-Q                      Send-Q                                                 Local Address:Port                                              Peer Address:Port                      
u_str                    ESTAB                      0                           0                                                 /var/run/ubus.sock 245                                                          * 0                         
u_str                    ESTAB                      0                           0                                                                  * 473                                                          * 0                         
u_str                    ESTAB                      0                           0                                                 /var/run/ubus.sock 1363                                                         * 0                         
u_str                    ESTAB                      0                           0                                                 /var/run/ubus.sock 1515                                                         * 0                         
u_str                    ESTAB                      0                           0                                                                  * 2486                                                         * 0                         
u_str                    ESTAB                      0                           0                                                                  * 2514                                                         * 0                         
u_str                    ESTAB                      0                           0                                                                  * 3197                                                         * 0                         
u_str                    ESTAB                      0                           0                                                                  * 3366                                                         * 0                         
u_str                    ESTAB                      0                           0                                                 /var/run/ubus.sock 3367                                                         * 0                         
u_str                    ESTAB                      0                           0                                                 /var/run/ubus.sock 3520                                                         * 0                         
u_str                    ESTAB                      0                           0                                                                  * 5924                                                         * 0                         
u_str                    ESTAB                      0                           0                                                                  * 5925                                                         * 0                         
u_str                    ESTAB                      0                           0                                                 /var/run/ubus.sock 600750                                                       * 0                         
u_str                    ESTAB                      0                           0                                                                  * 601742                                                       * 0                         
u_dgr                    ESTAB                      0                           0                                                           /dev/log 471                                                          * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 1576                                                         * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 1595                                                         * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 2445                                                         * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 2484                                                         * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 3188                                                         * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 5505                                                         * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 5929                                                         * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 565769                                                       * 0                         
u_dgr                    ESTAB                      0                           0                                                                  * 601735                                                       * 0                         
tcp                      ESTAB                      0                           0                                                    217.107.210.186:56580                                            99.144.24.156:https                     
tcp                      ESTAB                      0                           0                                                       192.168.1.10:https                                            192.168.1.100:45174                     
tcp                      ESTAB                      0                           0                                                    217.107.210.186:56212                                             88.112.99.68:https                     
tcp                      ESTAB                      0                           0                                                       192.168.1.10:https                                            192.168.1.100:45172                     
tcp                      ESTAB                      0                           0                                                    217.107.210.186:38370                                              71.13.64.51:https
```
Strace shows `socket(AF_NETLINK, SOCK_RAW|SOCK_CLOEXEC, NETLINK_SOCK_DIAG) = -1 EPROTONOSUPPORT (Protocol not supported)` errors.
Full strace log - https://paste.ubuntu.com/p/JnhWCNCZsH/

neheb, 2019-03-30:
should be @russellsenior

Strykar, 2019-06-08:
@RussellSenior Any update on this?

neheb, 2020-01-21:
The problem is here:

```
socket(AF_NETLINK, SOCK_RAW, NETLINK_SOCK_DIAG)
```

I tried replacing SOCK_RAW with SOCK_DGRAM but same error.

From reading the man page, it would appear that NETLINK_SOCK_DIAG is the issue. No idea what to replace with though.

neheb, 2020-01-21:
Found the issue. It's a kernel problem.

You need to add CONFIG_NETLINK_DIAG=y to your kernel config.

On desktop distros, this is a module loaded dynamically.

Strykar, 2020-01-21:
@neheb Nice! So I guess the Q now is if OpenWrt should enable that as a kernel default?

neheb, 2020-01-21:
Unlikely. That's partially what the debloat patch you linked to does.

Maybe create a kernel module package and have ss depend on that.

Strykar, 2020-01-21:
@neheb Perhaps @RussellSenior should see this thread then.

Strykar, 2020-01-22:
@neheb Correct me if I'm wrong, but isn't that just a config option for the module named `kmod-nfnetlink` and not a module on its own?

neheb, 2020-01-22:
Dunno.

If you install that, does ss work properly?

neheb, 2020-01-22:
Ah there's also a kmod-nfnetlink-queue. That one must be it.

Strykar, 2020-01-23:
@neheb I already had that installed and that does not resolve it. Did you try to re-compile the module with `CONFIG_NETLINK_DIAG=y`? I did, but opkg will not install it from `./` and just downloads the standard version instead.

neheb, 2020-01-23:
On further inspection, I don't think that module is the answer. I think a new one has to be specified.

Or run a custom kernel with CONFIG_NETLINK_DIAG=y

Strykar, 2020-01-23:
So it's not a module? I did re-compile the kernel with that option.

neheb, 2020-01-23:
Well, the object is netlink_diag.o , which can be built as a module.

I was referring to kmod-nfnetlink-queue. That one is unrelated.

Strykar, 2020-01-23:
Hmm, `targets/x86/64/packages/` only has one module named netlink called `kmod-nfnetlink_4.14.151-1_x86_64.ipk`. So where is the module located after being built?

neheb, 2020-01-24:
There's no kmod-* package for it. One has to be specified and built. Maybe in package/kernel/linux/modules/netfilter.mk

This patch as mentioned above makes it work:

```
--- a/target/linux/generic/config-4.19
+++ b/target/linux/generic/config-4.19
@@ -3164,7 +3164,7 @@ CONFIG_NETDEVICES=y
 # CONFIG_NETFILTER_XT_TARGET_TEE is not set
 # CONFIG_NETFILTER_XT_TARGET_TPROXY is not set
 # CONFIG_NETFILTER_XT_TARGET_TRACE is not set
-# CONFIG_NETLINK_DIAG is not set
+CONFIG_NETLINK_DIAG=y
 # CONFIG_NETLINK_MMAP is not set
 # CONFIG_NETPOLL is not set
 # CONFIG_NETROM is not set
```

Anyway, I will close this. As you mentioned, this must go in https://bugs.openwrt.org/ .

closed 2020-01-24 by neheb

RussellSenior, 2020-01-24:
fwiw, gmail buried my notifications on this. sorry for the awol.

Strykar, 2020-01-26:
@neheb Thanks, so reompile with that enabled in the config and then reinstall the kernel?
@RussellSenior No worries, I will file a bug at https://bugs.openwrt.org

neheb, 2020-01-26:
Maybe something like this would work. I have no idea:

```
--- a/package/kernel/linux/modules/netsupport.mk
+++ b/package/kernel/linux/modules/netsupport.mk
@@ -1231,3 +1231,18 @@ define KernelPackage/macsec/description
 endef
 
 $(eval $(call KernelPackage,macsec))
+
+
+define KernelPackage/netlink-diag
+  SUBMENU:=$(NETWORK_SUPPORT_MENU)
+  TITLE:=Netlink diag support for ss utility
+  KCONFIG:=CONFIG_NETLINK_DIAG
+  FILES:=$(LINUX_DIR)/net/netlink/netlink-diag.ko
+  AUTOLOAD:=$(call AutoLoad,31,netlink-diag)
+endef
+
+define KernelPackage/netlink-diag/description
+ Netlink diag is a module made for use with iproute2's ss utility
+endef
+
+$(eval $(call KernelPackage,netlink-diag))
```

neheb, 2020-04-08:
Try latest master.

Strykar, 2020-12-26:
> Try latest master.

@neheb
Sadly I do not have a device running master right now, I did try on `OpenWrt 19.07.4 r11208-ce6496d796` and the issue still exists.
I'm going to assume until there's a `kmod-netlink-diag` package available, this issue will persist?

neheb, 2020-12-26:
I think this was fixed after 19.07. master should work.