openwrt/packages #8234: LVM2: Syntax error: Unterminated quoted string

DocMAX, 2019-02-16:
i cant get around this error while compiling latest openwrt, any ideas?:
(i did a make dirclean...)

```
make[3]: Entering directory '/mnt/t420/hdd/software/linux/sources/openwrt/feeds/packages/utils/lvm2'
rm -f /mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02/.configured_*                                                                                                    
rm -f /mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/target-mips_24kc_musl/stamp/.lvm2_installed                                                                                                       
(cd /mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02/./; if [ -x ./configure ]; then find /mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02/ -name config.guess | xargs -r chmod u+w; find /mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02/ -name config.guess | xargs -r -n1 cp --remove-destination /mnt/t420/hdd/software/linux/sources/openwrt/scripts/config.guess; find /mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02/ -name config.sub | xargs -r chmod u+w; find /mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02/ -name config.sub | xargs -r -n1 cp --remove-destination /mnt/t420/hdd/software/linux/sources/openwrt/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02:LVM2.2.03.02 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02:LVM2.2.03.02 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02:LVM2.2.03.02 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/target-mips_24kc_musl/include -I/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/include -I/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include/fortify -I/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/include " LDFLAGS="-L/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/target-mips_24kc_musl/lib -L/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/usr/lib -L/mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.4.0_musl/lib -znow -zrelro "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --disable-o_direct --with-default-pid-dir=/var/run --with-default-dm-run-dir=/var/run --with-default-run-dir=/var/run/lvm --with-default-locking-dir=/var/lock/lvm ; fi; )
configure: loading site script /mnt/t420/hdd/software/linux/sources/openwrt/include/site/mips
checking build system type... x86_64-pc-linux-gnu
checking host system type... mips-openwrt-linux-gnu
checking target system type... mips-openwrt-linux-gnu
./configure: 2: ./configure: Syntax error: Unterminated quoted string
make[3]: *** [Makefile:105: /mnt/t420/hdd/software/linux/sources/openwrt/build_dir/target-mips_24kc_musl/LVM2.2.03.02/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2                                      
make[3]: Leaving directory '/mnt/t420/hdd/software/linux/sources/openwrt/feeds/packages/utils/lvm2'
time: package/feeds/packages/lvm2/compile#0.19#0.16#0.68
make[2]: *** [package/Makefile:109: package/feeds/packages/lvm2/compile] Error 2
make[2]: Leaving directory '/mnt/t420/hdd/software/linux/sources/openwrt'
make[1]: *** [package/Makefile:103: /mnt/t420/hdd/software/linux/sources/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2                                                               
make[1]: Leaving directory '/mnt/t420/hdd/software/linux/sources/openwrt'
make: *** [/mnt/t420/hdd/software/linux/sources/openwrt/include/toplevel.mk:218: world] Error 2

```

DocMAX, 2019-02-18:
Sorry, the problem is in the wolfssl package, moving there.

DocMAX, 2019-02-21:
strange, on arch linux it compiles well. never mind, i will close.

closed 2019-02-21 by DocMAX

jbbr, 2019-08-23:
The issue seems to be caused by some scripts, which use shell alias `/bin/sh` but are not POSIX compatible (rely on non POSIX Shell-Features, e.g bash).
Some distros like Arch use bash as shell for `/bin/sh`. Others like Ubuntu use more lightweight shells like `dash`.
Workaround: Change symlink of sh to bash.