openwrt/packages #8236: gcc: does not support generating statically-linked programs

MikePetullo, 2019-02-16:
Maintainer: @noblepepper
Environment: x86_64 master

Description:
It does not appear that any OpenWrt package provides libc.a, and thus GCC cannot statically link programs when run on OpenWrt. I have a requirement to build statically-linked programs using a GCC toolchain on an OpenWrt system. Would it be possible to create a gcc-static or musl-static package? It does seem that the musl build process creates libc.a, but it is not included in the musl (libc) package.

I am a little confused whether this should be filed against gcc or musl: musl owns /lib/libc.so, but gcc owns /usr/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/libc.so.

noblepepper, 2019-02-19:
It seems to me it is an issue with gcc since libc.a isn't needed otherwise.

Yes, static linking is not supported out of the box, this is noted in the package description:
To save disk space, this GCC only supports dynamic linking on the target box,
there are no static libraries shipped except libgcc.a on those architectures
that need it.

I'll look at including libc.a in gcc, in the world of cheap multi gig sdcards/thumb drives space isn't nearly so precious as 5 years ago.

If you have built openwrt locally you can copy libc.a from the host to the target and static link. For me it is found in staging_dir/target-mips_24kc_musl/root-ar71xx/lib/libc.a on the host and should be put in /usr/lib/gcc/mips-openwrt-linux-musl/5.4.0 on the target.

Example:
root@OpenWrt:~# ls -lh /usr/lib/gcc/mips-openwrt-linux-musl/5.4.0/libc.a
-rw-r--r--    1 root     root        7.9M Feb 19 19:18 /usr/lib/gcc/mips-openwrt-linux-musl/5.4.0/libc.a
root@OpenWrt:~# gcc hello.c -o hello
root@OpenWrt:~# gcc --static hello.c -o hellostatic
root@OpenWrt:~# ls -lh
-rwxr-xr-x    1 root     root        5.6K Feb 19 19:36 hello
-rw-r--r--    1 root     root          72 Feb 19 18:57 hello.c
-rwxr-xr-x    1 root     root       82.7K Feb 19 19:36 hellostatic
root@OpenWrt:~# ./hello
hello world 
root@OpenWrt:~# ./hellostatic 
hello world 
root@OpenWrt:~#

MikePetullo, 2019-02-19:
Yes, I had done exactly what you suggested: manually copy in libc.a, and it did and does indeed work. I am still interested in a "gcc-static" package, primarily to benefit from package updates but also to keep things cleaner. I think wrapping libc.a in an optional and separate would satisfy both people with small devices and those of us with a bit more room.

noblepepper, 2019-02-24:
I need to work on updating this package to match the toolchain version. I'll see about adding something for static linking in that pull request. I believe fixes to this version will find resistance from the main body of developers.

artynet, 2019-02-24:
As far as it can help you, I tried to cross-build an updated GCC compiler but I bumped in some static linking error (due to the cross-toolchain itself). Please find my open issue to get some pointers about it....best, Arturo

Shulyaka, 2019-04-10:
What about other static libraries? Some of them are used really often (such as libpthread.a) and some others not so often. Would it worth including them as well in the gcc package?

neheb, 2020-01-12:
GCC7 was merged. It should fix this.

closed 2020-01-12 by neheb

fishfish1337, 2023-04-01:
This is still an issue. Is there really no way to build a static binary without compiling all of OpenWRT ourselves and copying over the libc.a file?