openwrt/packages #8238: perl: spurious build failures on openwrt-18.06 branch

mweinelt, 2019-02-18:
Maintainer: @pprindeville
Environment: 
- Build Host: Debian 9.8 (x86_64)
- Build Target: brcm2708-bcm2710
- OpenWrt https://github.com/openwrt/openwrt/commit/62feabecd8a84cfe0d6a5aa3bad2720dbf328370
- packages.git https://github.com/openwrt/packages/commit/d05b98c6c86da58db5cbda3c945007be09583609

Description:
I'm regularly building OpenWrt 18.06 through the Gluon mesh framework. Building Perl spuriously fails every once in a while. I managed to capture a build log this time.

```
hexa@build:/scratch/hexa/build/gluon$ cat openwrt/logs/package/feeds/packages/perl/compile.txt
make[4]: Entering directory '/scratch/hexa/build/gluon/packages/packages/lang/perl'
touch /scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/.prepared_2b8d7106f2b8824c085fd46b0bcfa6eb_6664517399ebbbc92a37c5bb081b5c53_check
. /scratch/hexa/build/gluon/openwrt/include/shell.sh; xzcat /scratch/hexa/build/gluon/openwrt/dl/perl-5.28.0.tar.xz | tar -C /scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0

Applying ./patches/010-musl-compat.patch using plaintext: 
File perl.c is read-only; trying to patch anyway
patching file perl.c
Hunk #1 succeeded at 303 (offset 17 lines).

Applying ./patches/020-storables-stacksize.patch using plaintext: 
File dist/Storable/Makefile.PL is read-only; trying to patch anyway
patching file dist/Storable/Makefile.PL
File dist/Storable/stacksize is read-only; trying to patch anyway
patching file dist/Storable/stacksize

Applying ./patches/100-fix-cross-compile-endianness-detection.patch using plaintext: 
File config_h.SH is read-only; trying to patch anyway
patching file config_h.SH

Applying ./patches/110-always_use_miniperl.patch using plaintext: 
File Makefile.SH is read-only; trying to patch anyway
patching file Makefile.SH
Hunk #1 succeeded at 339 (offset 5 lines).

Applying ./patches/120-remove-build-timestamp.patch using plaintext: 
File perl.c is read-only; trying to patch anyway
patching file perl.c
Hunk #1 succeeded at 1987 (offset 109 lines).

Applying ./patches/300-add-relink-hack.patch using plaintext: 
patching file relink/Makefile.PL

Applying ./patches/320-copy-pod-hack.patch using plaintext: 
File cpan/podlators/Makefile.PL is read-only; trying to patch anyway
patching file cpan/podlators/Makefile.PL

Applying ./patches/710-threads_join-skip_ps_on_busybox.patch using plaintext: 
File dist/threads/t/join.t is read-only; trying to patch anyway
patching file dist/threads/t/join.t

Applying ./patches/900-use-rm-force.patch using plaintext: 
File Makefile.SH is read-only; trying to patch anyway
patching file Makefile.SH

Applying ./patches/910-miniperl-needs-inc-dot.patch using plaintext: 
File Makefile.SH is read-only; trying to patch anyway
patching file Makefile.SH
Hunk #1 succeeded at 333 (offset 5 lines).
Hunk #2 succeeded at 996 (offset 5 lines).
Hunk #3 succeeded at 1007 (offset 5 lines).

Applying ./patches/920-fix-no-locale.patch using plaintext: 
File embed.fnc is read-only; trying to patch anyway
patching file embed.fnc
Hunk #1 succeeded at 2793 (offset 28 lines).
patching file embed.h
Hunk #1 succeeded at 1796 (offset 72 lines).
[... 324 lines cut ...]
Writing Makefile for Archive::Tar
Writing MYMETA.yml and MYMETA.json
make[6]: Entering directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/cpan/Archive-Tar'
"../../../../../../staging_dir/hostpkg/usr/bin/perl" "-I../../lib" Errno_pm.PL Errno.pm
Generating a Unix-style Makefile
Writing Makefile for ExtUtils::ParseXS
Writing MYMETA.yml and MYMETA.json
make[6]: Entering directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/dist/ExtUtils-ParseXS'
make[6]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/cpan/CPAN-Meta'
Generating a Unix-style Makefile
Writing Makefile for HTTP::Tiny
Writing MYMETA.yml and MYMETA.json
make[6]: Entering directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/cpan/HTTP-Tiny'
make[6]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/cpan/Archive-Tar'
make[6]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/cpan/ExtUtils-MakeMaker'
make[6]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/cpan/ExtUtils-Manifest'
make[6]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/dist/ExtUtils-ParseXS'
make[6]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/cpan/HTTP-Tiny'
make[6]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/ext/Errno'
make[5]: Leaving directory '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0'
Makefile:143: recipe for target '/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/.built' failed
make[4]: *** [/scratch/hexa/build/gluon/openwrt/build_dir/target-aarch64_cortex-a53_musl/perl/perl-5.28.0/.built] Error 2
make[4]: Leaving directory '/scratch/hexa/build/gluon/packages/packages/lang/perl'
time: package/feeds/packages/perl/compile#162.40#11.91#30.32
```

mweinelt, 2019-06-10:
5d3cb594e49132158ca7b41f4246421078e8f92a

brcm2708-bcm2708
https://gist.github.com/mweinelt/9bec3e1559b94bd34f683c120a4c0761

mweinelt, 2020-04-11:
Still failing on

mweinelt, 2020-04-11:
Still failing on 619961596091449e8c95d98e591e02745b086a71.
https://github.com/freifunk-gluon/gluon/runs/578339551?check_suite_focus=true

Meanwhile I've disabled parallel builds of perl and I've yet to encounter this build failure again:

```
From: Martin Weinelt <martin@darmstadt.freifunk.net>
Date: Wed, 19 Jun 2019 01:28:44 +0200
Subject: parallel: don't build in parallel

diff --git a/lang/perl/Makefile b/lang/perl/Makefile
index b92b00279631ae07a0f23b063571d97eb8ed3efc..81dbfcdd5bd6cad92b9552f491b200fc85fd0084 100644
--- a/lang/perl/Makefile
+++ b/lang/perl/Makefile
@@ -33,8 +33,8 @@ PKG_BUILD_DIR:=$(BUILD_DIR)/perl/$(PKG_NAME)-$(PKG_VERSION)
 HOST_BUILD_DIR:=$(BUILD_DIR_HOST)/perl/$(PKG_NAME)-$(PKG_VERSION)
 PKG_INSTALL:=1
 PKG_BUILD_DEPENDS:=perl/host
-PKG_BUILD_PARALLEL:=1
-HOST_BUILD_PARALLEL:=1
+PKG_BUILD_PARALLEL:=0
+HOST_BUILD_PARALLEL:=0
 
 # Variables used during configuration/build
 HOST_PERL_PREFIX:=$(STAGING_DIR_HOSTPKG)/usr
```

pprindeville, 2020-05-25:
Odd that this isn't a problem with the nightlies.

pprindeville, 2021-02-16:
Is this still going on?  Do you have a root-cause?

mweinelt, 2021-02-16:
It's spurious, so it will take me a while to reproduce, as I've used said patch for the last year or so.

mweinelt, 2021-02-16:
Yep, still failing. Attaching a proper log from the package build.
[compile.txt](https://github.com/openwrt/packages/files/5992022/compile.txt)

blocktrron, 2021-08-14:
@pprindeville Any objections against merging this? This issue has been present for quite some time but never occurred with this fix.

mweinelt, 2021-09-02:
Can still reproduce this :tada: and would appreciate a decision.

mweinelt, 2021-12-05:
It's finally here! :)

closed 2024-07-28 by pprindeville