openwrt/packages #8256: Build failure in stable openwrt-18.06 branch caused by jool (never added).

mcarlsonO, 2019-02-19:
> Somehow I still face this issue on the `openwrt-18.06` branch (beyond the `18.06.2` release tag):
> 
> ```
> make[3]: Entering directory '/home/user/openwrt/feeds/packages/net/jool'
> rm -f /home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/.built
> touch /home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/.built_check
> make -C "/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.97" KERNEL_DIR="/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.97" ARCH="arm" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" SUBDIRS="/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful" V="s" modules
> make[4]: Entering directory '/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.97'
>   CC [M]  /home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.o
> /home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c: In function 'inet6_hash_frag':
> /home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c:96:14: error: 'INETFRAGS_HASHSZ' undeclared (first use in this function); did you mean 'INET_FRAG_LAST_IN'?
>   return c & (INETFRAGS_HASHSZ - 1);
>               ^~~~~~~~~~~~~~~~
>               INET_FRAG_LAST_IN
> /home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c:96:14: note: each undeclared identifier is reported only once for each function it appears in
> /home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c:97:1: warning: control reaches end of non-void function [-Wreturn-type]
>  }
>  ^
> scripts/Makefile.build:326: recipe for target '/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.o' failed
> make[5]: *** [/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.o] Error 1
> Makefile:1533: recipe for target '_module_/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful' failed
> make[4]: *** [_module_/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful] Error 2
> make[4]: Leaving directory '/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.14.97'
> Makefile:113: recipe for target '/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/.built' failed
> make[3]: *** [/home/user/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/.built] Error 2
> make[3]: Leaving directory '/home/user/openwrt/feeds/packages/net/jool'
> time: package/feeds/packages/jool/compile#1.29#0.37#1.66
> package/Makefile:107: recipe for target 'package/feeds/packages/jool/compile' failed
> make[2]: *** [package/feeds/packages/jool/compile] Error 2
> make[2]: Leaving directory '/home/user/openwrt'
> package/Makefile:103: recipe for target '/home/user/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
> make[1]: *** [/home/user/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
> make[1]: Leaving directory '/home/user/openwrt'
> /home/user/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
> make: *** [world] Error 2
> ```
> 
> Stranger still is that it happens with the following configuration options:
> 
> ```
> CONFIG_PACKAGE_kmod-jool=m
> # CONFIG_PACKAGE_jool is not set
> # CONFIG_PACKAGE_jool-tools is not set
> ```

```
make[3]: Entering directory '/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/feeds/packages/net/jool'
rm -f /home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/.built
touch /home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/.built_check
make -C "/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.156" KERNEL_DIR="/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.156" ARCH="mips" CROSS_COMPILE="mips-openwrt-linux-musl-" SUBDIRS="/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful" V="sc" modules
make[4]: Entering directory '/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.156'
  CC [M]  /home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.o
/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c: In function 'inet6_hash_frag':
/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c:96:14: error: 'INETFRAGS_HASHSZ' undeclared (first use in this function); did you mean 'INET_FRAG_LAST_IN'?
  return c & (INETFRAGS_HASHSZ - 1);
              ^~~~~~~~~~~~~~~~
              INET_FRAG_LAST_IN
/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c:96:14: note: each undeclared identifier is reported only once for each function it appears in
/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.c:97:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
scripts/Makefile.build:304: recipe for target '/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.o' failed
make[5]: *** [/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful/fragment_db.o] Error 1
Makefile:1528: recipe for target '_module_/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful' failed
make[4]: *** [_module_/home/mcarlson/OPENWRT/build_trees/ar71xx-generic-wzr-hp-ag300h/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/jool-9dfaf22e49f7905d94af9b73f9bee22c26d7dd4a/mod/stateful] Error 2
```
I am seeing the same issue on the stable openwrt-18.06 branch (updated a hour ago) with a nearly default config.
```
CONFIG_PACKAGE_kmod-jool=m
# CONFIG_PACKAGE_jool is not set
# CONFIG_PACKAGE_jool-tools is not set
```
So I suppose the solution here is to forcibly disable this thing?

_Originally posted by @mcarlsonO in https://github.com/openwrt/packages/issues/7095#issuecomment-465135475_

briskycat, 2019-02-20:
Yes, if you disable `CONFIG_PACKAGE_kmod-jool=m` (comment it out) everything should build just fine.

mcarlsonO, 2019-02-20:
> Yes, if you disable `CONFIG_PACKAGE_kmod-jool=m` (comment it out) everything should build just fine.

Is this documented in the development docs somewhere?

neheb, 2019-03-26:
https://github.com/openwrt/packages/pull/7807

neheb, 2019-03-30:
merged. closing.

closed 2019-03-30 by neheb