openwrt/packages #8260: netdata - 1.10.0-1

socomsystems, 2019-02-19:
Model:  MikroTik RBM33G
Architecture:  MediaTek MT7621 ver:1 eco:3

Firmware Version:  
OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152)

Kernel Version	4.14.63
Local Time:  Tue Feb 19 15:08:37 2019
Uptime:  5d 11h 3m 8s
Load Average:  1.05, 0.83, 0.63

Description:
```
[39584.399313] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-3: link becomes ready
[39584.406334] br-VLAN6_UNT: port 2(wlan0-3) entered blocking state
[39584.412484] br-VLAN6_UNT: port 2(wlan0-3) entered forwarding state
[39584.427675] br-VLAN4_UNT: port 2(wlan0-4) entered blocking state
[39584.433711] br-VLAN4_UNT: port 2(wlan0-4) entered disabled state
[39584.440823] device wlan0-4 entered promiscuous mode
[39584.446742] IPv6: ADDRCONF(NETDEV_UP): wlan0-4: link is not ready
[39584.453436] br-VLAN4_UNT: port 2(wlan0-4) entered blocking state
[39584.459543] br-VLAN4_UNT: port 2(wlan0-4) entered forwarding state
[39584.807900] br-VLAN4_UNT: port 2(wlan0-4) entered disabled state
[39585.065030] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-4: link becomes ready
[39585.072265] br-VLAN4_UNT: port 2(wlan0-4) entered blocking state
[39585.078351] br-VLAN4_UNT: port 2(wlan0-4) entered forwarding state
[119202.231737] mtk_soc_eth 1e100000.ethernet eth0: port 0 link down
[119251.709360] mtk_soc_eth 1e100000.ethernet eth0: port 0 link up
[119312.171430] mtk_soc_eth 1e100000.ethernet eth0: port 0 link down
[119331.686017] mtk_soc_eth 1e100000.ethernet eth0: port 0 link up
[390721.153731] do_page_fault(): sending SIGSEGV to netdata for invalid write access to 77dbed2c
[390721.162437] epc = 77f3ee20 in libc.so[77ec9000+92000]
[390721.167949] ra  = 0043c6fd in netdata[400000+95000]
[424279.087186] ------------[ cut here ]------------
[424279.091903] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:320 0x8036d4f8
[424279.099063] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[424279.106080] Modules linked in: ath9k ath9k_common pppoe ppp_async ath9k_hw ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY wireguard slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route
[424279.176909]  cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb ip6_udp_tunnel udp_tunnel leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[424279.202721] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.63 #0
[424279.208788] Stack : 00000006 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[424279.217210]         00000000 00000000 00000000 00000000 00000000 00000001 8fc0dd60 ac07f526
[424279.225628]         8fc0ddf8 00000000 00000000 0000fbb8 00000038 80476918 00000008 00000000
[424279.234046]         00000000 80520000 000317e1 70617773 8fc0dd40 00000000 80540000 8036d4f8
[424279.242469]         00000009 00000140 00000001 8fe56e40 00000000 80290678 00000004 80580004
[424279.250892]         ...
[424279.253420] Call Trace:
[424279.253438] [<80476918>] 0x80476918
[424279.259510] [<8036d4f8>] 0x8036d4f8
[424279.263077] [<80290678>] 0x80290678
[424279.266643] [<80010040>] 0x80010040
[424279.270203] [<80010048>] 0x80010048
[424279.273758] [<8045f8cc>] 0x8045f8cc
[424279.277316] [<8006f7b0>] 0x8006f7b0
[424279.280874] [<8002e388>] 0x8002e388
[424279.284434] [<8036d4f8>] 0x8036d4f8
[424279.287993] [<8002e3ec>] 0x8002e3ec
[424279.291549] [<8005440c>] 0x8005440c
[424279.295104] [<8036d4f8>] 0x8036d4f8
[424279.298661] [<80096d20>] 0x80096d20
[424279.302216] [<8036d34c>] 0x8036d34c
[424279.305773] [<80086354>] 0x80086354
[424279.309333] [<8008656c>] 0x8008656c
[424279.312889] [<80076b90>] 0x80076b90
[424279.316490] [<8047ce10>] 0x8047ce10
[424279.320050] [<80032ab4>] 0x80032ab4
[424279.323607] [<802491cc>] 0x802491cc
[424279.327171] [<8000b4e8>] 0x8000b4e8
[424279.330729] 
[424279.332386] ---[ end trace a0774e3f687ef3ea ]---
[424279.337088] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[424279.343361] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[424279.349486] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ea30000, max=0, ctx=2047, dtx=2047, fdx=2046, next=2047
[424279.360428] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e1f0000, max=0, calc=146, drx=147
[424279.372568] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[424279.385183] mtk_soc_eth 1e100000.ethernet: PPE started ```
```

diizzyy, 2019-02-19:
@socomsystems 
I can say with high probability that it's not going to be looked into for several reasons:

* Netdata has been bumped two times already (it's at 1.11.1 right now and there's a PR for 1.12.0)
* Several libraries such as libc (musl) has been bumped in master
* Half of your paste is related to a kernel/driver/subsystem crash which is not related to netdata pointing towards general system instability.
* There's little to no interest in debugging something that's already obsolete

Please give at a go with OpenWrt master (head/trunk) and the current package (head/trunk) by compiling it yourself or use the generated snapshots.

@BKPepe 
Anything you want to add?

BKPepe, 2019-02-19:
Thank you @diizzyy. You said everything.

socomsystems, 2019-02-19:
Thanks @diizzyy and @BKPepe for the direction.  I see theirs a busybox binary build.  PS, the maintainers link is 404 here:  https://docs.netdata.cloud/packaging/maintainers/

BKPepe, 2019-02-19:
Good catch. It was working a few days ago.  In the meantime, you can look [here](https://github.com/netdata/netdata/tree/master/packaging/maintainers). I will see what I can do about it.

socomsystems, 2019-02-21:
Hi @BKPepe and @diizzyy please excuse my ignorance regarding the following questions. I'm running OpenWRT 18.06.1 on an RMB33G board. The most up to date netdata package available I see using the following repos is netdata - 1.10.0-1. Of which I seem to be having page faults with also noted below. Does your 1.11.1 patching address such an issue and/or how would I go about obtaining your patches? I do see OpenWRT 18.06.2 addresses several issues with the MT7621 architecture as I'm running on MediaTek MT7621 ver:1 eco:3 - is this issue related and/or should I upgrade firmware? I greatly appreciate any advice or direction and if need be I'd like to start participating on getting NetData up to speed as much as I can. I've been researching setting up build environments etc... which will come in time.

diizzyy, 2019-02-22:
Without recompiling you need to upgrade to the current snapshots, these are moving targets meaning that you need to install everything you need in one sweep otherwise you'll get desynced packages.

socomsystems, 2019-02-22:
@diizzyy just so I'm clear, if I'm understanding your correctly I need to flash "sysupgrade" from 18.06.1 to 18.06.2, then opkg upgrade for netdata will be available which will bring it to 1.11.1?  Thank you for the feedback.

diizzyy, 2019-02-22:
@socomsystems 
No, that'll give you 1.10.0.
You need to either use snapshots (usually works fine but may occasionally have nasty bugs) or compile your own packages with 18.06.2 as base.

socomsystems, 2019-02-25:
Thanks for the advisory @diizzyy ... when will the new netdata package be committed to the stable 18.06.2 release?

diizzyy, 2019-02-25:
Most likely never

BKPepe, 2019-02-25:
We should update Netdata as well to OpenWrt 18.06.02 as there were some assigned CVEs and there is full backward compatibility, so there shouldn't be an issue, right? If my pull request for Netdata 1.12.1 would be accepted then I will create PR for updating it as well in OpenWrt 18.06.02, and let's see.

I read, there's planned OpenWrt 18.06.03, but there is no timeline, while there are plans and timelines for 19.03.

diizzyy, 2019-02-25:
In theory no but packages only get rebuilt upon release so might as well wait.

hnyman, 2019-03-13:
> >  there shouldn't be an issue, right? If my pull request for Netdata 1.12.1 would be accepted then I will create PR for updating it as well in OpenWrt 18.06.02, and let's see.
> > 
> > I read, there's planned OpenWrt 18.06.03, but there is no timeline, while there are plans and timelines for 19.03.

> In theory no **but packages only get rebuilt upon release** so might as well wait.

???

User-space packages are constantly rebuilt by the 18.06 buildbot, if necessary.

diizzyy, 2019-03-13:
Is that really the case? If that's true then a lot of packages would break if you'd update a library etc

hnyman, 2019-03-13:
Some are broken...
http://downloads.openwrt.org/releases/faillogs/

The recent ABI handling change in master is combating that problem, but that is not in 18.06.

diizzyy, 2019-03-13:
Wouldn't it make more sense to just do a complete clean build during releases especially since the majority of devices cant update pages in a "efficient" way?

diizzyy, 2019-03-28:
Resolved by https://github.com/openwrt/packages/pull/8402/commits/99c3670bb1d6f5b499701edb9f4105d3c7f62050

PR #8402 (2019-03-16): [OpenWrt 18.06] Netdata: update to version 1.12.2
Maintainer: none (previous @diizzyy)
Compile tested: cortexa53, Turris MOX, OpenWrt 18.06.02
Run tested: cortexa53, Turris MOX, OpenWrt 18.06.02

Description:
* update Netdata to version 1.12.2 in OpenWrt 18.06. (sync it with OpenWrt master)
* It fixes security issues:
CVE-2018-18836
CVE-2018-18837
CVE-2018-18838 
CVE-2018-18839
and a few others
* New Data Collection modules (Oracle DB, tor, unbound, monit, cups and so on)
* New feature: netdata.cloud (network view for all nodes)

_____

Closes: #8260
closed 2019-03-28 by diizzyy