openwrt/packages #8269: iw package in snaphost

mistepien, 2019-02-20:
iw-full - 5.0.1-1

> Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): Usage:	iw [options] command
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): Options:
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): 	--debug		enable netlink debugging
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): 	--version	show version (5.0.1)
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): Commands:
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): 	dev <devname> connect [-w] <SSID> [<freq in MHz>] [<bssid>] [key 0:abcde d:1:6162636465] [mfp:req/opt/no]
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): 		Join the network with the given SSID (and frequency, BSSID).
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): 		With -w, wait for the connect to finish or fail.
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202):
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202):
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): Commands that use the netdev ('dev') can also be given the
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): 'wdev' instead to identify the device.
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202):
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): You can omit the 'phy' or 'dev' if the identification is unique,
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): e.g. "iw wlan0 info" or "iw phy0 info". (Don't when scripting.)
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202):
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202): Do NOT screenscrape this tool, we don't consider its output stable.
Thu Feb 21 00:23:22 2019 daemon.notice netifd: radio0 (1202):
Thu Feb 21 00:23:22 2019 user.notice mac80211: Failed command: iw dev wlan0 set txpower fixed 2300

diizzyy, 2019-03-14:
@hynman @thess
Close, not valid

diizzyy, 2019-03-28:
Wrong repo, use https://bugs.openwrt.org/ to report bugs for the OpenWrt main repo

closed 2019-03-28 by diizzyy