openwrt/packages #8276: Transmission problem

fuqiang03, 2019-02-22:
[image]
Unable to start under openwrt 18.06 branch

neheb, 2019-02-25:
I've seen this issue under different circumstances. Either reboot the router or run /etc/init.d/transmission start manually.

I have no idea why this happens.

fuqiang03, 2019-02-26:
Upgrade to the new "Luci" problem
The SHELL file needs to be modified
I replaced it with someone else's source code.

fuqiang03, 2019-02-26:
[image]
[image]

diizzyy, 2019-03-14:
@hnyman @thess 
Please close

closed 2019-03-14 by hnyman