openwrt/packages #8277: ddns-scripts: Option bind_interface (Bind Network) not working with wget (default)

amialkow, 2019-02-22:
Maintainer: @chris5560
Package: ddns-scripts, Version: 2.7.8-1
Environment: OpenWrt 18.06.2, r7676-cddd7b4c77, Architecture: arm_xscale & brcm_47xx

Description:
I'm using ddns in multi-wan environment. Since both wan links are going through additional NAT I have to use web method to detect public IP. My objective of using bind_interface is to keep DNS name(s) assigned to specific interface. In default configuration (ddns + wget) I observed being updated with IP from wrong interface.

Per my debug it looks that wget-ssl --bind-address continues to use default route/mwan routing policy. Some sources I seen suggest this is expected. I observe different behavior with curl --interface. 
Please see experiment below:
ifconfig eth1
... inet addr:192.168.112.129
ifconfig wlan0
... inet addr:192.168.2.174
wget-ssl -q -O- --bind-addres 192.168.2.174 'https://domains.google.com/checkip'
188.146....
curl --interface wlan0 'https://domains.google.com/checkip'
89.69.... **// this is correct one**
curl --interface eth1 'https://domains.google.com/checkip'
188.146.... **// this is my default route address**
Having curl installed and selected to use solves problem. There is small challenge on second:
- ddns-scripts defaults to wget-ssl
- global option use_curl (uci set ddns.global.use_curl=1) is not easy to find.
- 18.06.2 specifically has also #6747 that requires presence of wget to have ssl option available in gui.

Option use_curl=1 is working fine for me, it just takes time figure it out. Not sure what could be done on scripts (wget?) vs. luci-app side to make this easier for others.

Configuration:
ddns-scripts 2.7.8-1
luci-app-ddns 2.4.9-3
wget 1.19.5-1
mwan3 2.6.18-1
curl 7.60.0-3

/etc/config/ddns:
config ddns 'global'
	option ddns_dateformat '%F %R'
	option ddns_loglines '250'
	option upd_privateip '0'
	option use_curl '1'

config service 'foo'
	option lookup_host 'foo.spdns.eu'
	option domain 'foo.spdns.eu'
	option username 'foo.spdns.eu'
	option ip_source 'web'
	option service_name 'spdyn.de'
	option check_interval '15'
	option use_https '1'
	option password '...'
	option interface 'wwan'
	option bind_network 'wwan'
	option ip_url 'https://domains.google.com/checkip'
	option enabled '1'

plntyk, 2021-03-12:
not much that can be done here

added a section in ddns wiki article under Issues about it

wget2 has bind-interface option since [this commit](https://gitlab.com/gnuwget/wget2/-/commit/91bba3f697d53643b151b4394fda7f1ea5ac371d) but it isnt released yet

plntyk, 2021-03-12:
https://openwrt.org/docs/guide-user/base-system/ddns#issues_and_known_bugs