openwrt/packages #8278: tor: Build failure in current

ratsputin, 2019-02-22:
Maintainer: @ja-pa 
Environment: ARMv7 Processor rev 1 (v7l) WRT3200ACM -- OpenWRT current pull

Description:
Build failure presumably related to the upgrade of OpenSSL impacting tor.  Has been broken for at least a couple of weeks.

```
  CC       src/lib/crypt_ops/src_lib_libtor_crypt_ops_a-crypto_hkdf.o
src/lib/crypt_ops/crypto_hkdf.c: In function 'crypto_expand_key_material_rfc5869_sha256_openssl':
src/lib/crypt_ops/crypto_hkdf.c:84:3: error: unknown type name 'EVP_PKEY_CTX'; did you mean 'EVP_PKEY_CTRL_PASS'?
   EVP_PKEY_CTX *evp_pkey_ctx = EVP_PKEY_CTX_new_id(EVP_PKEY_HKDF, NULL);
   ^~~~~~~~~~~~
   EVP_PKEY_CTRL_PASS
src/lib/crypt_ops/crypto_hkdf.c:84:32: warning: implicit declaration of function 'EVP_PKEY_CTX_new_id'; did you mean 'EVP_PKEY_CTX_hkdf_mode'? [-Wimplicit-function-declaration]
   EVP_PKEY_CTX *evp_pkey_ctx = EVP_PKEY_CTX_new_id(EVP_PKEY_HKDF, NULL);
                                ^~~~~~~~~~~~~~~~~~~
                                EVP_PKEY_CTX_hkdf_mode
src/lib/crypt_ops/crypto_hkdf.c:84:32: warning: nested extern declaration of 'EVP_PKEY_CTX_new_id' [-Wnested-externs]
src/lib/crypt_ops/crypto_hkdf.c:84:52: error: 'EVP_PKEY_HKDF' undeclared (first use in this function); did you mean 'EVP_PKEY_CTRL_HKDF_MD'?
   EVP_PKEY_CTX *evp_pkey_ctx = EVP_PKEY_CTX_new_id(EVP_PKEY_HKDF, NULL);
                                                    ^~~~~~~~~~~~~
                                                    EVP_PKEY_CTRL_HKDF_MD
src/lib/crypt_ops/crypto_hkdf.c:84:52: note: each undeclared identifier is reported only once for each function it appears in
src/lib/crypt_ops/crypto_hkdf.c:89:7: warning: implicit declaration of function 'EVP_PKEY_derive_init'; did you mean 'EVP_PKEY_CTRL_PASS'? [-Wimplicit-function-declaration]
   r = EVP_PKEY_derive_init(evp_pkey_ctx);
       ^~~~~~~~~~~~~~~~~~~~
       EVP_PKEY_CTRL_PASS
src/lib/crypt_ops/crypto_hkdf.c:89:7: warning: nested extern declaration of 'EVP_PKEY_derive_init' [-Wnested-externs]
In file included from src/lib/crypt_ops/crypto_hkdf.c:24:0:
src/lib/crypt_ops/crypto_hkdf.c:92:7: error: 'EVP_PKEY_OP_DERIVE' undeclared (first use in this function); did you mean 'EVP_PKEY_HKDF'?
   r = EVP_PKEY_CTX_set_hkdf_md(evp_pkey_ctx, EVP_sha256());
       ^
src/lib/crypt_ops/crypto_hkdf.c:92:7: error: 'EVP_PKEY_ALG_CTRL' undeclared (first use in this function); did you mean 'EVP_PKEY_OP_DERIVE'?
   r = EVP_PKEY_CTX_set_hkdf_md(evp_pkey_ctx, EVP_sha256());
       ^
src/lib/crypt_ops/crypto_hkdf.c:92:46: warning: implicit declaration of function 'EVP_sha256' [-Wimplicit-function-declaration]
   r = EVP_PKEY_CTX_set_hkdf_md(evp_pkey_ctx, EVP_sha256());
                                              ^
src/lib/crypt_ops/crypto_hkdf.c:92:46: warning: nested extern declaration of 'EVP_sha256' [-Wnested-externs]
src/lib/crypt_ops/crypto_hkdf.c:104:7: warning: implicit declaration of function 'EVP_PKEY_derive'; did you mean 'EVP_PKEY_CTRL_PASS'? [-Wimplicit-function-declaration]
   r = EVP_PKEY_derive(evp_pkey_ctx, key_out, &key_out_len);
       ^~~~~~~~~~~~~~~
       EVP_PKEY_CTRL_PASS
src/lib/crypt_ops/crypto_hkdf.c:104:7: warning: nested extern declaration of 'EVP_PKEY_derive' [-Wnested-externs]
src/lib/crypt_ops/crypto_hkdf.c:107:3: warning: implicit declaration of function 'EVP_PKEY_CTX_free'; did you mean 'EVP_PKEY_CTRL_PASS'? [-Wimplicit-function-declaration]
   EVP_PKEY_CTX_free(evp_pkey_ctx);
   ^~~~~~~~~~~~~~~~~
   EVP_PKEY_CTRL_PASS
src/lib/crypt_ops/crypto_hkdf.c:107:3: warning: nested extern declaration of 'EVP_PKEY_CTX_free' [-Wnested-externs]
Makefile:12168: recipe for target 'src/lib/crypt_ops/src_lib_libtor_crypt_ops_a-crypto_hkdf.o' failed
make[5]: *** [src/lib/crypt_ops/src_lib_libtor_crypt_ops_a-crypto_hkdf.o] Error 1
```

hauke, 2019-02-22:
Reading trough the changelog of version 0.3.5.8, this is probably fixed there:
https://gitweb.torproject.org/tor.git/plain/ReleaseNotes?id=tor-0.3.5.8

ja-pa, 2019-02-23:
I just compiled version  0.3.5.8 on master and failure is fixed. After some testing will push it to master. Thx for a report.

tripolar, 2019-02-23:
the update to 0.3.5.8 fixed the bug. See https://github.com/openwrt/packages/commit/03caca8e18c67b50ab52d25caddc3fefd3bf54ce

closed 2019-02-23 by tripolar