openwrt/packages #8282: python: python3-base conflicts with other packages

mittlereSchiene, 2019-02-23:
Maintainer: @commodo 
Environment: lantiq_xway, r9414-2d139450a3)

Description:
python3-base conflicts with other python packages:
```
Collected errors:
 * check_data_file_clashes: Package python3-light wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-light wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-urllib3.
 * check_data_file_clashes: Package python3-unittest wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-unittest wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-unittest.
 * check_data_file_clashes: Package python3-ncurses wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-ncurses wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-ncurses.
 * check_data_file_clashes: Package python3-ctypes wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-ctypes wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-ctypes.
 * check_data_file_clashes: Package python3-pydoc wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-pydoc wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-pydoc.
 * check_data_file_clashes: Package python3-logging wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-logging wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-logging.
 * check_data_file_clashes: Package python3-multiprocessing wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-multiprocessing wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-multiprocessing.
 * check_data_file_clashes: Package python3-codecs wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-codecs wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-codecs.
 * check_data_file_clashes: Package python3 wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3 wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-pip.
 * check_data_file_clashes: Package python3-gdbm wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-gdbm wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-gdbm.
 * check_data_file_clashes: Package python3-sqlite3 wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-sqlite3 wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-sqlite3.
 * check_data_file_clashes: Package python3-xml wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-xml wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
	But that file is already provided by package  * python3-base
 * opkg_install_cmd: Cannot install package python3-xml.
 * check_data_file_clashes: Package python3-distutils wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3
	But that file is already provided by package  * python3-base
 * check_data_file_clashes: Package python3-distutils wants to install file /home/felix/openwrt/build_dir/target-mips_24kc_musl/root-lantiq/usr/bin/python3.7
[... 750 lines cut ...]
# CONFIG_PACKAGE_libupm-vcap-python3 is not set
# CONFIG_PACKAGE_libupm-vdiv-python is not set
# CONFIG_PACKAGE_libupm-vdiv-python3 is not set
# CONFIG_PACKAGE_libupm-veml6070-python is not set
# CONFIG_PACKAGE_libupm-veml6070-python3 is not set
# CONFIG_PACKAGE_libupm-water-python is not set
# CONFIG_PACKAGE_libupm-water-python3 is not set
# CONFIG_PACKAGE_libupm-waterlevel-python is not set
# CONFIG_PACKAGE_libupm-waterlevel-python3 is not set
# CONFIG_PACKAGE_libupm-wfs-python is not set
# CONFIG_PACKAGE_libupm-wfs-python3 is not set
# CONFIG_PACKAGE_libupm-wheelencoder-python is not set
# CONFIG_PACKAGE_libupm-wheelencoder-python3 is not set
# CONFIG_PACKAGE_libupm-wt5001-python is not set
# CONFIG_PACKAGE_libupm-wt5001-python3 is not set
# CONFIG_PACKAGE_libupm-xbee-python is not set
# CONFIG_PACKAGE_libupm-xbee-python3 is not set
# CONFIG_PACKAGE_libupm-yg1006-python is not set
# CONFIG_PACKAGE_libupm-yg1006-python3 is not set
# CONFIG_PACKAGE_libupm-zfm20-python is not set
# CONFIG_PACKAGE_libupm-zfm20-python3 is not set
# CONFIG_PACKAGE_libelektra-python2 is not set
# CONFIG_PACKAGE_libelektra-python3 is not set
# CONFIG_PACKAGE_openvswitch-python is not set
```

jefferyto, 2019-02-23:
Sounds like #8263.

jefferyto, 2019-02-25:
@mittlereSchiene Can you see if this issue is still present?

mittlereSchiene, 2019-02-25:
@jefferyto 
The error above seems to be fixed, but I still have problems compiling.

Environment: lantiq_xway, r9433-f20fea9dcd)

Description:
bad interpreter

```
# Use host pip to install python-setuptools
/home/felix/openwrt_master/staging_dir/hostpkg/bin/pip3 install --ignore-installed --root=/home/felix/openwrt_master/build_dir/target-mips_24kc_musl/Python-3.7.2/install-setuptools --prefix=. /home/felix/openwrt_master/build_dir/target-mips_24kc_musl/Python-3.7.2/Lib/ensurepip/_bundled/setuptools-40.6.2-py2.py3-none-any.whl
bash: /home/felix/openwrt_master/staging_dir/hostpkg/bin/pip3: /home/felix/openwrt/staging_dir/hostpkg/bin/python3.7: bad interpreter: No such file or directory
Makefile:306: recipe for target '/home/felix/openwrt_master/build_dir/target-mips_24kc_musl/Python-3.7.2/.built' failed
make[3]: *** [/home/felix/openwrt_master/build_dir/target-mips_24kc_musl/Python-3.7.2/.built] Error 126
make[3]: Leaving directory '/home/felix/openwrt_master/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/compile#179.42#35.63#219.91
package/Makefile:107: recipe for target 'package/feeds/packages/python3/compile' failed
make[2]: *** [package/feeds/packages/python3/compile] Error 2
make[2]: Leaving directory '/home/felix/openwrt_master'
package/Makefile:103: recipe for target '/home/felix/openwrt_master/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/felix/openwrt_master/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/felix/openwrt_master'
/home/felix/openwrt_master/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

```
# CONFIG_PACKAGE_micropython is not set
# CONFIG_PACKAGE_micropython-lib is not set
# CONFIG_PACKAGE_python is not set
# CONFIG_PACKAGE_python-asn1crypto is not set
# CONFIG_PACKAGE_python-attrs is not set
# CONFIG_PACKAGE_python-attrs-src is not set
# CONFIG_PACKAGE_python-automat is not set
# CONFIG_PACKAGE_python-automat-src is not set
# CONFIG_PACKAGE_python-base is not set
# CONFIG_PACKAGE_python-base-src is not set
# CONFIG_PACKAGE_python-bcrypt is not set
# CONFIG_PACKAGE_python-bcrypt-src is not set
# CONFIG_PACKAGE_python-certifi is not set
# CONFIG_PACKAGE_python-certifi-src is not set
# CONFIG_PACKAGE_python-cffi is not set
# CONFIG_PACKAGE_python-cffi-src is not set
# CONFIG_PACKAGE_python-codecs is not set
# CONFIG_PACKAGE_python-codecs-src is not set
# CONFIG_PACKAGE_python-compiler is not set
# CONFIG_PACKAGE_python-compiler-src is not set
# CONFIG_PACKAGE_python-constantly is not set
# CONFIG_PACKAGE_python-constantly-src is not set
# CONFIG_PACKAGE_python-crcmod is not set
# CONFIG_PACKAGE_python-crypto is not set
# CONFIG_PACKAGE_python-crypto-src is not set
# CONFIG_PACKAGE_python-cryptography is not set
# CONFIG_PACKAGE_python-cryptography-src is not set
# CONFIG_PACKAGE_python-ctypes is not set
# CONFIG_PACKAGE_python-ctypes-src is not set
# CONFIG_PACKAGE_python-curl is not set
# CONFIG_PACKAGE_python-dateutil is not set
# CONFIG_PACKAGE_python-dateutil-src is not set
# CONFIG_PACKAGE_python-db is not set
# CONFIG_PACKAGE_python-db-src is not set
# CONFIG_PACKAGE_python-decimal is not set
# CONFIG_PACKAGE_python-decimal-src is not set
# CONFIG_PACKAGE_python-dev is not set
# CONFIG_PACKAGE_python-dev-src is not set
# CONFIG_PACKAGE_python-distutils is not set
# CONFIG_PACKAGE_python-distutils-src is not set
# CONFIG_PACKAGE_python-dns is not set
# CONFIG_PACKAGE_python-dpkt is not set
# CONFIG_PACKAGE_python-egenix-mx-base is not set
[... 643 lines cut ...]
# CONFIG_PACKAGE_libupm-vcap-python3 is not set
# CONFIG_PACKAGE_libupm-vdiv-python is not set
# CONFIG_PACKAGE_libupm-vdiv-python3 is not set
# CONFIG_PACKAGE_libupm-veml6070-python is not set
# CONFIG_PACKAGE_libupm-veml6070-python3 is not set
# CONFIG_PACKAGE_libupm-water-python is not set
# CONFIG_PACKAGE_libupm-water-python3 is not set
# CONFIG_PACKAGE_libupm-waterlevel-python is not set
# CONFIG_PACKAGE_libupm-waterlevel-python3 is not set
# CONFIG_PACKAGE_libupm-wfs-python is not set
# CONFIG_PACKAGE_libupm-wfs-python3 is not set
# CONFIG_PACKAGE_libupm-wheelencoder-python is not set
# CONFIG_PACKAGE_libupm-wheelencoder-python3 is not set
# CONFIG_PACKAGE_libupm-wt5001-python is not set
# CONFIG_PACKAGE_libupm-wt5001-python3 is not set
# CONFIG_PACKAGE_libupm-xbee-python is not set
# CONFIG_PACKAGE_libupm-xbee-python3 is not set
# CONFIG_PACKAGE_libupm-yg1006-python is not set
# CONFIG_PACKAGE_libupm-yg1006-python3 is not set
# CONFIG_PACKAGE_libupm-zfm20-python is not set
# CONFIG_PACKAGE_libupm-zfm20-python3 is not set
# CONFIG_PACKAGE_libelektra-python2 is not set
# CONFIG_PACKAGE_libelektra-python3 is not set
# CONFIG_PACKAGE_openvswitch-python is not set
```

mittlereSchiene, 2019-02-25:
@jefferyto 
Sorry my fault, everything works well now.
Thank you!

closed 2019-02-25 by mittlereSchiene