openwrt/packages #8283: python3-openssl: build error

leonardpitzu, 2019-02-23:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: ubuntu 18.04 64bit/openwrt mian/latest

Description:
python3-openssl depends on more packages - python3-light for example. during the build process python3-openssl tries to add the same files as python3-light and the snipped error pops out.

```
Collected errors:
 * check_data_file_clashes: Package python3-light wants to install file /home/vmdev/openwrt/build_dir/target-mips_24kc_musl/root-ath79/usr/bin/python3
	But that file is already provided by package  * python3-openssl
 * check_data_file_clashes: Package python3-light wants to install file /home/vmdev/openwrt/build_dir/target-mips_24kc_musl/root-ath79/usr/bin/python3.7
	But that file is already provided by package  * python3-openssl
 * opkg_install_cmd: Cannot install package python3-openssl.
 * check_data_file_clashes: Package python3-light wants to install file /home/vmdev/openwrt/build_dir/target-mips_24kc_musl/root-ath79/usr/bin/python3
	But that file is already provided by package  * python3-openssl
 * check_data_file_clashes: Package python3-light wants to install file /home/vmdev/openwrt/build_dir/target-mips_24kc_musl/root-ath79/usr/bin/python3.7
	But that file is already provided by package  * python3-openssl
 * opkg_install_cmd: Cannot install package python3-light.
package/Makefile:65: recipe for target 'package/install' failed
```

jefferyto, 2019-02-23:
Sounds like #8263.

jefferyto, 2019-02-25:
@leonardpitzu Can you see if this issue is still present?

leonardpitzu, 2019-02-25:
@jefferyto works as expected now - binaries are correctly copied.

jefferyto, 2019-02-25:
I think this can be closed :smile:

BKPepe, 2019-02-25:
@hnyman please close, thank you.

closed 2019-02-26 by hnyman