openwrt/packages #8285: python: python3 urllib.request import error

leonardpitzu, 2019-02-23:
Maintainer: @commodo
Environment: ubuntu 18.04 64bit/openwrt main/latest - ath79

Description:
urllib (py3) is missing some dependencies

```
File "/request.py", line 86, in <module>
ModuleNotFoundError: No module named 'email'
```

commodo, 2019-02-24:
Will take a look.

BKPepe, 2019-02-25:
I have successfully flashed OpenWrt SNAPSHOT, r9420-c17a68c to my router to be able to reproduce your issue, then I install python3-urllib3 and it installed a lot of stuff among python3-email.

See my log here: https://pastebin.com/dqrFi5um also you can find there that I have followed https://urllib3.readthedocs.io/en/latest/

leonardpitzu, 2019-02-26:
I have not used urllib3 but the stock urllib that comes with py3. That one misses the email lib. Point was the default is unusable as it missses a critical dependency.

commodo, 2019-02-26:
@leonardpitzu 

try this:
```
opkg update
opkg install python3-email
```

this seems to work for me

`python3-email` is a separate package, in order to reduce the size of Python3; I know it's not a great thing; maybe we could split `python3-urllib` into it's own package and see;

the split of python3 & python packages was done arbitrary
first to create the python[3]-base packages which are the minimal packages required for python[3] to start, then python[3]-light, some generic libraries from core Python, and then others were split away into other packages

leonardpitzu, 2019-02-26:
sure, it works like this (already did it, works as expected). The idea was to 1) remove urllib entirely and make it a stand-alone package, or 2) add the necessary dependencies to make it work "out of the box". urllib currently is useless and if you build openwrt from scratch you'll notice that upon running py3 it fails so you need to build it again to add email and then a third time to add idna (encodings) :)

commodo, 2019-02-26:
> sure, it works like this (already did it, works as expected).

oh, ok ; that wasn't obvious to me the first time;

i prefer 1) if that's ok

leonardpitzu, 2019-02-26:
@commodo 1) is perfect. A stripped down py interpreter is great. Then everybody can add stuff as needed.
Guess you can close the issue whenever you have time to split urllib into a package.

jefferyto, 2019-03-11:
@leonardpitzu @hnyman I think this can be closed?

leonardpitzu, 2019-03-11:
@jefferyto yes, thanks!

closed 2019-03-11 by hnyman