openwrt/packages #8289: Freeradius3 because openssl 1.1.1 Error

nasbdh9, 2019-02-24:
Maintainer: @cotequeiroz 
Environment: mipsel_24kc, R3G, OpenWrt master
                      mvebu, EBIN, OpenWrt master

**The following error occurred while using official build**
```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for freeradius3:
 *      libopenssl1.0.0
 * opkg_install_cmd: Cannot install package freeradius3.

root@OpenWrt:~# radiusd -X
Error loading shared library libcrypto.so.1.0.0: No such file or directory (needed by /usr/sbin/radiusd)
Error loading shared library libssl.so.1.0.0: No such file or directory (needed by /usr/sbin/radiusd)
Error loading shared library libtalloc.so.2: No such file or directory (needed by /usr/sbin/radiusd)
Error loading shared library libreadline.so.8: No such file or directory (needed by /usr/sbin/radiusd)
Error loading shared library libcrypto.so.1.0.0: No such file or directory (needed by /usr/lib/freeradius3/libfreeradius-server.so)
Error loading shared library libssl.so.1.0.0: No such file or directory (needed by /usr/lib/freeradius3/libfreeradius-server.so)
Error loading shared library libcrypto.so.1.0.0: No such file or directory (needed by /usr/lib/freeradius3/libfreeradius-radius.so)
Error loading shared library libssl.so.1.0.0: No such file or directory (needed by /usr/lib/freeradius3/libfreeradius-radius.so)
Error loading shared library libtalloc.so.2: No such file or directory (needed by /usr/lib/freeradius3/libfreeradius-radius.so)
Error loading shared library libreadline.so.8: No such file or directory (needed by /usr/lib/freeradius3/libfreeradius-radius.so)
Error loading shared library libpcap.so.1: No such file or directory (needed by /usr/lib/freeradius3/libfreeradius-radius.so)
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_get_size: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_steal_loc: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_set_destructor: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_version_minor: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: SSLeay_version: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_realloc_array: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_parent: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_zero: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_free: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_vasprintf: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_init: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_version_major: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_array: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_get_type_abort: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: SSLeay: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_strndup: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_strdup: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_set_name_const: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_memdup: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_strdup_append: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_zero_array: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_asprintf: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: talloc_named_const: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-server.so: _talloc_realloc: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_set_name_const: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_asprintf: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: pcap_set_buffer_size: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_disable_null_tracking: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: _talloc_set_destructor: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: pcap_lookupnet: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: _talloc_array: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: MD5_Update: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: pcap_dump_open: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: _talloc_realloc_array: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: pcap_datalink: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: pcap_open_live: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: _talloc_zero: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: MD5_Init: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_named_const: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_set_log_fn: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: pcap_dump_close: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_vasprintf: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_set_abort_fn: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_parent: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: _talloc_memdup: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: talloc_get_size: symbol not found
Error relocating /usr/lib/freeradius3/libfreeradius-radius.so: pcap_compile: symbol not found
[... 235 lines cut ...]
CC src/main/map.c
CC src/main/regex.c
CC src/main/tmpl.c
CC src/main/util.c
CC src/main/pair.c
CC src/main/xlat.c
/home/nasbdh9/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/7.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/freeradius-server-release_3_0_17/build/objs/src/main/tls.o: in function `tls_global_cleanup':
tls.c:(.text+0x2ef6): undefined reference to `ENGINE_cleanup'
collect2: error: ld returned 1 exit status
scripts/boiler.mk:628: recipe for target '/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/freeradius-server-release_3_0_17/build/bin/local/radeapclient' failed
make[4]: *** [/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/freeradius-server-release_3_0_17/build/bin/local/radeapclient] Error 1
make[4]: Leaving directory '/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/freeradius-server-release_3_0_17'
Makefile:607: recipe for target '/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/freeradius-server-release_3_0_17/.built' failed
make[3]: *** [/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/freeradius-server-release_3_0_17/.built] Error 2
make[3]: Leaving directory '/home/nasbdh9/openwrt/feeds/packages/net/freeradius3'
time: package/feeds/packages/freeradius3/compile#14.92#2.88#17.89
package/Makefile:107: recipe for target 'package/feeds/packages/freeradius3/compile' failed
make[2]: *** [package/feeds/packages/freeradius3/compile] Error 2
make[2]: Leaving directory '/home/nasbdh9/openwrt'
package/Makefile:103: recipe for target '/home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/nasbdh9/openwrt'
/home/nasbdh9/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

cotequeiroz, 2019-02-25:
@LucileQ is the actual maintainer for the package, you may want to adjust the post.
I imagine it depends on engine support, or deprecated API.  I will try to tackle this and post a PR if able--it shouldn't be too hard.  The first, binary package, error is a consequence of the failure to build, and should go away once the build problem is resolved.

cotequeiroz, 2019-02-25:
Engine support is not necessary, so my fix will be to conditionally disable that call, checking openssl's own opensslconfig.h definition of `OPENSSL_NO_ENGINE`.  I will silence that `RAND_load_file` warning as well.  I have opened FreeRADIUS/freeradius-server#2517 as well.

PR #8292 (2019-03-10): freeradius3: allow build without openssl engines
Maintainer: @LucileQ
Compile tested: ath79/mips_24kc & mvebu/arm, openwrt master
Run tested: None - @nasbdh9, can you run-test this, please.

Description:
This applies a patch comitted upstream that avoids a call to `ENGINE_cleanup` when the openssl library was built without engine support.

A workaround for a missing header check was used to silence a warning about the implicit definition of `RAND_load_file`.  A proper fix was submitted and merged upstream as well, and a note was left in the Makefile to check for the proper fix in future releas [... cut]
closed 2019-03-10 by hnyman