openwrt/packages #8300: smartmontools failing with buildbots

neheb, 2019-02-25:
https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/smartmontools/compile.txt as an example. I think it has to do with the new drivedb work.

@mstorchak

mstorchak, 2019-02-26:
It's working now. I was not able to download that log - it was already gone when I noticed this bug.
https://downloads.openwrt.org/snapshots/packages/x86_64/packages/smartmontools-drivedb_7.0-2_all.ipk

mstorchak, 2019-02-26:
https://downloads.openwrt.org/snapshots/faillogs/arm_arm1176jzf-s_vfp/packages/smartmontools/compile.txt contains
```
cp: cannot overwrite non-directory '/opt/buildbot/slaves/lede-slave-tah/arm_arm1176jzf-s_vfp/build/sdk/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/root-brcm2708/./usr/share/smartmontools' with directory '/opt/buildbot/slaves/lede-slave-tah/arm_arm1176jzf-s_vfp/build/sdk/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/smartmontools-7.0/.pkgdir/smartmontools-drivedb/./usr/share/smartmontools'
```
which looks like a result from improper cleanup of the previous build. Before #8220 the *file* `/usr/share/smartmontools` contained drive database, and now it's a directory containing `drivedb.h`

mstorchak, 2019-02-28:
@neheb , is there anything  I can do from my side?

diizzyy, 2019-03-14:
@neheb 
Looks like this is resolved on the buildbots?

neheb, 2019-03-14:
It appears from time to time. But doesn't look like a big deal. Closing.

closed 2019-03-14 by neheb