openwrt/packages #8303: miniupnpd floods syslog

xnoreq, 2019-02-26:
Maintainer: @neheb 
Environment: mips (tl-wdr3600 v1), openwrt 18.06.02

Description:
miniupnpd spams syslog with the following messages (seems to be with each notify, so every 30s):
```
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=477, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=473, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=459, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=445, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=465, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=469, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=397, dest=[ff0e::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=477, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=473, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=459, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=445, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=465, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=469, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=397, dest=[ff05::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=477, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=473, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=461, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=459, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=445, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=465, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=406, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=469, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=8, len=397, dest=[ff02::c]:1900): sendto: Permission denied
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=469, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=469, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=485, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=481, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=469, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=467, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=414, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=453, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=414, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=473, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=414, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=477, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto(sock=7, len=405, dest=239.255.255.250:1900): sendto: No such device
Tue Feb 26 23:11:09 2019 daemon.err miniupnpd[2105]: try_sendto failed to send 52 packets
```

What is going  wrong here?

neheb, 2019-02-27:
(I don't maintain miniupnpd)

The first problem is IPv6 specific. I've seen it on transmission (which I do maintain). The addresses shown I haven't seen before. Specific to your setup I guess.

neheb, 2019-07-02:
This was updated. You could try again.

th0ma7, 2019-11-18:
Having this exact same problem using 19.07.0-rc1.

Here is my config, pretty much default:
```
# uci show upnpd.config
upnpd.config=upnpd
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.config.enabled='1'
upnpd.config.uuid='51a30fff-5b23-4a14-a158-1ae6ead01096'
```
I noticed that the following doc does mention `config.external_iface='wan'` which is missing in my case so I tried adding it up:
```
root@th0ma7-router:~# uci set upnpd.config.external_iface='wan'
root@th0ma7-router:~# uci commit
root@th0ma7-router:~# /etc/init.d/miniupnpd restart
```
Perhaps due to restart but it does seems like fixing the issue...

greekstreet, 2019-12-05:
> Having this exact same problem using 19.07.0-rc1.
>
> Here is my config, pretty much default:
>
> ```
> # uci show upnpd.config
> upnpd.config=upnpd
> upnpd.config.enable_natpmp='1'
> upnpd.config.enable_upnp='1' [...]

Yes, restart miniupnpd manually fix the issue. I encountered the same issue. I Googled and found someone mentioned the fix in script located in /etc/hotplug.d/iface/50_miniupnpd. It's there but it seems not working well.

Mushoz, 2020-01-14:
Seeing the same spam in my log since updating to 19.07.0 final release. Using default upnp settings. So a simple restart should fix it? What's causing the issue in the first place, and why does a restart fix it?

neheb, 2020-02-09:
I've reproduced this.

I will fix with https://github.com/openwrt/packages/pull/11275

neheb, 2020-02-14:
Fixed by increasing the number of the hotplug script. I no longer get these errors.

Please reopen if the issue persists.

closed 2020-02-14 by neheb

locojohn, 2020-04-25:
I am getting the same errors in the system log now:

```
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=474, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=490, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=486, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=474, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=472, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=419, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=458, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=419, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=478, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=419, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=482, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=410, dest=[ff05::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=474, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=474, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=490, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=486, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=474, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=472, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=419, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=458, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=419, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=478, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=419, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=482, dest=[ff02::c]:1900): sendto: Address not available
Sat Apr 25 22:52:02 2020 daemon.err miniupnpd[3045]: try_sendto(sock=8, len=410, dest=[ff02::c]:1900): sendto: Address not available

```

# more /etc/os-release
NAME=q"OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r13062-ef5d37ba5e"
OPENWRT_BOARD="ipq806x/generic"
OPENWRT_ARCH="arm_cortex-a15_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r13062-ef5d37ba5e"

neheb, 2020-04-25:
That's unfortunate. Just restart the daemon.

ptpt52, 2020-08-13:
I am trying to add a lock to the hotplug script
Please try test
```
diff --git a/net/miniupnpd/files/miniupnpd.hotplug b/net/miniupnpd/files/miniupnpd.hotplug
index 63d6114b6..ea6fd238e 100644
--- a/net/miniupnpd/files/miniupnpd.hotplug
+++ b/net/miniupnpd/files/miniupnpd.hotplug
@@ -41,4 +41,6 @@ fi
 
 grep -qs "^ext_ifname=$ifname" "$tmpconf" && grep -qs "^ext_ifname6=$ifname6" "$tmpconf" && exit 0
 
+lock /var/run/miniupnpd.lock
 /etc/init.d/miniupnpd restart
+lock -u /var/run/miniupnpd.lock

```

ptpt52, 2020-08-13:
update: lock more range
```
diff --git a/net/miniupnpd/files/miniupnpd.hotplug b/net/miniupnpd/files/miniupnpd.hotplug
index 63d6114b6..4948b79e2 100644
--- a/net/miniupnpd/files/miniupnpd.hotplug
+++ b/net/miniupnpd/files/miniupnpd.hotplug
@@ -9,6 +9,8 @@
 
 [ "$ACTION" != "ifup" ] && /etc/init.d/miniupnpd running && exit 0
 
+lock /var/run/miniupnpd.lock
+
 tmpconf="/var/etc/miniupnpd.conf"
 external_iface=$(uci -q get upnpd.config.external_iface)
 external_iface6=$(uci -q get upnpd.config.external_iface6)
@@ -37,8 +39,16 @@ else
        fi
 fi
 
-[ "$DEVICE" != "$ifname" ] && [ "$DEVICE" != "$ifname6" ] && exit 0
+if [ "$DEVICE" != "$ifname" ] && [ "$DEVICE" != "$ifname6" ]; then
+       lock -u /var/run/miniupnpd.lock
+       exit 0
+fi
 
-grep -qs "^ext_ifname=$ifname" "$tmpconf" && grep -qs "^ext_ifname6=$ifname6" "$tmpconf" && exit 0
+if grep -qs "^ext_ifname=$ifname" "$tmpconf" && grep -qs "^ext_ifname6=$ifname6" "$tmpconf"; then
+       lock -u /var/run/miniupnpd.lock
+       exit 0
+fi
 
 /etc/init.d/miniupnpd restart
+
+lock -u /var/run/miniupnpd.lock

```